zigpy/zha-device-handlers #3589: _TZE204_ntcy3xu1 Smoke detector, battery missing

erkr, 2024-12-07:
### Bug description

Hi, I own a Tuya smoke detector.  With the released quirk, only the smoke sensor is exposed, but the battery entity is missing.
Model _TZE204_ntcy3xu1. 

### Steps to reproduce

Na

### Expected behavior

Please Include the battery entity as well. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

prairiesnpr, 2024-12-08:
That looks like expected behavior, DP 14 is a boolean that indicates if the battery is low according to the corresponding z2m converter.

erkr, 2024-12-08:
> That looks like expected behavior, DP 14 is a boolean that indicates if the battery is low according to the corresponding z2m converter.

Thanks. Then the 
Custom Quirk implementation will right. How to get is added to the official quirk for this device?

prairiesnpr, 2024-12-08:
Does this quirk work? Need to be on HA 2024.11 or newer.

```python
from zhaquirks.tuya import TuyaLocalCluster
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.tuya.builder import TuyaPowerConfigurationCluster2AAA, TuyaQuirkBuilder
from zigpy.quirks.v2 import BinarySensorDeviceClass, EntityType

class TuyaIAS(IasZone, TuyaLocalCluster):
    """Tuya local IAS cluster."""

    _CONSTANT_ATTRIBUTES = {
        IasZone.AttributeDefs.zone_type.id: IasZone.ZoneType.Fire_Sensor
    }

(
    TuyaQuirkBuilder("_TZE204_ntcy3xu1", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaIAS.ep_attribute,
        attribute_name=IasZone.AttributeDefs.zone_state.name,
        converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x == 0 else 0,
    )
    .tuya_dp(
        dp_id=14,
        ep_attribute=TuyaPowerConfigurationCluster2AAA.ep_attribute,
        attribute_name="battery_percentage_remaining",
        converter=lambda x: {0: 5, 1: 100}[x],
    )
    .tuya_binary_sensor(
        dp_id=4,
        attribute_name="tamper_status",
        device_class=BinarySensorDeviceClass.TAMPER,
        fallback_name="Tamper",
        entity_type=EntityType.DIAGNOSTIC,
    )
    .adds(TuyaPowerConfigurationCluster2AAA)
    .adds(TuyaIAS)
    .skip_configuration()
    .add_to_registry()
)
```

erkr, 2024-12-08:
@prairiesnpr   Yes that works!!! Wauw

erkr, 2024-12-08:
@prairiesnpr 
Done testing is good. There V2 quirk creates the right entities, but those don't seem to work.

Note that Tamper doesn't work in both quirks. That one probably could be left out. But smoke detect doesn't work in V2 (works fine in the V1)

prairiesnpr, 2024-12-08:
Can you grab logs on what value is being returned when it detects smoke and when cleared?

erkr, 2024-12-08:
Yes, but some guidance please. Should I enable the debug log for Zha component?

prairiesnpr, 2024-12-08:
Yes, enable debug logging in ZHA, then if you can cut it to just the timestamps when are testing we should see the traffic and be able to tell where I went wrong.

erkr, 2024-12-08:
I logged with my old V1 quirk. Device info: 
```
IEEE: a4:c1:38:66:3a:ee:86:34
Nwk: 0xcd84
Device Type: EndDevice
LQI: 255
RSSI: -79
Last seen: 2024-12-08T15:42:20
Power source: Battery or Unknown
Quirk: TS0601_TZE204_ntcy3xu1.TuyaSmokeDetector0601
```
Look at timestamp 15:42:19.464

Thanks for your help!
[ZHA debug Smoke.txt](https://github.com/user-attachments/files/18052681/ZHA.debug.Smoke.txt)

erkr, 2024-12-08:
Don't know if it is relevant; I enabled the V2 quirk by reloading ZHA, not a full restart

prairiesnpr, 2024-12-08:
> Don't know if it is relevant; I enabled the V2 quirk by reloading ZHA, not a full restart For reverting to the V1, reload wasn't enough, so I did full restart for that

Looked at logs, battery is an enum, and I had used zone_state instead of zone_status, please test the updated quirk above and let me know if that does it.

prairiesnpr, 2024-12-08:
> Note that Tamper doesn't work in both quirks. That one probably could be left out. But smoke detect doesn't work in V2 (works fine in the V1)

I didn't see any traffic in your logs related to the tamper sensor unless so going to remove it.

erkr, 2024-12-08:
Test it tomorrow 👌

prairiesnpr, 2024-12-09:
> > Test it tomorrow 👌
> 
> Just tried the new version. It only offers the battery. Alarm sensor is gone.
> 
> PS: I didn't trigger tamper in the provided log. I will record one for tamper as well

My bad, cp error, needed `.adds(TuyaIAS)`, added it above.

erkr, 2024-12-09:
@prairiesnpr 
Improved, the alarm sensor in present and triggers. But afterwards it doesn't clear!

prairiesnpr, 2024-12-09:
That's very odd, if you look at the PR, I actually have a test using the two packets from your logs above, the trigger and the following clear. Maybe worth trying a restart?

erkr, 2024-12-09:
I will try that!
I checked, there is noting logged when triggering tamper

prairiesnpr, 2024-12-09:
> Other note: All the testing seems to deplete the batteries quickly. Strange is that the battery reports 50% (instead of the coded 40% in the quirk). Doubth if this line is working (and maybe not needed): ` converter=lambda x: {0: 5, 1: 40, 2: 100}[x],`

Zigbee values are supposed to be doubled, I don't recall where the conversation takes place right now though. You can try `converter=lambda x: {0: 10, 1: 80, 2: 200}[x],`

I'm at work right now, will look at your logs this evening.

erkr, 2024-12-09:
> Zigbee values are supposed to be doubled, I don't recall where the conversation takes place right now though. You can try `converter=lambda x: {0: 10, 1: 80, 2: 200}[x],`

Changed the line and waiting for the next battery update...

Thanks for your help

prairiesnpr, 2024-12-09:
> Update: with this 'doubling modification' it reports 100% for raw value 0x02. These relevant lines where in the log when pressing the test button:

Perfect, PR is updated once it's merged you should be good.

erkr, 2024-12-09:
@prairiesnpr 
I'm not that familiar with reviewing commits, but looking at the latest battery update, the file ts0601_smoke contains this:
```
(
    TuyaQuirkBuilder("_TZE204_ntcy3xu1", "TS0601")
    .tuya_ias(dp_id=1)
    .tuya_dp(
        dp_id=14,
        ep_attribute=TuyaPowerConfigurationCluster2AAA.ep_attribute,
        attribute_name="battery_percentage_remaining",
        converter=lambda x: {0: 10, 1: 80, 2: 200}[x],
    )
    .adds(TuyaPowerConfigurationCluster2AAA)
    .skip_configuration()
    .add_to_registry()
)
```
Is the DP1 specialization not missing here?:
```
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaIAS.ep_attribute,
        attribute_name=IasZone.AttributeDefs.zone_status.name,
        converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x == 0 else 0,
    )
```

prairiesnpr, 2024-12-09:
> Or is that the default for:
> 
> ```
>     .tuya_ias(dp_id=1)
> ```

That's correct, see https://github.com/prairiesnpr/zha-device-handlers/blob/dee4296a65f49efee31231fd9807b5c3b67bb45b/zhaquirks/tuya/builder/__init__.py#L122

Since all of the smoke quirks used the same logic, I took the opportunity to add a helper method to TuyaQuirkBuilder and update all of the old style quirks in the same PR.

erkr, 2024-12-09:
Learning all the time, thanks, looks great. Looking forward to have it in the release
best Eric

erkr, 2024-12-09:
@prairiesnpr  Sorry, I have another small question. Since I played with this custom quirk, Reloading the ZHA integration gives this error: 
`zigpy.exceptions.MultipleQuirksMatchException: Multiple matches found for device <Device model='TS0601' manuf='_TZE204_ntcy3xu1'`
When I fully restart HA, ZHA loads normal. Is there some cache I can cleanup ?!

closed 2024-12-09 by TheJulianJES