zigpy/zha-device-handlers #3591: [BUG] Philips Hue RDM001 simultaneous press not processed

ericu65, 2024-12-07:
### Bug description

After updating to HA 2024.11 the associated ZHA quirk for Philips Hue Wall Switch RDM001 set in mode 3 (dual rocker) no longer processes simultaneous switching of both switches. When switching both only one of the switches is processed and the second never appears in HA logbook. 

After rolling back init.py and rdm001.py as a custom quirk from commit [c9c9e37](https://github.com/zigpy/zha-device-handlers/commit/c9c9e37395a9c8c1a84bc8e992a413916892e1c9) the behavior now works as before again: switching both switches at the same time will trigger events for left (unnamed) and right switch.

### Steps to reproduce

Switch both switches in mode 3. 
Only one events goes through.

### Expected behavior

Both events go through.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

ericu65, 2024-12-07:
Seems commit https://github.com/zigpy/zha-device-handlers/commit/d84d94f8c7289c048b3f89a7d6cb8e8d022d7570#diff-493c9c4c97806b1bcce0822f1ece0cad303ee32e4eac25b503e956ec83f3b587 introduced this issue. Version before this works fine in terms of simultaneous presses. 
Introduced with PR https://github.com/zigpy/zha-device-handlers/pull/2340

I would dig deeper but the diff is pretty large. @fholzer if you have any tips where I should look I could dig deeper myself?

fholzer, 2024-12-08:
Hey, you say you have a dual rocker and use mode 3. Dual rocker should be mode 2, see https://zigbee.blakadder.com/Philips_RDM001.html. Mode 3 is double push. Either way, I think I know what the issue is. The Button Queue doesn't support pressing multiple buttons at the same time. The main purpose of the queue is to support multi-press events, though it can also have impact on single presses. If different buttons are pressed withing the Button Queue threshold (300ms) only the last button received will be considered. I have am working on a fix, but would like to do some more testing in for the different switch modes, and also for the other remotes that use this Button Queue.

Could you maybe test using philips files from here: https://github.com/fholzer/zha-device-handlers/tree/fix-philips-multi-button-press

ericu65, 2024-12-08:
Wow, that was fast. Thanks for looking into it. 

Indeed you're right, I was using enum8.undefined_0x02 for mode (not mode 3 as I had mistakenly remembered). 

I tested both the fork and the PR without success. Is the __init__.py called correctly from the custom_quicks directory or do I need to place it someplace else?

If I should grab some events logs from RDM001 in mode 0x02 with simultaneous presses let me know.

fholzer, 2024-12-08:
Hey you'll need both, the init and wallswitch py files from my branch. Note that i made a change to the mode parameter names, so instead of using raw integer values when changing the mode, you can now use one of these strings:

* SingleRocker
* SinglePush
* DoubleRocker
* DoublePush

Please test, and in case you still experience issues, it would be great to get debug logs.

ericu65, 2024-12-08:
This last part is working: I get SwitchMode.DoubleRocker when reading the mode. Cool improvment.

I have both init and wallswitch in custom_quirks. Below the logs for a single simultaneous switch of left and right (but opposite - one from OFF to ON, the other from ON to OFF) from ZHA Debug logging. 

### Logs

<details><summary>Logs</summary>

```python
2024-12-08 19:09:40.021 DEBUG (MainThread) [zha.application.helpers] Global updater interval finished
2024-12-08 19:10:01.110 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ea6000700aa00
2024-12-08 19:10:01.112 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.device_state_changed{'status': <Status.SUCCESS: 0>, 'frame_length': 7, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE|32: 42>), 'reserved': 0} (seq 166)
2024-12-08 19:10:01.113 DEBUG (MainThread) [zigpy_deconz.api] Sending CommandId.aps_data_indication{'flags': <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>} (seq=166)
2024-12-08 19:10:01.116 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17a6002b002400220200000102a8e401040100fc0d001d0b107100010000300021000000affff43def01b7
2024-12-08 19:10:01.119 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.aps_data_indication{'status': <Status.SUCCESS: 0>, 'frame_length': 43, 'payload_length': 36, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE: 8>), 'dst_addr': DeconzAddress(address_mode=<AddressMode.NWK: 2>, address=0x0000), 'dst_ep': 1, 'src_addr': DeconzAddress(address_mode=<AddressMode.NWK: 2>, address=0xE4A8), 'src_ep': 1, 'profile_id': 260, 'cluster_id': 64512, 'asdu': b'\x1d\x0b\x10q\x00\x01\x00\x000\x00!\x00\x00', 'reserved1': 0, 'reserved2': 175, 'lqi': 255, 'reserved3': 244, 'reserved4': 61, 'reserved5': 239, 'reserved6': 1, 'rssi': -73} (seq 166)
2024-12-08 19:10:01.121 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 12, 8, 17, 10, 1, 121219, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE4A8), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=64512, data=Serialized[b'\x1d\x0b\x10q\x00\x01\x00\x000\x00!\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-73)
2024-12-08 19:10:01.123 DEBUG (MainThread) [zigpy.zcl] [0xE4A8:1:0xfc00] Received ZCL frame: b'\x1d\x0b\x10q\x00\x01\x00\x000\x00!\x00\x00'
2024-12-08 19:10:01.124 DEBUG (MainThread) [zigpy.zcl] [0xE4A8:1:0xfc00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x1D>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4107, tsn=113, command_id=0, *direction=<Direction.Server_to_Client: 1>)
2024-12-08 19:10:01.125 DEBUG (MainThread) [zigpy.zcl] [0xE4A8:1:0xfc00] Decoded ZCL frame: PhilipsWallSwitchRemoteCluster:notification(button=1, param2=3145728, press_type=0, param4=33, param5=0)
2024-12-08 19:10:01.125 DEBUG (MainThread) [zigpy.zcl] [0xE4A8:1:0xfc00] Received command 0x00 (TSN 113): notification(button=1, param2=3145728, press_type=0, param4=33, param5=0)
2024-12-08 19:10:01.127 DEBUG (MainThread) [zhaquirks.philips] PhilipsWallSwitchRemoteCluster - handle_cluster_request tsn: [113] command id: 0 - args: [notification(button=1, param2=3145728, press_type=0, param4=33, param5=0)]
2024-12-08 19:10:01.127 DEBUG (MainThread) [zhaquirks.philips] PhilipsWallSwitchRemoteCluster - handle_cluster_request button id: [1], button name: [<Button id=left, trigger=turn_on, action=left>]
2024-12-08 19:10:01.128 DEBUG (Thread-7) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f4e6258a0>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1733677801.121219, 'ieee': 00:17:88:01:0c:c3:de:45, 'min_update_delta': 30.0})
2024-12-08 19:10:01.131 DEBUG (Thread-7) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f4e6258a0>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1733677801.121219, 'ieee': 00:17:88:01:0c:c3:de:45, 'min_update_delta': 30.0}) completed
2024-12-08 19:10:01.133 DEBUG (Thread-7) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f4e6258a0>)
2024-12-08 19:10:01.134 DEBUG (Thread-7) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f4e6258a0>) completed
2024-12-08 19:10:01.142 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ea7000700aa00
2024-12-08 19:10:01.143 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.device_state_changed{'status': <Status.SUCCESS: 0>, 'frame_length': 7, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE|32: 42>), 'reserved': 0} (seq 167)
[... cut ...]
2024-12-08 19:10:01.704 DEBUG (Thread-7) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f4e6258a0>) completed
2024-12-08 19:10:02.143 DEBUG (MainThread) [zha.decorators] [<Task pending name='device-availability-checker_DeviceAvailabilityChecker' coro=<periodic.<locals>.scheduler.<locals>.wrapper() running at /usr/local/lib/python3.13/site-packages/zha/decorators.py:78> cb=[set.remove()]>] executing periodic task [zha.application.helpers::DeviceAvailabilityChecker.check_device_availability]

``` 
</details>

fholzer, 2024-12-08:
Are you sure you also replaced your `__init__.py` with the one from my branch?

ericu65, 2024-12-09:
I did. But I was wondering if placing it into the custom quirk is enough or it needs to go some other place. Here's the folder structure and a small sample of the init.py which was changed in the latest PR:
[image]

After placing them I reload the ZHA integration and test.

fholzer, 2024-12-09:
Right. What I found while initially working on these changes is that it doesn't work putting these in the custom quirks folder. Not entirely sure why. I usually put these files into python's site-packages folder. (Not directly, but into the proper path inside the zha-device-handlers directory inside site-packages)
Depending on how you run home assistant, there's different ways of doing that...

ericu65, 2024-12-09:
Test confirms issue is fixed in your fork @fholzer 
As suggested, the issue was `__init__.py` not being loaded from the custom folder. After replacing them in the home assistant image directly it now works properly with simultaneous presses. I can't confirm if there's any impact to multi-press events as I don't use them (but I'd like to with the Tap Dial once I move it from ZHA from Philips). 

Many thanks.

PR #3592 (2024-12-22): Support multi-button presses on Philips remotes
## Proposed change
ButtonPressQueue currently resets when receiving a button press from a different button than the previously received one. This prevents proper handling when pressing multiple buttons at the same time.

## Additional information
Fixes #3591 

This is still a draft. Would like to add another test to make sure multiple button presses are handled correctly. So far tests only check for multi-presses of the same button. Also want to wait for feedback from ericu65 in #3591

## Checklist

- [x] The changes are tested and work correctly [... cut]
closed 2024-12-22 by TheJulianJES