zigpy/zha-device-handlers #3597: [BUG] INVALID_DATA_TYPE error df

jeverley, 2024-12-08:
### Bug description

Using enum types to document attribute values in the new quirk I wrote has inadvertently resulted in the incorrect zigbee attribute type being sent to the device in write_attributes requests for the motion_sensitivity attribute.

I had not appreciated at the time that whilst it's derived from uint8_t, the code in https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/foundation.py maps the two types to separate zigbee type values (0x20 and 0x30).

This should be corrected by using the device's native uint8 type (raising a PR shortly).

### Steps to reproduce

Attempt to change the device Motion Sensitivity setting through the select entity/manage device attribute write.

### Expected behavior

The new value writes successfully to the device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python

2024-12-08 11:25:34.970 DEBUG (MainThread) [zigpy.zcl] [0xBBCE:1:0xfcc0] Sending request header: ZCLHeader(frame_control=FrameControl<0x04>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4660, tsn=3, command_id=<GeneralCommand.Write_Attributes: 2>, *direction=<Direction.Client_to_Server: 0>)
2024-12-08 11:25:34.972 DEBUG (MainThread) [zigpy.zcl] [0xBBCE:1:0xfcc0] Sending request: Write_Attributes(attributes=[Attribute(attrid=0x010C, value=TypeValue(type=AqaraMotionSensitivity, value=<AqaraMotionSensitivity.High: 3>))])
2024-12-08 11:25:34.977 DEBUG (MainThread) [zigpy.device] [0xbbce] Extending timeout for 0x03 request
2024-12-08 11:25:34.977 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 12, 8, 11, 25, 34, 977730, tzinfo=datetime.timezone.utc), priority=<PacketPriority.NORMAL: 0>, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBBCE), dst_ep=1, source_route=None, extended_timeout=True, tsn=3, profile_id=260, cluster_id=64704, data=Serialized[b'\x044\x12\x03\x02\x0c\x010\x03'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2024-12-08 11:25:34.978 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command  getExtendedTimeout: () {'remoteEui64': 54:ef:44:10:00:db:47:bb}
2024-12-08 11:25:34.987 DEBUG (MainThread) [bellows.ezsp.protocol] Received command getExtendedTimeout: {'extendedTimeout': <Bool.true: 1>}
2024-12-08 11:25:34.988 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command  sendUnicast: () {'type': <EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 'indexOrDestination': 0xBBCE, 'apsFrame': EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=3), 'messageTag': 37, 'messageContents': b'\x044\x12\x03\x02\x0c\x010\x03'}
2024-12-08 11:25:35.000 DEBUG (MainThread) [bellows.ezsp.protocol] Received command sendUnicast: {'status': <EmberStatus.SUCCESS: 0>, 'sequence': 71}
2024-12-08 11:25:35.038 DEBUG (MainThread) [bellows.ezsp.protocol] Received command messageSentHandler: {'type': <EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 'indexOrDestination': 48078, 'apsFrame': EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=71), 'messageTag': 37, 'status': <EmberStatus.SUCCESS: 0>, 'messageContents': b''}
2024-12-08 11:25:35.039 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 48078, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=71), 37, <EmberStatus.SUCCESS: 0>, b'']
2024-12-08 11:25:35.045 DEBUG (MainThread) [bellows.ezsp.protocol] Received command incomingSenderEui64Handler: {'senderEui64': 54:ef:44:10:00:db:47:bb}
2024-12-08 11:25:35.045 DEBUG (MainThread) [bellows.zigbee.application] Received incomingSenderEui64Handler frame with [54:ef:44:10:00:db:47:bb]
2024-12-08 11:25:35.051 DEBUG (MainThread) [bellows.ezsp.protocol] Received command incomingMessageHandler: {'type': <EmberIncomingMessageType.INCOMING_UNICAST: 0>, 'apsFrame': EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_SOURCE_EUI64|APS_OPTION_DESTINATION_EUI64: 3392>, groupId=0, sequence=128), 'lastHopLqi': 168, 'lastHopRssi': -58, 'sender': 0xBBCE, 'bindingIndex': 255, 'addressIndex': 9, 'messageContents': b'\x1c4\x12\x03\x04\x8d\x0c\x01'}
2024-12-08 11:25:35.052 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=64704, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_SOURCE_EUI64|APS_OPTION_DESTINATION_EUI64: 3392>, groupId=0, sequence=128), 168, -58, 0xBBCE, 255, 9, b'\x1c4\x12\x03\x04\x8d\x0c\x01']
2024-12-08 11:25:35.052 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 12, 8, 11, 25, 35, 52600, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBBCE), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=128, profile_id=260, cluster_id=64704, data=Serialized[b'\x1c4\x12\x03\x04\x8d\x0c\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=168, rssi=-58)
2024-12-08 11:25:35.055 DEBUG (MainThread) [zigpy.zcl] [0xBBCE:1:0xfcc0] Received ZCL frame: b'\x1c4\x12\x03\x04\x8d\x0c\x01'
2024-12-08 11:25:35.056 DEBUG (MainThread) [zigpy.zcl] [0xBBCE:1:0xfcc0] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x1C>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4660, tsn=3, command_id=4, *direction=<Direction.Server_to_Client: 1>)
2024-12-08 11:25:35.060 DEBUG (MainThread) [zigpy.zcl] [0xBBCE:1:0xfcc0] Decoded ZCL frame: OppleCluster:Write_Attributes_rsp(status_records=[WriteAttributesStatusRecord(status=<Status.INVALID_DATA_TYPE: 141>, attrid=0x010C)])
2024-12-08 11:25:35.062 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547359516000] Unexpected exception
Traceback (most recent call last):
[... 29 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
    await self.async_select_option(option)
  File "/usr/src/homeassistant/homeassistant/components/zha/helpers.py", line 1337, in handler
    raise HomeAssistantError(err) from err
homeassistant.exceptions.HomeAssistantError: Failed to write attribute motion_sensitivity=<AqaraMotionSensitivity.High: 3>: <Status.INVALID_DATA_TYPE: 141>
2024-12-08 11:25:35.092 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=system_log, service=write, service_data=logger=frontend.js.modern.202411276, message=Unhandled promise rejection from Edge 131.0.0.0 on Windows 10
{
  "code": "home_assistant_error",
  "message": "Failed to write attribute motion_sensitivity=<AqaraMotionSensitivity.High: 3>: <Status.INVALID_DATA_TYPE: 141>"
}, level=debug>

```

</details>

TheJulianJES, 2024-12-08:
Adding `zcl_type` should work: https://github.com/zigpy/zha-device-handlers/blob/97413e3ec2e5fdbcd17239b810d573cec4d3964d/zhaquirks/nodon/pilot_wire.py#L42

Otherwise, the attribute definition type can be changed from the enum to int.

jeverley, 2024-12-08:
> Adding `zcl_type` should work:
> 
> https://github.com/zigpy/zha-device-handlers/blob/97413e3ec2e5fdbcd17239b810d573cec4d3964d/zhaquirks/nodon/pilot_wire.py#L42
> 
> Otherwise, the attribute definition type can be changed from the enum to int.

I think I prefer this approach as it still allows for using the type as explicit documentation for the values - I'll amend PR https://github.com/zigpy/zha-device-handlers/pull/3598 accordingly (suspect I should rebase the branch anyway as it's pulling in all the pre-merge commits).

jeverley, 2025-01-05:
Confirmed this is now resolved in the 2025.1

closed 2025-01-05 by jeverley