zigpy/zha-device-handlers #3599: Heiman HS80S motion sensor

frillhouse, 2024-12-08:
### Problem description

TS0601
_TZ6210_duv6fhwt

No settings can be applied.

[image]

### Solution description

I want to access the device settings so I can use it in automation.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Vaarinvaliliha, 2025-01-07:
Same here! _TZ6210_duv6fhwt

petr-dolin, 2025-01-08:
[zha-01JDQ5MVS3N07XMS7BF8MZST1D-_TZ6210_duv6fhwt TS0601-e9090ea08bb5223a813ef7254e252639.json](https://github.com/user-attachments/files/18352698/zha-01JDQ5MVS3N07XMS7BF8MZST1D-_TZ6210_duv6fhwt.TS0601-e9090ea08bb5223a813ef7254e252639.json)
I just bought the same device. After adding it via ZHA, there are no entities available. Has anyone encountered a similar device, could similar quirks be used?
I added the diagnostic data from the device.

prairiesnpr, 2025-01-08:
> could similar quirks be used?

https://github.com/zigpy/zha-device-handlers/pull/3688 is based off the other motion quirks and a partial PR in z2m, feel free to test.

closed 2025-01-16 by TheJulianJES