zigpy/zha-device-handlers #3600: [BUG] Tuya TS130F Curtain (Roller Shutter) - not setting current_position_lift_percentage

mkz212, 2024-12-08:
### Bug description

Tuya TS130F TZ3000_eg7awg6a (single Roller Shutter) (Quirk: zhaquirks.tuya.ts130f.TuyaTS130GP) not saving current_position_lift_percentage - it is always 0 or 100.

Similar device, but Double Roller Shutter, TS130F _TZ3000_j1xl73iw (Quirk: zhaquirks.tuya.ts130f.TuyaTS130Double_GP_ESTC) works ok.

### Steps to reproduce

When I set position e.g. to 52% target lift position is 52 and curtain is moving to this position, but current_position_lift_percentage is 100. So after a while it shows wrong position in HA, even though the roller blind is in the correct position.

### Expected behavior

Fix position

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_eg7awg6a",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130GP"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

mkz212, 2024-12-10:
Using automation to set cluster works.

dabeschte, 2025-02-16:
hey @mkz212: you mentioned that using an automation works. what do you mean with that? 
My problem is that the position is always wrong after a couple hours independent of if I used HA or my hardware switches connected to the curtain module to control them.

Is there a solution to always save the current lift position with automations?

mkz212, 2025-02-17:
This is my automation as workaround for this. This one automation fix all covers. 

```python

alias: Napraw rolety
description: ""
triggers:
  - entity_id:
      - cover.m2_biuro_roleta
      - cover.m2_lazienka_roleta
      - cover.m2_kuchnia_wyspa_roleta
      - cover.m3_lazienka_roleta
      - cover.m1_salon_l_roleta
      - cover.m1_salon_r_roleta
      - cover.m1_sypialnia_roleta
      - cover.m4_roleta
      - cover.m5_roleta
    trigger: state
    to:
      - open
      - closed
    for:
      hours: 0
      minutes: 0
      seconds: 15
conditions: []
actions:
  - variables:
      device: "{{ trigger.entity_id }}"
      ieee: >-
        {{ device_attr(device, 'identifiers') | selectattr(0,'eq','zha') |
        map(attribute=1) | first }}
      position: "{{ state_attr(device, 'current_position') }}"
  - data:
      ieee: "{{ ieee }}"
      endpoint_id: 1
      cluster_id: 258
      cluster_type: in
      attribute: 8
      value: "{{ position }}"
    action: zha.set_zigbee_cluster_attribute
mode: parallel
max: 10

```

pchenriques, 2025-04-11:
It works, @mkz212. Thanks for the code. But for the absolut fix for the issue, what would we could expect from this open ticket?

pchenriques, 2025-04-11:
@mkz212 , thanks for the feedback. Hopefully we will get the fix at quirk level (I've read that the issue isn't happening while using MQTT).
Meanwhile, I'm following (again) your suggestion of splitting into device groups (I have 8 blinds and frequently get communication errors).

mkz212, 2025-04-11:
> [@mkz212](https://github.com/mkz212) , thanks for the feedback. Hopefully we will get the fix at quirk level (I've read that the issue isn't happening while using MQTT). Meanwhile, I'm following [...]

You don't have to make groups - all you have to do is divide commands in automations - so that it sends max 4-5 commands, then 2-3 seconds breaks and next commands. The second issue is in my case is that roller shutters devices have a poor range, so I had to add signal repeaters (Ikea has cheap outlets that works as repeaters).

dabeschte, 2025-04-11:
@mkz212's solution fixed it  90% of the time for me, but not always.
I bought a second zigbee adapter for z2m and can confirm that the position is always correct with that. My sonoff humidity sensor is also more stable with it. But z2h has worked well for me for everything so far except these 2 devices

mkz212, 2025-04-11:
> [@mkz212](https://github.com/mkz212)'s solution fixed it 90% of the time for me, but not always. I bought a second zigbee adapter for z2m and can confirm that the position is always correct with [...]

I know that my automation for repair will not work if the device has a poor range or if the zigbee network is busy (too many commands at once). I know that there is no queue mechanism in zha - if there are too many commands at the same time, some do not work. As far as I know, z2m has a queue mechanism, so there is no such problem.