zigpy/zha-device-handlers #3603: [Device Support Request] Tuya Thermostatic Radiator Valve (_TZE204_qyr2m29i)

39Northover, 2024-12-10:
### Problem description

Tuya Thermostatic Radiator Valve (_TZE204_qyr2m29i) is not working correctly in Home Assistant through ZHA. Despite creating a custom quirk (rookie), the device does not update attributes like current_temperature, target_temperature, or battery_level reliably or show entities. Logs show issues with duplicate TSNs, incomplete or unexpected data formats, and unhandled commands, leading to inconsistent behavior and updates in Home Assistant.

### Solution description

Update Home Assistant attributes correctly and reliably:

- Display accurate current_temperature.
- Allow setting and showing target_temperature.
- Reflect the current battery_level.

Handle device-specific commands properly:

- Interpret and process set_data_response commands to update attributes.
- Handle set_time_request commands 

Goal:
A working quirk or ZHA integration that enables a Tuya Thermostatic Radiator Valve to:

Fully integrate with Home Assistant.
Provide accurate and responsive controls and feedback for thermostat functions.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_qyr2m29i",
  "model": "TS0601",
[... 792 lines cut ...]
                "device_type": zha.DeviceType.THERMOSTAT,
                "input_clusters": [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaThermostaticCluster,  # Use the custom thermostat-specific cluster
                ],
                "output_clusters": [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    def update_entity(self, attr_name, value):
        """Update Home Assistant entity."""
        self.debug(f"Updating entity {attr_name} with value {value}")
        if hasattr(self, attr_name):
            setattr(self, attr_name, value)
            # Signal Home Assistant to refresh the entity
            self.hass.bus.fire("zha_custom_update", {"entity": attr_name, "value": value})]
```

</details>

prairiesnpr, 2024-12-10:
Looks like z2m has identified the Datapoints: https://github.com/Koenkk/zigbee2mqtt/issues/24958

I'd recommend building a v2 quirk, docs are here https://github.com/zigpy/zha-device-handlers/blob/dev/tuya.md and there is a guide on the wiki now. https://github.com/zigpy/zha-device-handlers/wiki/Tuya-%E2%80%90-v2-Quirk-with-TuyaQuirkBuilder

This should get you started. See the quirk in work here https://github.com/zigpy/zha-device-handlers/pull/3565

```python
from zigpy.quirks.v2 import EntityType
import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant import UnitOfTemperature,,
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass

(
    TuyaQuirkBuilder("_TZE204_qyr2m29i", "TS0601")
   .tuya_number(
        dp_id=4,
        type=t.int16s,
        attribute_name="current_heating_setpoint",
        min_value=5,
        max_value=35,
        step=1,
        unit=UnitOfTemperature.CELSIUS,
        translation_key="heating_setpoint",
        fallback_name="Heating setpoint",
    )
    .tuya_sensor(
        dp_id=5,
        attribute_name="local_temperature",
        type=t.int16s,
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfTemperature.CELSIUS,
        translation_key="local_temperature",
        fallback_name="Local temperature",
    )
    .skip_configuration()
    .add_to_registry()
)
```

BugSniper3000, 2024-12-11:
Thanks for the help @prairiesnpr , rather than battle on my own I have used Z2M and have got to the same point as the others with it partially working. Hopefully there is come more progress in the future :)

Bastix87, 2024-12-18:
@prairiesnpr unfortunately my devices will not use the quirk. 
I´v corrected the two "," here:
`from zigpy.quirks.v2.homeassistant import UnitOfTemperature,,` 

The quirk seems to be loaded but by adding the devices it will stuck on "configure":
[image]

Do you have any clue for me ?

Thank you

prairiesnpr, 2024-12-18:
Try something like this at the start of the quirk.

```python
from zigpy.quirks import DEVICE_REGISTRY

del DEVICE_REGISTRY._registry_v2[("_TZE204_qyr2m29i", "TS0601")]
```

Bastix87, 2024-12-18:
Then I get the error in Logs:
`Unexpected exception importing custom quirk 'TS0601_TZE204_qyr2m29i'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/TS0601_TZE204_qyr2m29i.py", line 2, in <module>
    del DEVICE_REGISTRY._registry_v2[("_TZE204_qyr2m29i", "TS0601")]
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: ('_TZE204_qyr2m29i', 'TS0601')`

prairiesnpr, 2024-12-18:
Just so I understand, this is during a reload or a restart?

This is a zigpy bug documented here https://github.com/zigpy/zigpy/issues/1508

You can remove the Del statement, then make sure you don't have any other quirks for this device, then attempt a restart and see if ZHA loads.

Bastix87, 2024-12-18:
It´s the only custom quirk. I´v put the the del statement in the quirk and done a restart. After restart the error was in the log.
After remove the del statement and a another restart I get this:
[image]

Instead 16°C it shows 160. I can´t change the temperature in HA (will not send to trv) but if I change the temperature on trv it will be shown in HA.

prairiesnpr, 2024-12-18:
> Instead 16°C it shows 160. 

You don't keep your house at 160c?

We can fix that. 

```python

from zigpy.quirks.v2 import EntityType
import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant import UnitOfTemperature,
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass

(
    TuyaQuirkBuilder("_TZE204_qyr2m29i", "TS0601")
   .tuya_number(
        dp_id=4,
        type=t.int16s,
        attribute_name="current_heating_setpoint",
        min_value=5,
        max_value=35,
        step=1,
        multiplier=0.1,
        unit=UnitOfTemperature.CELSIUS,
        translation_key="heating_setpoint",
        fallback_name="Heating setpoint",
    )
    .tuya_sensor(
        dp_id=5,
        attribute_name="local_temperature",
        type=t.int16s,
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfTemperature.CELSIUS,
        translation_key="local_temperature",
        fallback_name="Local temperature",
    )
    .skip_configuration()
    .add_to_registry()
)
```

Sanderrtjee, 2025-01-24:
I got the same thermostat valves but with the code from above I can't seem to get it working. 

What am I doing wrong? Am I missing something?

prairiesnpr, 2025-01-24:
> I got the same thermostat valves but with the code from above I can't seem to get it working.
> 
> What am I doing wrong? Am I missing something?

Post the signature of your device.

Sanderrtjee, 2025-01-24:
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_qyr2m29i",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}

I got the quirk working but still cant seem to see for example what the temperature is

prairiesnpr, 2025-01-24:
OK, couple things, the above was just a start of a quirk. I didn't go any further with it since the original requester left for z2m and nobody could test it. If you want to do some testing, we can probably put one together based on the last few thermostat and trv quirks I built, but it will take some iteration.

Do you have any entities with the current quirk? Are you just reloading ZHA, have you tried rebooting, and removing and rejoining the device?

ctepelea, 2025-02-03:
Hello guys and thank you for all your hard work. I am using the quirk provided, but I get a very strange behaviour for the device _TZE204_qyr2m29i. Whenever I want to change the thermostat temperature I immediately get insteat of the set temperature in ℃ the message "Heat/cool" and this message from ZHA: "Failed to perform the action climate/set_hvac_mode. expected HVACMode or one of 'off', 'heat', 'cool', 'heat_cool', 'auto', 'dry', 'fan_only' for dictionary value @ data['hvac_mode']"

[image]
This is that it looks like before attempting to change the temperature.
[image]

I suppose something is mapped wrongly.

prairiesnpr, 2025-02-03:
@ctepelea is this with h the quirk in #3769?

ctepelea, 2025-02-04:
Yes, specifically this part:

(
    base_tuya_trv.clone()
    .applies_to("_TZE204_qyr2m29i", "TS0601")
    .applies_to("_TZE204_ltwbm23f", "TS0601")
    # DPs 3, 4, 5, and 47 from base
    # preset mode and system mode are on the same datapoint, using only system mode
    # until we support multiple values on a single DP
    # .tuya_enum(
    #     dp_id=2,
    #     attribute_name="preset_mode",
    #     enum_class=TuyaPresetMode,
    #     translation_key="preset_mode",
    #     fallback_name="Preset mode",
    # )
    .tuya_dp(
        dp_id=2,
        ep_attribute=TuyaThermostatV2.ep_attribute,
        attribute_name=TuyaThermostatV2.AttributeDefs.system_mode.name,
        converter=lambda x: {
            TuyaPresetMode.Auto: Thermostat.SystemMode.Auto,
            TuyaPresetMode.Antifrost: Thermostat.SystemMode.Auto,
            TuyaPresetMode.Eco: Thermostat.SystemMode.Auto,
            TuyaPresetMode.Comfort: Thermostat.SystemMode.Auto,
            TuyaPresetMode.Heat: Thermostat.SystemMode.Heat,
            TuyaPresetMode.Off: Thermostat.SystemMode.Off,
        }[x],
        dp_converter=lambda x: {
            Thermostat.SystemMode.Auto: TuyaPresetMode.Auto,
            Thermostat.SystemMode.Heat: TuyaPresetMode.Heat,
            Thermostat.SystemMode.Off: TuyaPresetMode.Off,
        }[x],
    )
    .tuya_battery(dp_id=6)
    .tuya_switch(
        dp_id=7,
        attribute_name="child_lock",
        translation_key="child_lock",
        fallback_name="Child lock",
    )
    .tuya_number(
        dp_id=9,
        attribute_name="max_temperature",
        type=t.uint16_t,
        unit=UnitOfTemperature.CELSIUS,
        min_value=15,
        max_value=35,
        step=1,
        multiplier=0.1,
        translation_key="max_temperature",
        fallback_name="Max temperature",
    )
    .tuya_number(
        dp_id=10,
        attribute_name="min_temperature",
        type=t.uint16_t,
        unit=UnitOfTemperature.CELSIUS,
        min_value=1,
        max_value=15,
        step=1,
        multiplier=0.1,
        translation_key="min_temperature",
        fallback_name="Min temperature",
    )
    .tuya_switch(
        dp_id=14,
        attribute_name="window_detection",
        translation_key="window_detection",
        fallback_name="Open window detection",
    )
    .tuya_enum(
        dp_id=15,
        attribute_name="window_open",
        enum_class=TuyaWindowOpen,
        entity_platform=EntityPlatform.SENSOR,
        entity_type=EntityType.STANDARD,
        translation_key="window_open",
        fallback_name="Window open",
    )
    .tuya_enum(
        dp_id=110,
        attribute_name="motor_thrust",
        enum_class=TuyaMotorThrust,
        translation_key="motor_thrust",
        fallback_name="Motor thrust",
    )
    .tuya_enum(
        dp_id=111,
        attribute_name="display_brightness",
        enum_class=TuyaDisplayBrightness,
        translation_key="display_brightness",
        fallback_name="Display brightness",
    )
    .tuya_enum(
        dp_id=113,
        attribute_name="display_orientation",
        enum_class=TuyaDisplayOrientation,
        translation_key="display_orientation",
        fallback_name="Display orientation",
    )
    .tuya_sensor(
        dp_id=114,
        attribute_name="valve_position",
        type=t.int16s,
        divisor=10,
        state_class=SensorStateClass.MEASUREMENT,
        unit=PERCENTAGE,
        translation_key="valve_position",
        fallback_name="Valve position",
    )
    .tuya_number(
        dp_id=119,
        attribute_name="comfort_temperature",
        type=t.uint16_t,
        unit=UnitOfTemperature.CELSIUS,
        min_value=5,
        max_value=30,
        step=1,
        multiplier=0.1,
        translation_key="comfort_temperature",
        fallback_name="Comfort temperature",
    )
    .tuya_number(
        dp_id=120,
        attribute_name="eco_temperature",
        type=t.uint16_t,
        unit=UnitOfTemperature.CELSIUS,
        min_value=5,
        max_value=30,
        step=1,
        multiplier=0.1,
        translation_key="eco_temperature",
        fallback_name="Eco temperature",
    )
    .tuya_number(
        dp_id=121,
        attribute_name="holiday_temperature",
        type=t.uint16_t,
        unit=UnitOfTemperature.CELSIUS,
        min_value=5,
        max_value=30,
        step=1,
        multiplier=0.1,
        translation_key="holiday_temperature",
        fallback_name="Holiday temperature",
    )
    .tuya_enum(
        dp_id=127,
        attribute_name="hysteresis_mode",
        enum_class=TuyaDisplayOrientation,
        translation_key="hysteresis_mode",
        fallback_name="Hysteresis mode",
    )
    .add_to_registry()
)

closed 2025-02-26 by TheJulianJES

Bert3660, 2025-02-26:
Did someone complete the quirk for this model? thanks in advance

TheJulianJES, 2025-02-26:
Should work out-of-the-box if you upgrade to Home Assistant Core 2025.3.0b0 or later. Beta is already out, full release comes out in 7 days.

PR #3769 (2025-02-26): Add Moes (Tuya) TRV602Z and TRV801Z TRVs
## Proposed change

Adds Moes _TZE204_qyr2m29i and _TZE204_ltwbm23f

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/79e4c8560f96e6ba52a94370137d420e2a9641b8/src/devices/tuya.ts#L5169

## Additional information

DP 2 is used for system mode and preset mode. Currently this quirk ignores preset mode and maps those values to a system mode value. #3643 may address, but I haven't had time to see how that would work when setting attributes.

Closes: #3603
Closes: #3620
Closes: #3639
Closes: #3609
Closes: #3578

## Checklist

- [ ] The chan [... cut]