zigpy/zha-device-handlers #3604: [BUG] TS004F device incorrect quirk?

vknmakes, 2024-12-11:
### Bug description

I've got a TS004F by _TZ3000_402vrq2i that has Quirk: zhaquirks.tuya.ts004f.TuyaSmartRemote004F loaded.  I believe this quirk is for a multi-button dimmer switch.  My device is actually a rotary encoder knob that appears to have at least 2 button events: long/short press.  Using the device now does capture ccw/cw rotation, but, I am unable to create any automations with the device as none of the triggers seem to be the actual triggers the device sends to HA.

Go easy on my, I only learned about quirks a couple hours ago :).

### Steps to reproduce

Pair rotary encoder knob dimmer which identifies as TS004F 

### Expected behavior

I think it should just have a few device trigger options that match the actions of the device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```
"node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2024-12-11:
Without quirk all devices is only working in light controller mode.
Then making the original quirk i also was leaving one not with model so its getting some functionality.
You device need being added in the right device class with its manufacture ID and then it shall working OK (likely in this class) https://github.com/zigpy/zha-device-handlers/blob/6d16517b9ddf3a306dedf56d19da90740077a1ba/zhaquirks/tuya/ts004f.py#L68

vknmakes, 2024-12-11:
How can I help, is any additional info required?  I will play with this a bit myself.  I an HA n00b, but not worried about python.

MattWestb, 2024-12-11:
By adding the model info in one new line here:
https://github.com/zigpy/zha-device-handlers/blob/6d16517b9ddf3a306dedf56d19da90740077a1ba/zhaquirks/tuya/ts004f.py#L74-L80
The problem is that the quirk is living in one docker container and its not so easy editing and is going away if updating HA.
Little old information and also how to installing it as local quirk:https://github.com/zigpy/zha-device-handlers/discussions/693

trsdn, 2024-12-28:
same issue here.

I see that the events are triggered:
_TZ3000_abrsvsou TS004F Rotate Type event was fired with parameters: {'rotate_type': 0}
00:33:37 - Vor 3 Minuten
_TZ3000_abrsvsou TS004F Device Rotated - Right event was fired
00:33:37 - Vor 3 Minuten

but automation is not triggered via: "right" device_rotated_slow

vknmakes, 2025-01-02:
I think this is just going to be an ongoing thing since it appears these devices are proliferating :). 
Editing the quirk is relatively easy:

[me@home]$ docker exec -it homeassistant sh
/config # vi /usr/local/lib/python3.13/site-packages/zhaquirks/tuya/ts004f.py 

And add as indicated by MattWestb above.

closed 2025-08-02 by github-actions[bot]