zigpy/zha-device-handlers #3606: [Device Support Request]  Tuya TS0601 TZE204_dhotiauw

Theseku, 2024-12-12:
### Problem description

Link
https://www.aliexpress.com/item/1005007499685162.html

Here are the datapointids from Tuya:

19 | Power total
20 | Voltage
101 | Current
102 | Current
103 | Power
104 | Power
105 | Supply frequency
107 | Reverse energy total
110 | Forward energy Total
115 | Forward energy
116 | Reverse energy
117 | Forward energy
118 | Reverse energy
120 | Power factor
121 | Power factor
122 | Report frequency
123 | Voltage calibration
124 | AC frequency calibretion
125 | Current  calibretion
126 | Power  calibretion
127 | Forword energy  calibretion
128 | Reverse energy  calibretion
130 | Current calibration
131 | Power calibration
132 | Forword energy calibretion
133 | Reverse energy calibretion
140 | Chanel set
141 | Current A reverse
142 | Current B reverse
143 | Alarm overcurrent
144 | Alarm overcurrent
145 | Overcurrent alarm
146 | Overcurrent alarm
147 | A电流过流提醒
148 | B电流过流提醒

External definition
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const utils = require('zigbee-herdsman-converters/lib/utils');
const logger = require('zigbee-herdsman-converters/lib/logger');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_dhotiauw',
        },
    ],
    model: 'TS0601_new',
    vendor: 'Tuya',
    description: 'Bidirectional energy meter with 80A current clamp',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.ac_frequency(),
        e.voltage(),
        tuya.exposes.powerWithPhase('a'),
        tuya.exposes.powerWithPhase('b'),
        tuya.exposes.powerWithPhase('ab'),
        tuya.exposes.currentWithPhase('a'),
        tuya.exposes.currentWithPhase('b'),
        tuya.exposes.powerFactorWithPhase('a'),
        tuya.exposes.powerFactorWithPhase('b'),
        tuya.exposes.energyFlowWithPhase('a', ['sign']),
        tuya.exposes.energyFlowWithPhase('b', ['sign']),
        tuya.exposes.energyWithPhase('a'),
        tuya.exposes.energyWithPhase('b'),
        tuya.exposes.energyProducedWithPhase('a'),
        tuya.exposes.energyProducedWithPhase('b'),
        e
            .numeric('update_frequency', ea.STATE_SET)
            .withUnit('s')
            .withDescription('Update frequency')
            .withValueMin(3)
            .withValueMax(60)
            .withPreset('default', 10, 'Default value'),
        // Timestamp a and b are basically equivalent to last_seen
        // but they indicate when the unsigned value of power_a and power_b
        // were received. They can be several seconds in the past if
        // the publication was delayed because of the late_energy_flow options.
        e.numeric('timestamp_a', ea.STATE).withDescription('Timestamp at power measure (phase a)'),
        e.numeric('timestamp_b', ea.STATE).withDescription('Timestamp at power measure (phase b)'),
    ],
    meta: {
        multiEndpointSkip: ['power_factor', 'power_factor_phase_b', 'power_factor_phase_c', 'energy'],
        tuyaDatapoints: [
            [105, 'ac_frequency', tuya.valueConverter.divideBy100],
            [103, 'power_a', tuya.valueConverter.divideBy10],
            [104, 'power_b', tuya.valueConverter.divideBy10],
            [19, 'power', tuya.valueConverter.divideBy10],
            [20, 'voltage', tuya.valueConverter.divideBy10],
            [101, 'current_a', tuya.valueConverter.divideBy1000],
            [102, 'current_b', tuya.valueConverter.divideBy1000],
            [120, 'power_factor_a', tuya.valueConverter.raw],
            [121, 'power_factor_b', tuya.valueConverter.raw],
            [110, 'energy', tuya.valueConverter.divideBy100],
            [115, 'energy_a', tuya.valueConverter.divideBy100],
            [117, 'energy_b', tuya.valueConverter.divideBy100],
            [116, 'energy_produced_a', tuya.valueConverter.divideBy100],
            [118, 'energy_produced_b', tuya.valueConverter.divideBy100],
            [107, 'energy_produced', tuya.valueConverter.divideBy100],
            [122, 'update_frequency', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

### Solution description

Support for ZHA

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

beatles1, 2024-12-31:
I'd also like to see this work. The quirk for a similar single monitor device doesn't appear to work (from https://smarthomescene.com/reviews/tuya-zigbee-single-clamp-energy-meter-review/) when tweaked to include this device.

MarinX, 2025-01-28:
I just modified from _TZE204_ugekduaj to _TZE204_dhotiauw and quirk gets loaded. Need to test with a proper load to see if I can get any data in.  

 https://github.com/zigpy/zha-device-handlers/blob/ab4e4f5e1b3184aa73caf8f4e7e87a1f4ad9576c/zhaquirks/tuya/ts0601_din_power.py#L255

[image]

Theseku, 2025-01-28:
I think, it doesn´t work good, I will try to change datapoints.

[image]

Tell you later

MarinX, 2025-02-16:
I managed to get power a,b phase and current a,b phase by using this configuration

```python
    .tuya_sensor(
        dp_id=103,
        attribute_name="power_consumed_ph_a",
        type=t.uint32_t,
        divisor=10,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfPower.WATT,
        translation_key="power_ph_a",
        fallback_name="Power phase A",
    )
    .tuya_sensor(
        dp_id=104,
        attribute_name="power_consumed_ph_b",
        type=t.uint32_t,
        divisor=10,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfPower.WATT,
        translation_key="power_ph_b",
        fallback_name="Power phase B",
    )
    .tuya_sensor(
        dp_id=101,
        attribute_name="current_ph_a",
        type=t.uint32_t,
        divisor=1000,
        device_class=SensorDeviceClass.CURRENT,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfElectricCurrent.AMPERE,
        translation_key="current_ph_a",
        fallback_name="Current phase A",
    )
    .tuya_sensor(
        dp_id=102,
        attribute_name="current_ph_b",
        type=t.uint32_t,
        divisor=1000,
        device_class=SensorDeviceClass.CURRENT,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfElectricCurrent.AMPERE,
        translation_key="current_ph_b",
        fallback_name="Current phase B",
    )
```

Theseku, 2025-02-21:
> I managed to get power a,b phase and current a,b phase by using this configuration
> 
>     .tuya_sensor(
>         dp_id=103,
>         attribute_name="power_consumed_ph_a",
>         type=t.uint32_t,
>         divisor=10,
>         device_class=SensorDeviceClass.POWER,
>         state_class=SensorStateClass.MEASUREMENT,
>         unit=UnitOfPower.WATT,
>         translation_key="power_ph_a",
>         fallback_name="Power phase A",
>     )
>     .tuya_sensor(
>         dp_id=104,
>         attribute_name="power_consumed_ph_b",
>         type=t.uint32_t,
>         divisor=10,
>         device_class=SensorDeviceClass.POWER,
>         state_class=SensorStateClass.MEASUREMENT,
>         unit=UnitOfPower.WATT,
>         translation_key="power_ph_b",
>         fallback_name="Power phase B",
>     )
>     .tuya_sensor(
>         dp_id=101,
>         attribute_name="current_ph_a",
>         type=t.uint32_t,
>         divisor=1000,
>         device_class=SensorDeviceClass.CURRENT,
>         state_class=SensorStateClass.MEASUREMENT,
>         unit=UnitOfElectricCurrent.AMPERE,
>         translation_key="current_ph_a",
>         fallback_name="Current phase A",
>     )
>     .tuya_sensor(
>         dp_id=102,
>         attribute_name="current_ph_b",
>         type=t.uint32_t,
>         divisor=1000,
>         device_class=SensorDeviceClass.CURRENT,
>         state_class=SensorStateClass.MEASUREMENT,
>         unit=UnitOfElectricCurrent.AMPERE,
>         translation_key="current_ph_b",
>         fallback_name="Current phase B",
>     )

Can you show us, full file configuratión?

I gert errors with my file config

Regards

MarinX, 2025-02-22:
sure, here is the full reference file

```python

"""Tuya Din Power Meter."""

from zigpy.profiles import zha
from zigpy.quirks.v2 import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import PERCENTAGE, UnitOfEnergy, UnitOfElectricCurrent, UnitOfPower
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaLocalCluster,
    TuyaManufClusterAttributes,
    TuyaOnOff,
    TuyaSwitch,
)
from zhaquirks.tuya.builder import TuyaQuirkBuilder

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

"""Hiking Power Meter Attributes"""
HIKING_DIN_SWITCH_ATTR = 0x0110
HIKING_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
HIKING_TOTAL_ENERGY_RECEIVED_ATTR = 0x0266
HIKING_VOLTAGE_CURRENT_ATTR = 0x0006
HIKING_POWER_ATTR = 0x0267
HIKING_FREQUENCY_ATTR = 0x0269
HIKING_POWER_FACTOR_ATTR = 0x026F
HIKING_TOTAL_REACTIVE_ATTR = 0x026D
HIKING_REACTIVE_POWER_ATTR = 0x026E

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_deliver_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value / 10)
        elif attrid == TUYA_VOLTAGE_ATTR:
[... 291 lines cut ...]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    HikingManufClusterDinPower,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

cedricdev, 2025-11-15:
#Hello,
This one works for me.

[ts0601_TZE204_dhotiauw_din_power.py](https://github.com/user-attachments/files/23560775/ts0601_TZE204_dhotiauw_din_power.py)

[image]