zigpy/zha-device-handlers #3609: [Device Support Request]

Smandurlo, 2024-12-13:
### Problem description

Hello,
I bought some Tuya TRV _TZE204_ltwbm23f they are able to pair with ZHA, but no entities. The Tuya model is TRV601

Is it possible for someone to create the quirk? I am not really able to :-(

Thank you!

### Solution description

To support the device :-)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_ltwbm23f",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[... 606 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

tco99ttocs, 2024-12-13:
should be the same like [https://github.com/zigpy/zha-device-handlers/issues/3578](https://github.com/zigpy/zha-device-handlers/issues/3578)

Smandurlo, 2024-12-13:
> should be the same like #3578

it looks very different. Mine is https://www.aliexpress.com/item/1005005584973831.html

bangert, 2024-12-28:
even though it looks different, the exposed zigbee endpoints (and the internal electronics) may be the same.

PR #3769 (2025-02-26): Add Moes (Tuya) TRV602Z and TRV801Z TRVs
## Proposed change

Adds Moes _TZE204_qyr2m29i and _TZE204_ltwbm23f

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/79e4c8560f96e6ba52a94370137d420e2a9641b8/src/devices/tuya.ts#L5169

## Additional information

DP 2 is used for system mode and preset mode. Currently this quirk ignores preset mode and maps those values to a system mode value. #3643 may address, but I haven't had time to see how that would work when setting attributes.

Closes: #3603
Closes: #3620
Closes: #3639
Closes: #3609
Closes: #3578

## Checklist

- [ ] The chan [... cut]
closed 2025-02-26 by TheJulianJES