zigpy/zha-device-handlers #3610: [Device Support Request] TUYA Smoke _TZE284_rccxox8p

netdrg, 2024-12-13:
### Problem description

Hello,
I bought some Tuya TS601 _TZE284_rccxox8p they are able to pair with ZHA, but no entities just firmware. The Tuya model is same TS601 _TZE200_rccxox8p.
Is it possible for someone to create the quirk? I am not really able to :-(

Thank you!

### Solution description

To support the device :-)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_rccxox8p",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

baldisos, 2024-12-16:
Same problem here.

Some further Informations:

```
Nwk: 0xa9ce
```
```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_rccxox8p",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

prairiesnpr, 2024-12-16:
This is likely the same as the other Tuya smoke detectors. They all seem to use DP 1, if it's functional we can update the existing quirk.

```python
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zhaquirks.tuya import TuyaLocalCluster
from zigpy.zcl.clusters.security import IasZone

class TuyaIasFire(IasZone, TuyaLocalCluster):
    """Tuya local IAS smoke/fire cluster."""

    _CONSTANT_ATTRIBUTES = {
        IasZone.AttributeDefs.zone_type.id: IasZone.ZoneType.Fire_Sensor
    }

(
    TuyaQuirkBuilder("_TZE284_rccxox8p", "TS0601")
    .tuya_dp(
        dp_id=1,
        TuyaIasFire.ep_attribute,
        IasZone.AttributeDefs.zone_status.name,
        converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x == 0 else 0,
    )
    .adds(TuyaIasFire)
    .skip_configuration()
    .add_to_registry()
)
```

baldisos, 2024-12-27:
Can you give me a hint on how to test if this works?

prairiesnpr, 2024-12-27:
> Can you give me a hint on how to test if this works?

You need to enable custom quirks, see the ZHA documentation. Once enabled, place that code snippet in a file located in the directory configuted as the custom quirk directory, name it `tuya_TZE284_rccxox8p.py` then restart HA. I think there are some forum posts that have more detail on setting up custom quirks also.

baldisos, 2024-12-27:
Okay, i tried that. But i still don't see any Alarm Entitiy.
Is there a way to see if that Quirk was applied?

I can only see this in the logs:
```
[0xEB85:1:0x0000] Received ZCL frame: b'\x18\x06\x01\x04\x00\x00B\x10_TZE284_rccxox8p\x05\x00\x00B\x06TS0601'
[0xEB85:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=6, command_id=1, *direction=<Direction.Server_to_Client: 1>)
[0xEB85:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=4, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE284_rccxox8p')), ReadAttributeRecord(attrid=5, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0xeb85] Read model 'TS0601' and manufacturer '_TZE284_rccxox8p' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000, None:0xED00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xeb85] Discovered basic device information for <Device model='TS0601' manuf='_TZE284_rccxox8p' nwk=0xEB85 ieee=a4:c1:38:f0:2a:64:19:b5 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE284_rccxox8p' nwk=0xEB85 ieee=a4:c1:38:f0:2a:64:19:b5 is_initialized=True>
Emitting event raw_device_initialized with data RawDeviceInitializedEvent(device_info=RawDeviceInitializedDeviceInfo(ieee=a4:c1:38:f0:2a:64:19:b5, nwk=0xEB85, pairing_status=<DevicePairingStatus.INTERVIEW_COMPLETE: 2>, model='TS0601', manufacturer='_TZE284_rccxox8p', signature={'manufacturer': '_TZE284_rccxox8p', 'model': 'TS0601', 'node_desc': {'logical_type': <LogicalType.EndDevice: 2>, 'complex_descriptor_available': 0, 'user_descriptor_available': 0, 'reserved': 0, 'aps_flags': 0, 'frequency_band': <FrequencyBand.Freq2400MHz: 8>, 'mac_capability_flags': <MACCapabilityFlags.AllocateAddress: 128>, 'manufacturer_code': 4417, 'maximum_buffer_size': 66, 'maximum_incoming_transfer_size': 66, 'server_mask': 10752, 'maximum_outgoing_transfer_size': 66, 'descriptor_capability_field': <DescriptorCapability.NONE: 0>}, 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.SMART_PLUG: 81>, 'input_clusters': [4, 5, 61184, 0, 60672], 'output_clusters': [25, 10]}}}), event_type='zha_gateway_message', event='raw_device_initialized') (1 listeners)
(ZHAGatewayProxy) handling event protocol for event: RawDeviceInitializedEvent(device_info=RawDeviceInitializedDeviceInfo(ieee=a4:c1:38:f0:2a:64:19:b5, nwk=0xEB85, pairing_status=<DevicePairingStatus.INTERVIEW_COMPLETE: 2>, model='TS0601', manufacturer='_TZE284_rccxox8p', signature={'manufacturer': '_TZE284_rccxox8p', 'model': 'TS0601', 'node_desc': {'logical_type': <LogicalType.EndDevice: 2>, 'complex_descriptor_available': 0, 'user_descriptor_available': 0, 'reserved': 0, 'aps_flags': 0, 'frequency_band': <FrequencyBand.Freq2400MHz: 8>, 'mac_capability_flags': <MACCapabilityFlags.AllocateAddress: 128>, 'manufacturer_code': 4417, 'maximum_buffer_size': 66, 'maximum_incoming_transfer_size': 66, 'server_mask': 10752, 'maximum_outgoing_transfer_size': 66, 'descriptor_capability_field': <DescriptorCapability.NONE: 0>}, 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.SMART_PLUG: 81>, 'input_clusters': [4, 5, 61184, 0, 60672], 'output_clusters': [25, 10]}}}), event_type='zha_gateway_message', event='raw_device_initialized')
Checking quirks for _TZE284_rccxox8p TS0601 (a4:c1:38:f0:2a:64:19:b5)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Creating cluster handler for cluster id: 4 class: <class 'zha.zigbee.cluster_handlers.general.GroupsClusterHandler'>
Creating cluster handler for cluster id: 5 class: <class 'zha.zigbee.cluster_handlers.general.ScenesClusterHandler'>
Creating cluster handler for cluster id: 61184 class: <class 'zha.zigbee.cluster_handlers.manufacturerspecific.TuyaClusterHandler'>
Creating cluster handler for cluster id: 0 class: <class 'zha.zigbee.cluster_handlers.general.BasicClusterHandler'>
Creating cluster handler for cluster id: 60672 class: <class 'zha.zigbee.cluster_handlers.ClusterHandler'>
Discovering entities for endpoint: a4:c1:38:f0:2a:64:19:b5-1
'sensor' platform -> 'RSSISensor' using ['basic']
'sensor' platform -> 'LQISensor' using ['basic']
'update' platform -> 'FirmwareUpdateEntity' using ['ota']
Discovering entities for device: a4:c1:38:f0:2a:64:19:b5-_TZE284_rccxox8p TS0601
Attempting to discover quirks v2 entities for device: a4:c1:38:f0:2a:64:19:b5-_TZE284_rccxox8p TS0601
Device: a4:c1:38:f0:2a:64:19:b5-_TZE284_rccxox8p TS0601 is not a quirks v2 device - skipping discover_quirks_v2_entities
device - 0xEB85:a4:c1:38:f0:2a:64:19:b5 entering async_device_initialized - is_new_join: True
device - 0xEB85:a4:c1:38:f0:2a:64:19:b5 has joined the ZHA zigbee network
[0xEB85](TS0601): Update device availability -  device available: True - new availability: True - changed: False

```

prairiesnpr, 2024-12-27:
Look at device info, should see the quirk applied. Should look like this.
[image]

baldisos, 2024-12-27:
Nope, nothing there.
[image]

Just to confirm i did everything right, i put this in configuration.yaml:
```
zha:
  database_path: /config/zigbee.db 
  enable_quirks: true
  custom_quirks_path: /config/zha_custom_quirks/
```

Then in folder /config/zha_custom_quirks i put the file tuya_TZE284_rccxox8p.py with your content.

Looks right to me. I don't see why it's not loading the Quirk tho.

prairiesnpr, 2024-12-27:
So that looks right to me, fair disclosure, I've never actually used custom quirks before, nor set them up, so bit of the blind leading the blind. You might try removing the trailing forward slash in the path.

baldisos, 2024-12-27:
Nope, still nothing. Maybe it's something with the Quirk itself. I don't have the knowledge to go from here, so i just wait for more instructions in case anyone has a clue on this. 😄

prairiesnpr, 2024-12-27:
Just to confirm, you are on the latest HA version?

baldisos, 2024-12-27:
Yes.

I just got it to work with the help of Github Copilot.

The Quirk looks like this now:
```
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zhaquirks.tuya import TuyaLocalCluster
from zigpy.zcl.clusters.security import IasZone

class TuyaIasFire(IasZone, TuyaLocalCluster):
    """Tuya local IAS smoke/fire cluster."""

    _CONSTANT_ATTRIBUTES = {
        IasZone.AttributeDefs.zone_type.id: IasZone.ZoneType.Fire_Sensor
    }

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self.endpoint.device.ias_zone = self

# Ensure TuyaQuirkBuilder is correctly configured
(
    TuyaQuirkBuilder("_TZE284_rccxox8p", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaIasFire.ep_attribute,
        attribute_name=IasZone.AttributeDefs.zone_status.name,
        converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x == 0 else 0,
    )
    .adds(TuyaIasFire)
    .skip_configuration()
    .add_to_registry()
)
```

And confirmation can be seen in this screenshot:
[image]

The history showing smoke detect events and back to normal:
[image]

prairiesnpr, 2024-12-27:
I was going to go add this, but it was already added in https://github.com/zigpy/zha-device-handlers/pull/3596. It should work in the beta.

baldisos, 2024-12-27:
Looking through the PR i can't see the "_TZE**284**_rccxox8p", there's only

 .applies_to("_TZE**200**_rccxox8p", "TS0601")
_(Notice the 284, which is mine vs the 200 which is in the PR)_

Would this need adaptation?

prairiesnpr, 2024-12-27:
> Looking through the PR i can't see the "_TZE**284**_rccxox8p", there's only
> 
> .applies_to("_TZE**200**_rccxox8p", "TS0601") _(Notice the 284, which is mine vs the 200 which is in the PR)_
> 
> Would this need adaptation?

Whoops, missed that, hang on, new PR incoming.

baldisos, 2024-12-27:
Thank you! That was a really nice experience contributing to this and working together with you!

closed 2024-12-27 by TheJulianJES

kodacy, 2025-01-03:
Can you please confirm what would be the right ts0601_smoke.py Quirk for TS601 _TZE200_rccxox8p that can report fire and battery state? tested a bunch with no success. Thank you.

prairiesnpr, 2025-01-03:
It will be part of the next HA release, 2025.1.

cosmico89, 2025-01-22:
I don't understand, I have HA 2025.1.2 and zigbee2MQTT 2.0.0
I pair the smoker, it detects it but says it is not compatible.
Do I have to do anything else? I think it was integrated
Thanks

prairiesnpr, 2025-01-22:
> zigbee2MQTT 2.0.0

You are in the wrong place

mattfox27, 2025-01-25:
I can't this to work in HA or at least report the battery

prairiesnpr, 2025-01-25:
> I can't this to work in HA or at least report the battery

What exactly can't you get to work? I'd suggest a new issue with more details on what you have and what you are trying to do. This is in HA now, so you shouldn't need to do anything other than delete any custom quirks you may have.

dcigir, 2025-03-11:
I just installed this smoke detector and is showing detection in Home Assistant, but it doesn't show battery state. Can somebody let me know if this can be fixed and how?
Thanks

plasma99new, 2025-03-27:
Hi,

in another post I found the information that this one shoul be identical to the PA-44Z is this one suported by ZHA?

The problem I have is, i do only get the smoke entity but no battery live,

yvoictra, 2025-08-11:
> Hi,
> 
> in another post I found the information that this one should be identical to the PA-44Z is this one suported by ZHA?
> 
> The problem I have is, i do only get the smoke entity but no battery live, 

Same issue here.

kgooris, 2025-12-02:
Also same issue, battery status is really usefull on these smoke detectors....