zigpy/zha-device-handlers #3613: [Device Support Request] TS0601 by _TZE200_nqo5ou9z

AdmiralKirk02, 2024-12-14:
### Problem description

Helloo
I bought a thermostat which I can add to HA via ZHA but without any entities.

Its called Essentials MATRIX Zigbee. 

Would anyone be so kind and add a quirk? :)

### Solution description

Is a custom Quirk the solution?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_nqo5ou9z",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MeatHunter95, 2024-12-16:
Hi there!

Check out [this post](https://github.com/zigpy/zha-device-handlers/issues/3416#issuecomment-2434427006); the TS0601 is identified as a thermostat there.

I added the quirk (for the Beok TGM50-ZigBee Thermostat) 1 hour ago and it works fine for me.

Cheers!

AdmiralKirk02, 2024-12-17:
Did not work sadly :( 

Dunno if it matters, but the manufacturer differs from the given quirk.

debritde, 2025-01-17:
hi, did you find a working quirk ?
i also have the same thermostate :)

debritde, 2025-01-17:
@AdmiralKirk02