zigpy/zha-device-handlers #3628: [Device Support Request] Tuya TRV TS0601 _TZE204_ogx8u5z6

moedude, 2024-12-21:
### Problem description

I got a new ["Tuya" TRV from China](https://www.banggood.com/Tuya-Smart-Zigbee3_0-Thermostat-Radiator-Valve-Temperature-Schedule-Setting-Window-Open-Detection-Function-APP-Remote-Voice-Control-Work-with-Alexa-Google-Assistance-p-2020881.html?P=XV070160825191202106&cur_warehouse=CN), which isn’t supported by the current device handlers.
[image]

### Solution description

Maybe this is just another clone of existing hardware.
I tried adding a custom quirk (a copy of TS0601_trv.py) in HA but didn’t succeed. Either the signature is incorrect, or my quirk isn’t being loaded.
[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_ogx8u5z6",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 626 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

prairiesnpr, 2024-12-21:
Please try this custom quirk

https://github.com/prairiesnpr/zha-device-handlers/blob/add-Tuya-_TZE204_ogx8u5z6/zhaquirks/tuya/ts0601_trv_v2.py

T3pco, 2024-12-22:
> Please try this custom quirk
> 
> https://github.com/prairiesnpr/zha-device-handlers/blob/add-Tuya-_TZE204_ogx8u5z6/zhaquirks/tuya/ts0601_trv_v2.py

Tested (I have the same device, seems like a new version of common zigbee thermostats).
Looks much better, after re-adding the device thermostat entity appeared.

There seems to be an issue with HVAC mode. When on "heat", HA reacts to the manually set target temp on the device (read).
Writing heating target values has no result.

"Failed to perform the action climate/set_hvac_mode. expected HVACMode or one of 'off', 'heat', 'cool', 'heat_cool', 'auto', 'dry', 'fan_only' for dictionary value @ data['hvac_mode']." occurs when pressing the physical device button. Ith

I think mode "auto" is not working correctly.
local_temperature_calibration is missing, which is quite important and exists at least from the readings.

Z2M has the follwing (reported working) code (maybe it helps?)
https://github.com/Koenkk/zigbee2mqtt/issues/25199#issuecomment-2552109312

[image]
[image]

moedude, 2024-12-22:
Hello [prairiesnpr](https://github.com/prairiesnpr) and [T3pco](https://github.com/T3pco),
I can confirm the quirk (signature) matches, but the heating mode does not respond; however, it does reflect the set temperature on the device.
Child lock works! :D
I'm not entirely sure, but I think the "HVAC action" is also incorrect—it shows "idle" when heating is active and vice versa.
[image]

Anyway, I’d like to express my heartfelt thanks for all your efforts!

prairiesnpr, 2024-12-22:
I inverted the running state and corrected an error in the system mode. Please test the new version. On the set point failure, please put ZHA in debug logging and post the logs when attempting to change the set point.

prairiesnpr, 2024-12-22:
> local_temperature_calibration is missing, which is quite important and exists at least from the readings.

This is exposed in the quirk. Does the climate entity have UI elements for it normally? We can expose a number; I just assumed the HA climate entity would take care of it. Does it at least show as an attribute?

T3pco, 2024-12-22:
> I inverted the running state and corrected an error in the system mode. Please test the new version. On the set point failure, please put ZHA in debug logging and post the logs when attempting to change the set point.

[debug_set_temp_heat_mode.txt](https://github.com/user-attachments/files/18223138/debug_set_temp_heat_mode.txt)
[debug_switching_mode_with_button.txt](https://github.com/user-attachments/files/18223139/debug_switching_mode_with_button.txt)

Thanks for your effort :-)

HVAC action value now seems to be correct. When heating symbol is "on" at the display, it shows heating in HA. If it's not heating (set temp below current temp) status is "off".

HVAC mode is definitly wrong with the latest version of the quirk.
Freshly initialized device has HVAC status "Off". When pressing the device button to go to scheduled mode (clock icon) - there is an error. This should be the HVAC "auto" mode I guess.
When pressing the button again to go to normal heating mode - HA switches to HVAC mode "Off".

So "Off" should be "Heat" hvac mode.
And Auto is not working. Thermostat entity shows text "Heat/Cool".

Debug logs attached.

T3pco, 2024-12-22:
> This is exposed in the quirk. Does the climate entity have UI elements for it normally? We can expose a number; I just assumed the HA climate entity would take care of it. Does it at least show as [...]

No UI element for the climate entity. It should change the current temperature value in the climate entity.
It's the "local_temperature_calibration" attribute. I can read and set values here, but they seem to have no effect.

prairiesnpr, 2024-12-22:
I made a couple changes, the mode was absolutely wrong, cleaned those up. Also, updated `local_temperature_calibration` to match the z2m logic used in `localTempCalibration2`.

T3pco, 2024-12-23:
> I made a couple changes, the mode was absolutely wrong, cleaned those up. Also, updated `local_temperature_calibration` to match the z2m logic used in `localTempCalibration2`.

sorry for late reply. tested it. sad to say but it still does not work. Readded multiple devices to verifiy that.

Setting temperatures in heat hvac mode has no action on the device itself. Value reading (when using the knob) works like in the beginning.
Auto mode still not existent (and error shows like mentioned above) - but thats not relevant.

local_temperature_calibration is a bit weird. When I adjusted the quirk file and readded the device, I saw a slider to adjust under "Configuration". After HA restart and re-adding again, local_temperature_calibration do not appear at all. I could not make it appearing again.

Hope that helps. Thank you again :)

prairiesnpr, 2024-12-23:
I'm a little stumped on setting the temp, I see the traffic in the logs, correct DP, or it matches z2m anyways and correct format, thought maybe it was due to the sytem mode mismatch. At this point about all I could do would be to identify the raw incoming packet for each system mode and build a quick test to verify the logic. 

On the local calibration, no idea how you sea a slider, we aren't exposing one anywhere, I may go ahead and expose one for now so we can see if it's at least working.

T3pco, 2024-12-24:
> I'm a little stumped on setting the temp, I see the traffic in the logs, correct DP, or it matches z2m anyways and correct format, thought maybe it was due to the sytem mode mismatch. At this point about all I could do would be to identify the raw incoming packet for each system mode and build a quick test to verify the logic.
> 
> On the local calibration, no idea how you see a slider, we aren't exposing one anywhere, I may go ahead and expose one for now so we can see if it's at least working.

Thanks for your reply. I experimented with some values - I'm not sure if I screwed up my setup.
@moedude Can you please verify the last version? To double check as it does not behave for me like it should.

In parallel I try to reset my environment.

prairiesnpr, 2024-12-24:
I went ahead and added some tests based on your previous logs for `system_mode`, just running through the possible values and ensuring they convert to the correct ZCL values. Everything lines up with the z2m converter.

T3pco, 2024-12-30:
> I went ahead and added some tests based on your previous logs for `system_mode`, just running through the possible values and ensuring they convert to the correct ZCL values. Everything lines up with the z2m converter.

Tested a bit more now. Can eleminate local setup issues now.

Main issue is, that temperature change in HA entitiy do not lead to action on the device.

occupied_heating_setpoint is available and readable with currently set temperature on the device. You can even set a value (like 2800 for 28 degree). This value is then persisted and readable from the device.

When turning the knob, the set temp is shown in BA within seconds.

Still, the device do not adjust the actual temperature. Like its ignoring it. System mode is "heat".
Setting other values (like child lock) works immediately.

Idk whats causing this.

prairiesnpr, 2024-12-30:
I'll double check what z2m is doing, but we may need to get an actual capture of the traffic from a Tuya hub to the device and then from ZHA to compare.

prairiesnpr, 2024-12-31:
After reviewing, I think we are stuck, I don't see how we are doing anything different then what z2m is doing. 

The only thing I can suggest and I'm sure you've already tried is to completly remove the device and then add it back.

It would also be interesting to know if @moedude is having the same results.

Beyond that, we need to capture the trsffic when the command is sent from a Tuya hub or z2m to when it's sent by ZHA and see what the difference is.

T3pco, 2024-12-31:
> After reviewing, I think we are stuck, I don't see how we are doing anything different then what z2m is doing.
> 
> The only thing I can suggest and I'm sure you've already tried is to completly remove the device and then add it back.
> 
> It would also be interesting to know if @moedude is having the same results.
> 
> Beyond that, we need to capture the traffic when the command is sent from a Tuya hub or z2m to when it's sent by ZHA and see what the difference is.

Thanks for double checking. I asked for assistance in the z2m issue - maybe we can solve this together.
Re-adding etc. I did several times.
I agree we need more testers here :-)

prairiesnpr, 2024-12-31:
Thinking on this more, I noticed that the z2m converter marks the device as `configure: tuya.configureMagicPacket` Which is defined here, https://github.com/Koenkk/zigbee-herdsman-converters/blob/8f6e15f1c5f6464766982e13281cb0d3c53ae89d/src/lib/tuya.ts#L450

We have similar logic in https://github.com/zigpy/zha-device-handlers/blob/812441104427e74e13a8168ea916597f87534d72/zhaquirks/tuya/__init__.py#L532

My thought is, that this device requires the enchanted packet to be sent on join to allow `occupied_heating_setpoint` to be set and without it, it's ignoring the command. If it indeed works in z2m, that's the only difference I see. If so, we will need to determine a way to add that same logic in a v2 quirk. Since v2 quirks use CustomDeviceV2 https://github.com/zigpy/zigpy/blob/0dc50756fb44f7438c37709dc35cf29e597c3648/zigpy/quirks/v2/__init__.py#L65

I'm not certain how best to add the enchantment, have to think on it a bit.

prairiesnpr, 2024-12-31:
Ok, I reworked to add the enchantment on initial join. Please try the new quirk and let me know if that changes anything. You will need to repair the device.

T3pco, 2024-12-31:
> Ok, I reworked to add the enchantment on initial join. Please try the new quirk and let me know if that changes anything. You will need to remove and rejoin the device.

Ah thats what I also found but did not expect it to be a potential thing as the device pairing works fine.

Lets see tomorrow when I can do the next testing session.

Thanks for your motivation in this :-)

prairiesnpr, 2024-12-31:
> Thanks for your motivation in this :-)

All good, I was trying to get some high quality Tuya examples for each device type in the repo and made the mistake of thinking yours would be easy. Now I'm just invested.

moedude, 2025-01-01:
Hello everyone,

Please forgive me for the delay; I’ve been occupied with family matters. Unfortunately, the interim changes submitted haven’t yielded any success on my end either. The set temperature point is still not being adopted by the radiator valve.

The idea of using the MagicPacket or Spell sounds very promising!

To support further development, I’ve ordered the Tuya Zigbee Gateway. Once it arrives, I’ll be able to provide the necessary logs for analysis.

Thank you all for the effort and dedication you’ve shown so far—it’s truly appreciated!

prairiesnpr, 2025-01-01:
@moedude which version did you test? The latest one, updated yesterday, has the enchantment added, curious if that made any difference. You would have to remove and then rejoin the device since it's sent during the initial join.

t3pco, 2025-01-01:
I tested the new version, with cleanup and re-pairing. Sadly makes no difference.
I'm currently setting up Z2M to check proper functionality there.

moedude, 2025-01-02:
I’ve tested with the new quirk and ensured the device was cleanly removed beforehand.
Attached is my cleaned-up "Join" log.

Are the unknown cluster 0xEF00 or the two "Error handling '_save_attribute' event with..." errors of interest?

`Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 2, 0, 11, 16, 985628, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=3, profile_id=0, cluster_id=32822, data=Serialized[b'\x03\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
Feeding watchdog
[<Task pending name='global-updater_GlobalUpdater' coro=<periodic.<locals>.scheduler.<locals>.wrapper() running at /usr/local/lib/python3.13/site-packages/zha/decorators.py:78> cb=[set.remove()]>] executing periodic task [zha.application.helpers::GlobalUpdater.update_listeners]
Global updater interval starting
Global updater interval finished
[<Task pending name='device-availability-checker_DeviceAvailabilityChecker' coro=<periodic.<locals>.scheduler.<locals>.wrapper() running at /usr/local/lib/python3.13/site-packages/zha/decorators.py:78> cb=[set.remove()]>] executing periodic task [zha.application.helpers::DeviceAvailabilityChecker.check_device_availability]
Device availability checker interval starting
Checking device availability
[0x86CC](TS0505B): Device seen - marking the device available and resetting counter
[0x86CC](TS0505B): Update device availability -  device available: True - new availability: True - changed: False
[0x952B](SOMRIG shortcut button): Device seen - marking the device available and resetting counter
[0x952B](SOMRIG shortcut button): Update device availability -  device available: True - new availability: True - changed: False
[0xD161:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0xD161:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0x81EC:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0x81EC:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0xD8FB:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0xD8FB:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0xd8fb] Extending timeout for 0x01 request
[0x7D02:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0x7D02:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0x7d02] Extending timeout for 0x01 request
[0x3752](TS0601): Device seen - marking the device available and resetting counter
[0x3752](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x313F](TS0601): Device seen - marking the device available and resetting counter
[0x313F](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x81F7](TS0601): Device seen - marking the device available and resetting counter
[0x81F7](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x3543](TS0601): Device seen - marking the device available and resetting counter
[0x3543](TS0601): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 2, 0, 11, 35, 868515, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3C66), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=54, profile_id=0, cluster_id=19, data=Serialized[b'6f<2-,\xef\x118\xc1\xa4\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
New device 0x3c66 (a4:c1:38:11:ef:2c:2d:32) joined the network
[... cut ...]
[0x3C66:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=2, command_id=11, *direction=<Direction.Server_to_Client: 1>)
[0x3C66:1:0xef00] Decoded ZCL frame: TuyaReplacementCluster:Default_Response(command_id=36, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
[0x3C66:1:0xef00] Received command 0x0B (TSN 2): Default_Response(command_id=36, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
Previously delayed device request is now running, delayed by 0.11s
`

moedude, 2025-01-02:
I’ve tested with the new quirk and ensured the device was cleanly removed beforehand.
Attached is my cleaned-up "Join" log.

Are the unknown cluster 0xEF00 or the two "Error handling '_save_attribute' event with..." errors of interest?

Log:
<details>

```
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 2, 0, 11, 16, 985628, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=3, profile_id=0, cluster_id=32822, data=Serialized[b'\x03\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
Feeding watchdog
[<Task pending name='global-updater_GlobalUpdater' coro=<periodic.<locals>.scheduler.<locals>.wrapper() running at /usr/local/lib/python3.13/site-packages/zha/decorators.py:78> cb=[set.remove()]>] executing periodic task [zha.application.helpers::GlobalUpdater.update_listeners]
Global updater interval starting
Global updater interval finished
[<Task pending name='device-availability-checker_DeviceAvailabilityChecker' coro=<periodic.<locals>.scheduler.<locals>.wrapper() running at /usr/local/lib/python3.13/site-packages/zha/decorators.py:78> cb=[set.remove()]>] executing periodic task [zha.application.helpers::DeviceAvailabilityChecker.check_device_availability]
Device availability checker interval starting
Checking device availability
[0x86CC](TS0505B): Device seen - marking the device available and resetting counter
[0x86CC](TS0505B): Update device availability -  device available: True - new availability: True - changed: False
[0x952B](SOMRIG shortcut button): Device seen - marking the device available and resetting counter
[0x952B](SOMRIG shortcut button): Update device availability -  device available: True - new availability: True - changed: False
[0xD161:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0xD161:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0x81EC:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0x81EC:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0xD8FB:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0xD8FB:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0xd8fb] Extending timeout for 0x01 request
[0x7D02:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0x7D02:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0x7d02] Extending timeout for 0x01 request
[0x3752](TS0601): Device seen - marking the device available and resetting counter
[0x3752](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x313F](TS0601): Device seen - marking the device available and resetting counter
[0x313F](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x81F7](TS0601): Device seen - marking the device available and resetting counter
[0x81F7](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x3543](TS0601): Device seen - marking the device available and resetting counter
[0x3543](TS0601): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 2, 0, 11, 35, 868515, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3C66), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=54, profile_id=0, cluster_id=19, data=Serialized[b'6f<2-,\xef\x118\xc1\xa4\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
New device 0x3c66 (a4:c1:38:11:ef:2c:2d:32) joined the network
[... cut ...]
[0x3C66:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=2, command_id=11, *direction=<Direction.Server_to_Client: 1>)
[0x3C66:1:0xef00] Decoded ZCL frame: TuyaReplacementCluster:Default_Response(command_id=36, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
[0x3C66:1:0xef00] Received command 0x0B (TSN 2): Default_Response(command_id=36, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
Previously delayed device request is now running, delayed by 0.11s
```

</details>

pberenyi, 2025-01-03:
Hi, I have the same thermostat and I have been using it with this quirk so far: https://github.com/zigpy/zha-device-handlers/blob/94e15bdcdeeab9dc333fbd587416bffd1b76048f/zhaquirks/tuya/ts0601_trv_tze204.py
With this setting the temperature worked properly. Other functions (like battery state) don't necessarily work.

prairiesnpr, 2025-01-03:
> Hi, I have the same thermostat and I have been using it with this quirk so far

If possible, please enable debug logging and capture logs of you setting the temperature, would like to compare to the quirk. In worm here.

pberenyi, 2025-01-04:
These are the debug logs. I've increased the temperature to 28 °C from HA and then decreased it back to 22.5 °C from the device.
The device address is a4:c1:38:52:25:9c:1c:b2. Let me know if you need anything else.
[zb-trv-ogx8u5z6-home-assistant_zha_2025-01-04T09-43-46.908Z.zip](https://github.com/user-attachments/files/18306435/zb-trv-ogx8u5z6-home-assistant_zha_2025-01-04T09-43-46.908Z.zip)

prairiesnpr, 2025-01-04:
> These are the debug logs.

Thanks for this, the working quirk is sending it with `is_manufacturer_specific=False` and the non-working quirk is using `is_manufacturer_specific=True`

prairiesnpr, 2025-01-04:
I made a change to `write_attributes` on the new quirk, please capture logs again when modifying `occupied_heating_setpoint` from HA.

pberenyi, 2025-01-06:
I've tested with the latest version and the temperature setting seems to work fine in both directions. The device address I tested with is: a4:c1:38:53:33:d5:fc:5e
Here are the logs.
[zb-trv-ogx8u5z6-home-assistant_zha_2025-01-06T14.zip](https://github.com/user-attachments/files/18320041/zb-trv-ogx8u5z6-home-assistant_zha_2025-01-06T14.zip)

prairiesnpr, 2025-01-06:
Nice! Is there anything not working at this point, or should I clean it up and PR it?

prairiesnpr, 2025-01-06:
I didn't see a DP for battery percentage, if you have a Tuya hub you can follow the z2m guide for capturing DPs from their dev console. If there are any missing we can add those. It wasn't in z2m, so guessing it doesn't exist.

pberenyi, 2025-01-06:
I've added the TRV to the Smart Life app via the gateway, and other then the temperature calibration nothing relevant seems to be missing. (Battery percentage isn't present in the app either, and while ECO mode is missing, that can be implemented with HA rules, so I would call that less relevant.)

prairiesnpr, 2025-01-06:
The only other thing I would like to know, is do we actually need the magic packet logic? If you can comment out `.device_class(EnchantedDeviceV2)` remove and rejoin and everything works, we can drop all of that. I don't think it's required even though it was in z2m.

closed 2025-01-16 by TheJulianJES