zigpy/zha-device-handlers #3636: [BUG] TZE204_qasjif9e presence sensor configuration values incorrectly handled.

gnif, 2024-12-25:
### Bug description

After upgrading HA, the quirk for the TZE204_qasjif9e that has overridden my local quirk I was previously using has broken configuration of these devices. The values for Maximum range and Minimum range are not being correctly handled.

1) I have a device that is correctly configured using the older quirk, with a range of 1.8m, however the UI reports this as 180m.
2) Adjusting the value makes it impossible to set a useful value, as setting it to 9m results in it really only being set to 9cm.

### Steps to reproduce

1. Go to the device configuration
2. Adjust the maximum range to any value

### Expected behavior

The device is configured with the correct range.

### Screenshots/Video

Correctly configured device as seen by HA
[image]

Incorrectly configured device by HA
[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
[... 769 lines cut ...]
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

Not Applicable.

</details>

gnif, 2024-12-25:
Just noted too that the "Presence Target Distance" is out by an order of magnitude also, reporting 30m instead of 3m.

prairiesnpr, 2024-12-25:
Thanks, this is an easy fix, the question is how broadly this applies. This base quirk is used for 9 other motion devices, maybe the z2m source would show if they all require scaling by 100 or just a subset.

TheJulianJES, 2024-12-28:
@gnif Can you try the updated quirk from https://github.com/zigpy/zha-device-handlers/pull/3645?  You'll need to add this at the top of the custom quirk file though:

```python
from zigpy.quirks import DEVICE_REGISTRY
# other existing imports from custom quirks

# remove upstream quirk from registry
del DEVICE_REGISTRY._registry_v2[("TZE204_qasjif9e", "TS0601")]

# add custom quirk...
```

Then, just put that file (`ts0601_motion.py`) into your custom quirks folder.
You may need to create a `custom_zha_quirks` folder in your `/config` and set the `custom_quirks_path` in your HA `configuration.yaml` if you haven't done that already.

If you can't test it at the moment or it doesn't work, let me know. I think the PR changes *should* be fine to merge as-is.

gnif, 2024-12-28:
This works but it's now missing the presence sensitivity configuration value.

closed 2024-12-30 by TheJulianJES