zigpy/zha-device-handlers #3638: [Device Support Request] TS0601 _TZE204_l8xiyymq 6-way DIN relay

bulcke, 2024-12-25:
### Problem description

6-way DIN mounted zigbee relay https://vi.aliexpress.com/item/1005008082245864.html
Device is easily added to HomeAssistant (I'm using a Sonoff Zigbee 3.0 USB Dongle Plus). But no entities are created

### Solution description

Probaly very similar to https://github.com/zigpy/zha-device-handlers/issues/2945 or https://github.com/Koenkk/zigbee2mqtt/issues/18371#issuecomment-1976836792

I tried adding the correct model to https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py

```
    signature = {
        MODELS_INFO: [
            ("_TZE204_dqolcpcp", "TS0601"),
            ("_TZE204_l8xiyymq", "TS0601"),
        ],
```

But I don't know much about quirks or python. I'm not a coder. Obviously it does not work. I would appreciate some help adding this.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_l8xiyymq",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```
[... 620 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

bulcke, 2024-12-25:
Oh wow, did not expect this to work that easy: I added the model info to the TuyaSextupleSwitchTO class in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py :

```
class TuyaSextupleSwitchTO(TuyaSwitch):
    """Tuya sextuple channel switch time on out cluster device."""

    quirk_id = TUYA_PLUG_MANUFACTURER

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81
        # input_clusters=[0x0000,0x0004,0x0005,0xEF00]
        # output_clusters=[0x000A,0x0019]>
        MODELS_INFO: [
            ("_TZE200_9mahtqtg", "TS0601"),
            ("_TZE200_wnp4d4va", "TS0601"),
            ("_TZE204_l8xiyymq", "TS0601"),
        ],
```

Reloaded HA and the switches appeared and work fine.

closed 2025-01-05 by bulcke

TomeNikov, 2025-02-19:
@bulcke I'm a super noob with this but while searching the web I can across this and it seems I've facing the same issue here as I've purchased the same product.
Any chance you could share step by step how you got the relay to work properly?

Thanks in advance, your help will be greatly appreciated.