zigpy/zha-device-handlers #3642: [BUG] TS0601 _TZE200_3yp57tby cannot set temperature

MacSergey, 2024-12-25:
### Bug description

I have issues with this device. I can see current temperature and it is immediatelly updated when I change temperature on thermastat, but when i am trying to set temperature from HA, it is not appied.

### Steps to reproduce

trying to adjust target temperature

### Expected behavior

target temperature is updated

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x000d",
        "0x0201",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_3yp57tby",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

Sir-Udo, 2024-12-27:
if its help, here some dagnostic data
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 2096,
    "manufacturer": "_TZE200_hhrtiq0x",
    "model": "TS0601",
    "name": "_TZE200_hhrtiq0x TS0601",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.tuya.ts0601_trv.SiterwellGS361_Type2",
    "quirk_id": null,
    "manufacturer_code": 4098,
    "power_source": "Battery or Unknown",
    "lqi": 230,
    "rssi": -35,
    "last_seen": "2024-12-27T07:24:52",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 128,
        "manufacturer_code": 4098,
        "maximum_buffer_size": 82,
        "maximum_incoming_transfer_size": 82,
        "server_mask": 11264,
        "maximum_outgoing_transfer_size": 82,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0301",
          "input_clusters": [
            "0x0000",
            "0x0001",
            "0x0004",
            "0x0005",
            "0x0201",
            "0x0204",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_hhrtiq0x",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "climate.thermostat_bad_thermostat",
        "name": "_TZE200_hhrtiq0x TS0601"
      },
      {
        "entity_id": "select.thermostat_bad_tastatursperre",
        "name": "_TZE200_hhrtiq0x TS0601"
      },
      {
        "entity_id": "sensor.thermostat_bad_batterie",
        "name": "_TZE200_hhrtiq0x TS0601"
      },
      {
[... 797 lines cut ...]
                "value": null
              },
              "0xfffe": {
                "attribute": "ZCLAttributeDef(id=0xFFFE, name='reporting_status', type=<enum 'AttributeReportingStatus'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x000b": {
                "attribute": "ZCLAttributeDef(id=0x000B, name='upgrade_activation_policy', type=<enum 'UpgradeActivationPolicy'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0000": {
                "attribute": "ZCLAttributeDef(id=0x0000, name='upgrade_server_id', type=<class 'zigpy.types.named.EUI64'>, zcl_type=<DataTypeId.EUI64: 240>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x000c": {
                "attribute": "ZCLAttributeDef(id=0x000C, name='upgrade_timeout_policy', type=<enum 'UpgradeTimeoutPolicy'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": [
              2,
              "current_file_version"
            ]
          }
        }

MacSergey, 2024-12-27:
> hi there, 
> 
> that’s exactly my problem too.
> 
> I have this devices:
> 
> TS0601 von _TZE200_hhrtiq0x
> 
> Quirk: zhaquirks.tuya.ts0601_trv.SiterwellGS361_Type2,
> 
> with a perfect ZigBee connection 
> 
> LQI: 230
> 
> RSSI: -35
> 
> I cannot change anything at the device via HA. no temperature, no on or off. 
> 
> Please Help 
> 
> THX
> 
> Udo
> 
> 

I found the solution, I installed custom quirk for my device from this repo and everything started working correctly 

https://github.com/jacekk015/zha_quirks

Sir-Udo, 2024-12-27:
oh, ok, thx.
i have no idee ybout "quirks"
i only create the directory and copy the file into?
no more to do? 
please help
thx udo

MacSergey, 2024-12-27:
> oh, ok, thx.
> 
> i have no idee ybout "quirks"
> 
> i only create the directory and copy the file into?
> 
> no more to do? 
> 
> please help
> 
> thx udo
> 
> 

Yes, follow the instructions from the repo

Sir-Udo, 2024-12-27:
ok, thx
lol "He who can read, is clearly in advantage" me ;-)
thx

Sir-Udo, 2024-12-27:
ok, it is the same error like bevor. 
some one a other idee?

prairiesnpr, 2024-12-27:
> ok, it is the same error like bevor. some one a other idee?

Your error is probably related to overall zigbee mesh issues or a device problem. A delivery error isn't going to be solved by a quirk.

Sir-Udo, 2024-12-27:
at 10 devices?
LQI: 224
RSSI: -44
LQI: 244
RSSI: -42
LQI: 242
RSSI: -41
etc. etc.
with this koordinator
SLZB-06M

runesl, 2025-02-28:
> I found the solution, I installed custom quirk for my device from this repo and everything started working correctly
> 
> https://github.com/jacekk015/zha_quirks

Thanks. https://github.com/jacekk015/zha_quirks/blob/main/trv_saswell.py also fixed mine, so I can set the temperature remotely for holiday/night mode a.s.o.