zigpy/zha-device-handlers #3647: [Device Support Request] TS0601 by _TZE284_qyflbnbj

dpozimski, 2024-12-27:
### Problem description

Temperature & humidity sensor with external sensor - support request
TS0601 by _TZE284_qyflbnbj ZTHS3 EAN5903794123298
https://botland.store/tuya-home-automation/24810-tuya-zigbee-temperature-and-humidity-sensor-lcd-display-external-probe-rtx-zths3-5903794123298.html

### Solution description

device should report:
temperature & humidity 
optiona: battery level

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_qyflbnbj",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[... 683 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

prairiesnpr, 2024-12-27:
This looks like a very new device, I don't see anything on z2m for it. It's a temp and humidity sensor, should be easy to support, but we would need to know the Data Points.

You could try the following, it's likely that it uses the same DPs as the other Tuya sensors.

```python
from zhaquirks.tuya.builder import  TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE284_qyflbnbj", "TS0601") 
    .tuya_temperature(dp_id=1, scale=10)
    .tuya_humidity(dp_id=2)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)
```

dpozimski, 2024-12-28:
can I kindly ask you how how i can consume it in my home assistance+zha instance?

dpozimski, 2024-12-28:
[image]
[zha-c617d448614aae6fcd54aacdfc88b6c4-_TZE284_qyflbnbj TS0601-22d7c576a20978bab7aa91413e8a5081 (2).json](https://github.com/user-attachments/files/18265001/zha-c617d448614aae6fcd54aacdfc88b6c4-_TZE284_qyflbnbj.TS0601-22d7c576a20978bab7aa91413e8a5081.2.json)

custom quirk added, however temp, humidity & battery shows unknown value in HA

prairiesnpr, 2024-12-28:
Ok,next we need to trigger an update from the device. You can remove and repair it, wait several hours, or try to change the temperature or humidity enough that it sends an update. If it still says unavailable, then enable debug logging in ZHA, get the device to send an update for all three of the DPs, battery, temperature, and humidity, then post the logs.

dpozimski, 2024-12-28:
yep it works, now all parameters are readable

prairiesnpr, 2024-12-28:
> yep it works, all parameters are readable now

Just to confirm, the scaling is correct on everything?

dpozimski, 2024-12-28:
yes

closed 2024-12-29 by TheJulianJES