zigpy/zha-device-handlers #3655: [Device Support Request] Tuya TS0121 (new device_type?) Not getting quirks applied

doff-1, 2024-12-30:
### Problem description

I've recently bougt two new Connecte power sockets (rebranded tuya) which doesn't report power usage correctly, not firmware either.
From before I have 3 other working perfectly fine. The two new ones are sold as the same socket.

Current measurements are having spikes at almost 3000A, and measure around 130-140A while the power measurement is around 40W (seems about correct for a small fridge)
Summation delivered is also increasing too quickly showing about 35kWh in a day.
Voltage seems to be reported correctly.

### Solution description

I don't know how to write or adjust quirks, but I imagine it is the same quirk as before, it just needs to apply to the new identifier as well.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_5ity3zyu",
  "model": "TS0121",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

The log file was too long to paste, so it's attatched as a file.
[zha-7de48d8e8d99ebff82bd942e0e2db63a-_TZ3000_5ity3zyu TS0121-6162eb5fd75c2a402a425d4a6e8da4d2.json](https://github.com/user-attachments/files/18271295/zha-7de48d8e8d99ebff82bd942e0e2db63a-_TZ3000_5ity3zyu.TS0121-6162eb5fd75c2a402a425d4a6e8da4d2.json)

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I'll gladly help testing out a new quirk to get this working for as many people as possible.

prairiesnpr, 2024-12-30:
What is the manufacturer and model for the other switches that work?

doff-1, 2024-12-30:
They're all the same manufacturer and model. 
Manufacturer is Connecte (now part of Futurehome)
Product nr.: [Co017](https://www.connecte.no/stikkontakt-semi-flush)
It's registered in the norwegian "EFO database" with El.nr.: [4500993](https://efobasen.no/produkt/4500993)

Here is the signature of a working device:
```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0009",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_5ity3zyu",
  "model": "TS0121",
  "class": "zhaquirks.tuya.ts0121_plug.TS0121_Var03"
}
```

prairiesnpr, 2024-12-30:
If you get a chance, try this one and see if it works for you.

```python
from zigpy.quirks import DEVICE_REGISTRY

try:
    del DEVICE_REGISTRY._registry_v2[("_TZ3000_5ity3zyu", "TS0121")]
except KeyError:
    pass

from zigpy.quirks.v2 import QuirkBuilder

from zhaquirks.tuya import (
    TuyaNewManufCluster,
    TuyaZBElectricalMeasurement,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)

(
    QuirkBuilder("_TZ3000_5ity3zyu", "TS0121")
    .replaces(TuyaZBOnOffAttributeCluster)
    .replaces(TuyaZBMeteringCluster)
    .replaces(TuyaZBElectricalMeasurement)
    .replaces(TuyaNewManufCluster)
    .add_to_registry()
)
```

doff-1, 2024-12-31:
Thanks a lot! I need to read up on how to apply a custom quirk.
Next week will be hectic, but after that I'll definetly have time.
I'll get back to you with results.

doff-1, 2024-12-31:
Okay, so I found time to figure out and try this new custom quirk, and it WORKS! 😁
Thank you so much for the help. Will it get merged in ZHA?

prairiesnpr, 2024-12-31:
> firmware is still reported as unknown, I will report if it shows up in a while.

Firmware is very hit and miss, I think out of my 80 devices only 2 are showing a firmware other than unknown. So, it may never show.

On the merging, the exiting quirks are odd, since they only match on the [model](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0121_plug.py) and not the manufacturer. I have to discuss with @TheJulianJES on how we can proceed with this, since v2 quirks require both and I don't have a list of all possible manufacturers.

TheJulianJES, 2025-01-01:
It's unfortunately one of the things we can't really do nicely with quirks v2. There are a lot of different "manufacturer names" for these plugs, so the v1 signature matching is useful for this.

In the future, we might be able to filter to just match on the TSxxx model name and then have a `.filter()` that checks if the Tuya manufacturer cluster IDs are present (and e.g. OnOff). We can be relatively sure to have a Tuya plug quirk then, without needing to keep a complete list of Tuya plug manufacturer names.
If it's unavoidable to keep a list, it'll be somewhat 

We also need v2 quirks to expose "quirk ids", but there's a ticket for that in the ZHA repo.

For now, I'd just add a v1 quirk that matches this plug, similar to existing quirks. (You can also use the quirk-generator package for this if it helps.)

closed 2025-08-02 by github-actions[bot]