zigpy/zha-device-handlers #3658: [Device Support Request] TS0601 _TZE284_81yrt3lo Dual energy meter

Ralim, 2024-12-31:
### Problem description

Would like help in adding support for this small 2 channel AC power meter.
Purchased from [Aliexpress](https://www.aliexpress.com/item/1005007670042835.html).

### Solution description

Able to see the power readings for the two channels

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE284_81yrt3lo",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>
[... 654 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

```

</details>

### Additional information

Happy to test quirks, help debug etc. New to custom quirks for Zigbee, but not new to python & programming. Can't find a good guide for this so filing the request while I look more.

sasojerman, 2025-01-12:
Hello,
I’m experiencing the same issue with the recognition of this meter. I believe the problem lies with the incorrect identifier _TZE284_81yrt3lo. I have an identical meter recognized as _TZE204_81yrt3lo, and it is successfully integrated into Zigbee2MQTT, working without any issues. Could the solution be to adjust the external converter for the _TZE204_81yrt3lo meter to support this new identifier?

mitxol, 2025-01-13:
Hi,
I have the same problem but it is when integrating into ZHA. ZHA discovers the device, configures it but it has no entities or sensors. It detects that it is _TZE284_81yrt3lo and does not apply the quirk. I have other devices with quirks applied but this one is not applied.

sasojerman, 2025-01-13:
Hi,
I switched from ZHA to Z2M some time ago. Somehow, I find it more user-friendly. Regarding the converter for this meter in Z2M, I managed to put something together, and it now recognizes the device, and the measurements seem correct. If there are any possible improvements, I’d greatly appreciate them. As for the quirk in ZHA, I’m not sure. This is external converter that worked for me:

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE284_81yrt3lo'},
    ],
    model: 'TS0601_Energy_Meter',
    vendor: 'Tuya',
    description: 'Energy meter with 80A clamp',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [],
    exposes: [ //only report data
        e.ac_frequency(), // Frequency in Hz
        exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
        exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
        exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
        exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
        exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured pow>
        exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured pow>
        exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/rev>
        exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/rev>
        exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward>
        exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward>
        exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse>
        exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse>
        exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
    ],
    meta: {
        tuyaDatapoints: [//only report data
            [111, 'ac_frequency', tuya.valueConverter.divideBy100],
            [101, 'total_power_A', tuya.valueConverter.divideBy10],
            [105, 'total_power_B', tuya.valueConverter.divideBy10],
            [115, 'total_power_AB', tuya.valueConverter.divideBy10],
            [112, 'voltage', tuya.valueConverter.divideBy10],
            [113, 'current_A', tuya.valueConverter.divideBy1000],
            [114, 'current_B', tuya.valueConverter.divideBy1000],
            [110, 'power_factor_A', tuya.valueConverter.divideBy100],
            [121, 'power_factor_B', tuya.valueConverter.divideBy100],
            [102, 'power_direction A', tuya.valueConverter.raw],
            [104, 'power_direction B', tuya.valueConverter.raw],
            [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
            [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
            [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
            [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
            [129, 'update_frequency' , tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

`

Eguar-rc, 2025-01-13:
Hi,
I have the same problem and I would like to try your converter but with the new version 2.0 I don't know where to put the js file.
I use the z2m addon from Home Assistant, can you help me?

Thanks

sasojerman, 2025-01-13:
Hi,
I was also on version 2.0, but during one of my attempts, I reverted to version 1.42. Now, I’m not sure what will happen when I update.
I think there shouldn’t be any issues. Try this procedure:

1. Create a Directory for External Converters
 - Access your terminal (via SSH) or use the File Editor in Home Assistant.
 - Create the directory if it doesn’t already exist:
 ```
mkdir -p /config/zigbee2mqtt/data/external_converters/

```
2. Create a New External Converter File
 - In the /config/zigbee2mqtt/data/external_converters/ directory, create a new file:
 ```
nano /config/zigbee2mqtt/data/external_converters/Tuya_TZE284_81yrt3lo.js

```
 - Paste the code for your device.
 ```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE284_81yrt3lo'},
    ],
    model: 'PJ-1203A',
    vendor: 'Tuya',
    description: 'Energy meter with 80A clamp',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [ //only report data
        e.ac_frequency(), // Frequency in Hz
        exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
        exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
        exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
        exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
        exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
        exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
        exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
        exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
        exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
        exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
        exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
        exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
        exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
    ],
    meta: {
        tuyaDatapoints: [//only report data
            [111, 'ac_frequency', tuya.valueConverter.divideBy100],
            [101, 'total_power_A', tuya.valueConverter.divideBy10],
            [105, 'total_power_B', tuya.valueConverter.divideBy10],
            [115, 'total_power_AB', tuya.valueConverter.divideBy10],
            [112, 'voltage', tuya.valueConverter.divideBy10],
            [113, 'current_A', tuya.valueConverter.divideBy1000],
            [114, 'current_B', tuya.valueConverter.divideBy1000],
            [110, 'power_factor_A', tuya.valueConverter.divideBy100],
            [121, 'power_factor_B', tuya.valueConverter.divideBy100],
            [102, 'power_direction A', tuya.valueConverter.raw],
            [104, 'power_direction B', tuya.valueConverter.raw],
            [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
            [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
            [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
            [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
            [129, 'update_frequency' , tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

```
- Save the file: Press CTRL+O, confirm with Enter, then CTRL+X

3. Add the Converter Path to configuration.yaml
 - Open the configuration.yaml file:
  ```
nano /config/zigbee2mqtt/configuration.yaml

```
- Add or update the external_converters section to include your new file:
 ```
external_converters:
  - data/external_converters/Tuya_TZE284_81yrt3lo.js

```
- Save the changes.

4. Restart Zigbee2MQTT

Eguar-rc, 2025-01-13:
It finally worked! Thank you very much.
In my case the directory is \config\zigbee2mqtt\external_converters, without \data.

tuga89, 2025-01-14:
Hello, I have also installed this module, but I am using ZHA. Could someone share the quirk to make it report correctly with the data? Thank you.Hi, I use

Celsius456, 2025-01-15:
Hello, I just wanted to thank you. Also, I think in next Z2M version realeasing 1 feb, they gonna add official support

simonedimarzio, 2025-01-16:
Hello, do you have any suggestions for solving the issue with the device "_TZE284_81yrt3lo" using ZHA in Home Assistant?  

Currently, it is not supported by the quirk available here: https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter_devices/zhaquirks/tuya/ts0601_energy_meter.py  

Thank you very much!

Edsol, 2025-01-18:
> Hi, I was also on version 2.0, but during one of my attempts, I reverted to version 1.42. Now, I’m not sure what will happen when I update. I think there shouldn’t be any issues. Try this [...]

it works great! thanks!

lio0909, 2025-01-25:
Hello,
same here, any news on support of this device through ZHA?
Thank you!

IreBrezOvire, 2025-01-31:
> Hi, I was also on version 2.0, but during one of my attempts, I reverted to version 1.42. Now, I’m not sure what will happen when I update. I think there shouldn’t be any issues. Try this [...]

Works like a charm, thank you very much Mr. Z2M Wizard :)

wolsty7, 2025-02-07:
> Hello, do you have any suggestions for solving the issue with the device "_TZE284_81yrt3lo" using ZHA in Home Assistant?
> 
> Currently, it is not supported by the quirk available here: https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter_devices/zhaquirks/tuya/ts0601_energy_meter.py
> 
> If it helps, I have the device and am available to capture data and debug logs.
> 
> Thank you very much!

I'm a beginner so I'm sure I have made some mistakes, but I have added a _TZE284_81yrt3lo version to the bottom of @jeverley quirk and it now works perfectly, I hope this helps everyone.

```
"""Tuya Energy Meter."""

from enum import Enum
from typing import Any, Callable, Dict, Optional, Tuple, Type, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaZBElectricalMeasurement,
    TuyaZBMeteringClusterWithUnit,
)
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

# from zigpy.zcl.clusters.homeautomation import MeasurementType

# Manufacturer cluster identifiers for device signatures
EARU_MANUFACTURER_CLUSTER_ID = 0xFF66

# Offset of 512 (0x200) for transating DP ID to Attribute ID
# Attribute IDs don't need to match every device's specific values
DP_ATTR_OFFSET = 512

# Power direction acttributes
POWER_FLOW = 102 + DP_ATTR_OFFSET  # PowerFlow (0: forward, 1: reverse)
POWER_FLOW_B = 104 + DP_ATTR_OFFSET  # PowerFlow (0: forward, 1: reverse)

# Calibration attributes
AC_FREQUENCY_COEF = 122 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_DELIVERED_COEF = 119 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_DELIVERED_COEF_B = 125 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_RECEIVED_COEF = 127 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_RECEIVED_COEF_B = 128 + DP_ATTR_OFFSET  # uint32_t_be
INSTANTANEOUS_DEMAND_COEF = 118 + DP_ATTR_OFFSET  # uint32_t_be
INSTANTANEOUS_DEMAND_COEF_B = 124 + DP_ATTR_OFFSET  # uint32_t_be
RMS_CURRENT_COEF = 117 + DP_ATTR_OFFSET  # uint32_t_be
RMS_CURRENT_COEF_B = 123 + DP_ATTR_OFFSET  # uint32_t_be
RMS_VOLTAGE_COEF = 116 + DP_ATTR_OFFSET  # uint32_t_be

# Device configuration attributes
UPDATE_PERIOD = 129 + DP_ATTR_OFFSET  # uint32_t_be (3-60 seconds supported)

# Local configuration attributes
CHANNEL_CONFIGURATION = 0x5000
SUPPRESS_REVERSE_FLOW = 0x5010
SUPPRESS_REVERSE_FLOW_B = 0x5011
[... 1703 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaElectricalMeasurement,
                    TuyaMetering,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaElectricalMeasurement,
                    TuyaMetering,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

tuga89, 2025-02-07:
TOP !!! thank you !

Duntch144, 2025-02-08:
Here is a converter that work for me, it allows to modify the update frequency to other frequencies

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE284_81yrt3lo'},
    ],
    model: 'TS0601_Energy_Meter',
    vendor: 'Tuya',
    description: 'Energy meter with 80A clamp',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        {
            key: ['update_frequency'],
            convertSet: async (entity, key, value, meta) => {
                await tuya.sendDataPointValue(entity, 129, value);
            },
        },
    ],
    exposes: [
        e.ac_frequency(), // Frequency in Hz
        exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
        exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power AB'),
        exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
        exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
        exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
        exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
        exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
        exposes.numeric('power_direction_A', ea.STATE).withDescription('Power direction A (0: forward, 1: reverse)'),
        exposes.numeric('power_direction_B', ea.STATE).withDescription('Power direction B (0: forward, 1: reverse)'),
        exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
        exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
        exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
        exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
        exposes.numeric('update_frequency', ea.STATE_SET).withUnit('sec').withDescription('Update frequency'),
    ],
    meta: {
        tuyaDatapoints: [
            [111, 'ac_frequency', tuya.valueConverter.divideBy100],
            [101, 'total_power_A', tuya.valueConverter.divideBy10],
            [105, 'total_power_B', tuya.valueConverter.divideBy10],
            [115, 'total_power_AB', tuya.valueConverter.divideBy10],
            [112, 'voltage', tuya.valueConverter.divideBy10],
            [113, 'current_A', tuya.valueConverter.divideBy1000],
            [114, 'current_B', tuya.valueConverter.divideBy1000],
            [110, 'power_factor_A', tuya.valueConverter.divideBy100],
            [121, 'power_factor_B', tuya.valueConverter.divideBy100],
            [102, 'power_direction_A', tuya.valueConverter.raw],
            [104, 'power_direction_B', tuya.valueConverter.raw],
            [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
            [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
            [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
            [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
            [129, 'update_frequency', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

dorianmartinez31, 2025-02-20:
Hello,

I configure this device automatically (supported from 1feb).
I get a state after pairing, but after no more state seems to be received. Time-stamp of state stay unchanged.

Any id?

I also noticed that it is recognized as a Router and not EndPoint in my database.db

seppl0815, 2025-02-25:
Same problem as @dorianmartinez31 here
HA 2025.2.5

LEANOSPA, 2025-03-11:
+1 same problem as  @dorianmartinez31  ZHA (SONOFF ZIGBEE 3.0 USB DONGLE)

https://es.aliexpress.com/item/1005007670042835.html?gatewayAdapt=glo2esp
MODEL BOX: PJ-1203A
TS0601 _TZE284_81yrt3lo

HOME ASSISTANT
Core: 2025.3.1
Supervisor: 2025.03.2
Operating System: 14.2
Frontend: 20250306.0

Any idea to help me ?

jeverley, 2025-03-11:
The custom quirk v1 in comment https://github.com/zigpy/zha-device-handlers/issues/3658#issuecomment-2643002772 should work with ZHA.

The v2 quirk I'm working on the linked pull request above is work in progress and won't work on the main releases until some other dependencies have been merged.

spplecxer, 2025-03-12:
Got it working. Thanks a bunch. Will compare to my clamp meter and come back with the result.

LEANOSPA, 2025-03-16:
> La peculiaridad personalizada v1 en el comentario [#3658 (comentario)](https://github.com/zigpy/zha-device-handlers/issues/3658#issuecomment-2643002772) debería funcionar con ZHA.
> 
> La peculiaridad v2 en la que estoy trabajando, la solicitud de extracción vinculada anterior, es un trabajo en progreso y no funcionará en las versiones principales hasta que se hayan fusionado algunas otras dependencias.
> 
> Editar:
> 
> ¿Puedes compartir la firma de tu dispositivo si sigues teniendo problemas después de intentar lo anterior?
> 
> Tendrás que eliminar cualquier peculiaridad personalizada del dispositivo y volver a cargar ZHA para asegurarte de que muestra la firma "no peculiar" original del dispositivo.

Working!! Although from what I've seen, you'll need to give it time to start seeing the readings... it's not instantaneous, thank you very much.

One small detail: what would I have to change to see the values ​​in kW instead of VA?

Good job!!

cguhring67, 2025-03-22:
Hi, I've bought 2 modules (TS0601 by _TZE284_81yrt3lo) on Aliexpress, last week, and installed today. As said here, ZHA recognized the devices, but without any sensors.
I've searched and found this quirk, and also the method to install a custom quirk.
It worked instantaneously after the restat of HA, but only for one channel. The second (and the third, I think calculated), shows "unknown" on all readings.
How can I obtain the second channel ?

PhilCreasey, 2025-03-24:
I'm new to HA, and have one of these devices.

I'm using ZHA will these devices get added in an update at some point, if so I'm happy to wait, or will I have to install a quirk?

Thanks

LEANOSPA, 2025-03-30:
> Working!! Although from what I've seen, you'll need to give it time to start seeing the readings... it's not instantaneous, thank you very much.
>
> One small detail: what would I have to change to see [...]

LEANOSPA, 2025-03-30:
> Working!! Although from what I've seen, you'll need to give it time to start seeing the readings... it's not instantaneous, thank you very much.
>
> One small detail: what would I have to change to see [...]

The device works for me. In my case, I only use one of the two available channels.

But... How can I modify the detection interval for status values? Let me explain...

Today, when I tried to automate a NOTICE (via Telegram) for a NO VOLTAGE alert after 1 minute of detection, it didn't work.

IT DOESN'T WORK, probably because the voltage detection interval is predefined for a longer time.
I see on the panel that the status change isn't instantaneous, as it is with other types of devices. This is IMPORTANT, especially when you want to automate power FAILURE warnings so you can take action, or to be notified that power is back on.

Where can I change it so that it measures every 1 minute, for example?

n00bus, 2025-04-04:
> The custom quirk v1 in comment [#3658 (comment)](https://github.com/zigpy/zha-device-handlers/issues/3658#issuecomment-2643002772) should work with ZHA.
> Are you able to share your device's signature if you're still seeing an issue after trying the above?
 @jeverley thank you for your huge work.
In case you need a feedback. I have no issues with wolsty7's quirk v1 based on yours. Not sure about correct current direction values in all cases though (i need to check it later). Other than that, everything works for both TZE284_81yrt3lo and TZE204_81yrt3lo (both channels and the third virtual summarizing).

ouafnico, 2025-04-04:
Thanks for this quick.
Actually It have added sensors to _TZE204_81yrt3lo, but they are all stuck as "unknown".
Any idea how to fix this?

ouafnico, 2025-04-07:
I tried the method, restarting homeassistant, but no clue, now the devices are empty, and on homeassistant I got this
```
2025-04-07 18:36:22.224 ERROR (SyncWorker_0) [zhaquirks] Unexpected exception importing custom quirk 'ts0601_energy_meter'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_energy_meter.py", line 598, in <module>
    .tuya_dp_attribute(
     ~~~~~~~~~~~~~~~~~^
        dp_id=102,
        ^^^^^^^^^^
    ...<2 lines>...
        converter=lambda x: TuyaEnergyDirection(x),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/builder/__init__.py", line 579, in tuya_dp_attribute
    self.tuya_dp(
    ~~~~~~~~~~~~^
        dp_id=dp_id,
        ^^^^^^^^^^^^
    ...<5 lines>...
        dp_handler=dp_handler,
        ^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/builder/__init__.py", line 528, in tuya_dp
    self.tuya_dp_multi(
    ~~~~~~~~~~~~~~~~~~^
        dp_id,
        ^^^^^^
    ...<9 lines>...
        dp_handler,
        ^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/builder/__init__.py", line 552, in tuya_dp_multi
    raise ValueError(f"DP {dp_id} is already mapped.")
ValueError: DP 102 is already mapped.
```

I didn't tried to fix it yet.

boennhoff, 2025-04-11:
@ouafnico that's strange, are you sure you copied the code exactly and you are running the latest (stable) HA release together with ZHA?

I also have many (>300/day) bad log entries related to ZHA, but those seem related to connectivity. I am pretty sure something is wrong here and an exception shouldn't be thrown, but everything seems to be running okay:

```
[0xBF57:1:0x0020] Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 381, in request return await req.result ^^^^^^^^^^^^^^^^ asyncio.exceptions.CancelledError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zigpy/zcl/__init__.py", line 378, in request return await self._endpoint.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ ...<9 lines>... ) ^ File "/usr/local/lib/python3.13/site-packages/zigpy/endpoint.py", line 270, in request return await self.device.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^ ...<11 lines>... ) ^ File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 380, in request async with asyncio_timeout(timeout): ~~~~~~~~~~~~~~~^^^^^^^^^ File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__ raise TimeoutError from exc_val TimeoutError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/general.py", line 630, in check_in_response await self.checkin_response(True, self.CHECKIN_FAST_POLL_TIMEOUT, tsn=tsn) File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 85, in wrapper with wrap_zigpy_exceptions(): ~~~~~~~~~~~~~~~~~~~~~^^ File "/usr/local/lib/python3.13/contextlib.py", line 162, in __exit__ self.gen.throw(value) ~~~~~~~~~~~~~~^^^^^^^ File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 70, in wrap_zigpy_exceptions raise ZHAException("Failed to send request: device did not respond") from exc zha.exceptions.ZHAException: Failed to send request: device did not respond
[0x7BC8:1:0x0020] Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 381, in request return await req.result ^^^^^^^^^^^^^^^^ asyncio.exceptions.CancelledError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zigpy/zcl/__init__.py", line 378, in request return await self._endpoint.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ ...<9 lines>... ) ^ File "/usr/local/lib/python3.13/site-packages/zigpy/endpoint.py", line 270, in request return await self.device.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^ ...<11 lines>... ) ^ File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 380, in request async with asyncio_timeout(timeout): ~~~~~~~~~~~~~~~^^^^^^^^^ File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__ raise TimeoutError from exc_val TimeoutError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/general.py", line 630, in check_in_response await self.checkin_response(True, self.CHECKIN_FAST_POLL_TIMEOUT, tsn=tsn) File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 85, in wrapper with wrap_zigpy_exceptions(): ~~~~~~~~~~~~~~~~~~~~~^^ File "/usr/local/lib/python3.13/contextlib.py", line 162, in __exit__ self.gen.throw(value) ~~~~~~~~~~~~~~^^^^^^^ File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 70, in wrap_zigpy_exceptions raise ZHAException("Failed to send request: device did not respond") from exc zha.exceptions.ZHAException: Failed to send request: device did not respond
[0xBF57:1:0x0020] Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 381, in request return await req.result ^^^^^^^^^^^^^^^^ asyncio.exceptions.CancelledError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zigpy/zcl/__init__.py", line 378, in request return await self._endpoint.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ ...<9 lines>... ) ^ File "/usr/local/lib/python3.13/site-packages/zigpy/endpoint.py", line 270, in request return await self.device.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^ ...<11 lines>... ) ^ File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 380, in request async with asyncio_timeout(timeout): ~~~~~~~~~~~~~~~^^^^^^^^^ File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__ raise TimeoutError from exc_val TimeoutError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/general.py", line 632, in check_in_response await self.set_long_poll_interval(self.LONG_POLL) File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 85, in wrapper with wrap_zigpy_exceptions(): ~~~~~~~~~~~~~~~~~~~~~^^ File "/usr/local/lib/python3.13/contextlib.py", line 162, in __exit__ self.gen.throw(value) ~~~~~~~~~~~~~~^^^^^^^ File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line [... cut ...] neral, or my zigbee stick (conbee II)... _Might this info help you!_

A little update on my own situation: My two devices are running for a few days now, and I got the impression that the main problem seems to be an unreliable detection of the flow direction. I draw power (>1kW) but it still toggles it's sign... imho, that can't be right... So with `suppress_reverse_flow = true` I can at least reliably determine if power is drawn over some time, but its integration (energy) is obviously wrong...

alexder21, 2025-05-21:
To work in ZHA, I used the [v2 from @jeverley](https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter_v2_quirk/zhaquirks/tuya/ts0601_energy_meter.py) with removal of extra devices and adding missing classes and this worked

```
"""Tuya Energy Meter."""

from __future__ import annotations

from collections.abc import Callable
from typing import Any, Final

from zigpy.quirks.v2.homeassistant import PERCENTAGE, EntityType, UnitOfTime
import zigpy.types as t
from zigpy.zcl import Cluster
from zigpy.zcl.clusters.homeautomation import MeasurementType
from zigpy.zcl.foundation import BaseAttributeDefs, ZCLAttributeDef

from zhaquirks import LocalDataCluster
from zhaquirks.tuya import (
    DPToAttributeMapping,
    PowerPhaseVariant2,
    PowerPhaseVariant3,
    TuyaLocalCluster,
    TuyaZBElectricalMeasurement,
    TuyaZBMeteringClusterWithUnit,
)
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zhaquirks.tuya.mcu import TuyaMCUCluster

ENERGY_DIRECTION: Final = "energy_direction"

class PowerPhaseVariant2:
    """Methods for extracting values from a Tuya Power Phase datapoint.

    'TS0601': (
        '_TZE200_lsanae15',
        '_TZE204_lsanae15'
    )
    """

    @staticmethod
    def voltage_dV(data: ByteString) -> t.uint_t:
        """Return the voltage in decivolts (V * 10)."""
        return data[1] | (data[0] << 8)

    @staticmethod
    def current_mA(data: ByteString) -> t.uint_t:
        """Return the current in milliamperes (A * 1000)."""
        return data[4] | (data[3] << 8)

    @staticmethod
    def power_W(data: ByteString) -> int:
        """Return the signed power in watts (W)."""
        power = data[7] | (data[6] << 8)
        if power > 0x7FFF:
            power = (0x999A - power) * -1
        return power

class PowerPhaseVariant3:
    """Methods for extracting values from a Tuya Power Phase datapoint.

    'TS0601': (
        '_TZE204_ac0fhfiq',
        '_TZE200_qhlxve78',
        '_TZE204_qhlxve78'
    )
    """

    @staticmethod
    def voltage_dV(data: ByteString) -> t.uint_t:
        """Return the voltage in decivolts (V * 10)."""
[... 765 lines cut ...]
        step=0.1,
        multiplier=0.1,
        translation_key="calibrate_current_b",
        fallback_name="Calibrate current B",
        entity_type=EntityType.CONFIG,
        initially_disabled=True,
    )
    .tuya_number(
        dp_id=116,  # % * 10 (1 decimal precision)
        attribute_name="rms_voltage_coefficient",
        type=t.uint32_t_be,
        unit=PERCENTAGE,
        min_value=0,
        max_value=2000,
        step=0.1,
        multiplier=0.1,
        translation_key="calibrate_voltage",
        fallback_name="Calibrate voltage",
        entity_type=EntityType.CONFIG,
        initially_disabled=True,
    )
    .add_to_registry()
)

```

macpip, 2025-05-28:
Hi there, I'm new here and in Home Assistant too (but I'm a veteran Linux user)

I have HomeAssistantOS on Raspberry PI with Sonoff Dongle E as zigbee coordinator with Many Tuya devices working fine with ZHA, **but** TS0601 _TZE284_81yrt3lo is just pairing and seen by HA, but does not expose data (has no entities)

Home Assistant and Sonoff Dongle E are updated to the last stable and "official" release.

I already tried many quirks.

As suggested in many posts, every time I:

1. removed the device from HA
2. removed the content of __pycache__ directory
3. rebooted
4. changed quirk
5. re-addded the device in HA
6. rebooted

Everytime HA sow and added the device, but no entity.

Also tried the following quirks:

- [v2 from @jeverley](https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter_v2_quirk/zhaquirks/tuya/ts0601_energy_meter.py)
- the one posted by @alexder21 in [this post](https://github.com/zigpy/zha-device-handlers/issues/3658#issuecomment-2898969162)

In both cases I get in the logs  (/homeassistant/home-assistant.log):

```
2025-05-28 05:41:38.749 ERROR (SyncWorker_5) [zhaquirks] Unexpected exception importing custom quirk 'ts0601_energy_meter'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_energy_meter.py", line 15, in <module>
    from zhaquirks.tuya import (
    ...<6 lines>...
     )
ImportError: cannot import name 'PowerPhaseVariant2' from 'zhaquirks.tuya' (/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/__init__.py)
```

In my installation the file /usr/local/lib/python3.13/site-packages/zhaquirks/\_\_init\_\_.py mentioned in the logs does not exist (there's no /usr/local/lib/python* directory)

Using some quirks from zhaquirks/tuya/ts0601_din_power.py (eg. [eg. this one](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py) despite I know that it's not suitable for my TS0601 _TZE284_81yrt3lo) HA log says that the quirk gets correctly loaded.

I'm not a coder, so I've not been able to debug.

Could anyone, please, help me?

I'm avaliable for testing

macpip, 2025-05-29:
@alexder21 : Thank you for your [quick response](https://github.com/zigpy/zha-device-handlers/issues/3658#issuecomment-2917015987).
I'm away now. I'll test ASAP and let you know. #

macpip, 2025-05-30:
Here I am. It woks, Now I can see a lot of entities. Thank you @alexder21 !!!

But...

**1.** the value of a lot of entities is "Unknown".

Just the following report values:

- Current (A,B,A+B)
- Power factor (A,B)
- Frequency (A) 
- Sum delivered (A,B)
- Sum received (A,B)
- Tension (A)

as you can see in the following image. Eg. Power values are Sconosciuto (italian for unknown)

[image]

**2.** I had to manually rename all the entities by adding (A) or (B) or (A+B).

Should be possible to have the names automagically differentiated by channel?

clapper66, 2025-06-06:
Thank you @alexder21 I was able to get this working with my TS0601 by _TZE284_81yrt3lo. I'm just seeing the same thing as @macpip where some sensors are unknown.

ouafnico, 2025-06-06:
@alexder21 Very nice job, almost good, just got some unknown

[image]

darmach, 2025-06-13:
@ouafnico @macpip At least some of the `unknown` entitites are in that state because they are present in the firmware (probably shared among different meters) but the measurement sensor is not present. For Example we have 3 voltages, which could suggest 3 meter, but we have only one voltage input, so we could never measure current on separate phases. Same goes for these that only 2 out of 3 have value. This would seem to be our 2 coils (proper readout) on a 3 coil (i.e. three phase) generic meter firmware.

darmach, 2025-06-17:
@macpip @alexder21  Attaching below a different version - working power entities:

<details>

<summary>working power entities</summary>

"""Tuya Energy Meter."""

from enum import Enum
from typing import Any, Callable, Dict, Optional, Tuple, Type, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaZBElectricalMeasurement,
    TuyaZBMeteringClusterWithUnit,
)
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

# from zigpy.zcl.clusters.homeautomation import MeasurementType

# Manufacturer cluster identifiers for device signatures
EARU_MANUFACTURER_CLUSTER_ID = 0xFF66

# Offset of 512 (0x200) for transating DP ID to Attribute ID
# Attribute IDs don't need to match every device's specific values
DP_ATTR_OFFSET = 512

# Power direction acttributes
POWER_FLOW = 102 + DP_ATTR_OFFSET  # PowerFlow (0: forward, 1: reverse)
POWER_FLOW_B = 104 + DP_ATTR_OFFSET  # PowerFlow (0: forward, 1: reverse)

# Calibration attributes
AC_FREQUENCY_COEF = 122 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_DELIVERED_COEF = 119 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_DELIVERED_COEF_B = 125 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_RECEIVED_COEF = 127 + DP_ATTR_OFFSET  # uint32_t_be
CURRENT_SUMM_RECEIVED_COEF_B = 128 + DP_ATTR_OFFSET  # uint32_t_be
INSTANTANEOUS_DEMAND_COEF = 118 + DP_ATTR_OFFSET  # uint32_t_be
INSTANTANEOUS_DEMAND_COEF_B = 124 + DP_ATTR_OFFSET  # uint32_t_be
RMS_CURRENT_COEF = 117 + DP_ATTR_OFFSET  # uint32_t_be
RMS_CURRENT_COEF_B = 123 + DP_ATTR_OFFSET  # uint32_t_be
RMS_VOLTAGE_COEF = 116 + DP_ATTR_OFFSET  # uint32_t_be

# Device configuration attributes
UPDATE_PERIOD = 129 + DP_ATTR_OFFSET  # uint32_t_be (3-60 seconds supported)

# Local configuration attributes
CHANNEL_CONFIGURATION = 0x5000
SUPPRESS_REVERSE_FLOW = 0x5010
SUPPRESS_REVERSE_FLOW_B = 0x5011
POWER_FLOW_PREEMPT = 0x5020

# Suffix for device attributes which need power flow direction applied
UNSIGNED_POWER_ATTR_SUFFIX = "_attr_unsigned"

[... 1699 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaElectricalMeasurement,
                    TuyaMetering,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaElectricalMeasurement,
                    TuyaMetering,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
</details>

mmorenocab, 2025-06-19:
> [@macpip](https://github.com/macpip) [@alexder21](https://github.com/alexder21) Attaching below a different version - working power entities:
> 
>
Tested and seems to be working fine.
It seems that now only the unknown Voltage and Frequency entities from clamp B are missing.

ouafnico, 2025-06-30:
Tried it, but all values except current voltage are at 0 🤔

tinamar10, 2025-07-02:
@darmach 
Thanks, is working for me

mmorenocab, 2025-07-08:
Work fine, Thanks!

vinc, 2025-07-10:
Works for me too, thank you! 🙏

Rihan9, 2025-08-13:
> [@macpip](https://github.com/macpip) [@alexder21](https://github.com/alexder21) Attaching below a different version - working power entities:
> 
> working power entities

With this version, the summation have the wrong unit:

[image]

I'm pretty sure that I didn't use 39MWh of energy in 2 days 😆 .

Another think. I don't know why the entities got automatic translated, I can acept it, however I cannot distinguish with entities is the Clamp A, B or the sum of them. What a mess.

Rihan9, 2025-08-14:
> @Rihan9 Maybe the "point" ( . ) is the decimal separator: is it possible you used ~ 39 KWh in 2 days ?

Yes, It is possible. But the problem is that if I add the sensor in my energy dashboard, the graph scale itself in MWh, so I'm sure the point is being ignored.

gillenoarbe, 2025-08-27:
> Tried it, but all values except current voltage are at 0 🤔

Same here, Power and Current are 0 all the time, Voltage and Frequency are not shown. I also have a TZE204_81yrt3lo (bought both from the same seller) and this worked with the custom ZHA... but they are not compatible with the TZE284.

epenet, 2025-09-08:
On my machine, I have configured to use quirks from `zha_quirks` directory

[image]

and

[image]

Once it is working, you should get extra "Quirk" property in the device properties

[image]

ovisopa, 2025-09-08:
@epenet first of all, Where is the Donate me a beer/cofee button in your git ? 

After an restart of HA the values are displayed in my dashboard !!! 

[image]

Thank you very much !!

Does yours to show negative values for watts ? I will check the clamps, maybe I need to rotate them.

Thank you again.

I'll happily donate a cofee/beer .. just give me an revolut id :)

ArieIV, 2025-09-18:
Aha, this could be the thing, I will check it later on.

ArieIV, 2025-09-20:
Didn't work. I guess there is something wrong with the device.

jsanchezf, 2026-02-14:
Thanks @epenet for this quirk. I nearly returned the device until I found this.

I find, however, something weird: I don't get individual instantaneous demand (W) per clamp. I can see the current (A) reading, but the voltage (V) is unknown. I guess that's why I don't get a power (W) reading.

One difference I see with screenshots by others is that I get three voltage (V) attributes: one is for the power input (the L ad N lines) and has data. The other two are for the two clamps and show voltage as unknown. A clamp cannot read voltage so mixing that "unknown" voltage reading with the current (A) reading to get the instantaneous demand (W) is a losing strategy.

I have the feeling that other versions of the device provide only one voltage reading that is combined with the current to get the instantaneous demand but that my device, having the three separate readings trips the calculation. Am I making sense?

The individual energy (Wh) readings, however, are correct and that blows my mind. 

Any ideas on this? I am thinking of writing a couple of helpers to mix the main voltage readig with the individual current readings to get a estimate of instantaneous demand for each clamp. But it would be nice to have it compensated for upstream.

Legolas66-teszt, 2026-02-17:
Subject: Successful DP mapping for TS0601 _TZE284_81yrt3lo (Firmware 0x4e)

Message:
Hi everyone,

I've spent some time debugging this device (Firmware version 0x4e) with physical testing (clamping and unclamping loads). We've successfully identified the most important Data Points, including a "Net Power" summation DP that wasn't clear before.

Key findings:

DP 115: This is a Net Power Summation (A+B). If you have loads on both clamps, this DP shows the combined wattage. It uses a divisor of 10.

DP 110: This is the Power Factor (cos φ) for the system. It uses a divisor of 100.

DP 101/105: Real-time Power for Clamp A and B (divisor 10).

DP 113/114: Current for Clamp A and B (divisor 1000).

Here is the working quirk I am using. It includes the Greek φ symbol as the unit for the power factor for better visibility in Home Assistant.

A kód (magyar kommentekkel):
Python
"""TS0601 _TZE284_81yrt3lo – Végleges, javított Cos φ és Nettó Teljesítmény kód"""

import zigpy.types as t
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant import (
    UnitOfElectricCurrent,
    UnitOfElectricPotential,
    UnitOfEnergy,
    UnitOfFrequency,
    UnitOfPower,
)
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass

(
    TuyaQuirkBuilder("_TZE284_81yrt3lo", "TS0601")
    
    # --- RENDSZERADATOK ---
    .tuya_sensor(dp_id=111, attribute_name="frequency", type=t.uint32_t, divisor=100, unit=UnitOfFrequency.HERTZ, device_class=SensorDeviceClass.FREQUENCY, state_class=SensorStateClass.MEASUREMENT, fallback_name="Hálózati Frekvencia")
    .tuya_sensor(dp_id=112, attribute_name="voltage", type=t.uint32_t, divisor=10, unit=UnitOfElectricPotential.VOLT, device_class=SensorDeviceClass.VOLTAGE, state_class=SensorStateClass.MEASUREMENT, fallback_name="Hálózati Feszültség")

    # --- "A" ÁRAMKÖR ---
    .tuya_sensor(dp_id=101, attribute_name="power_a", type=t.uint32_t, divisor=10, unit=UnitOfPower.WATT, device_class=SensorDeviceClass.POWER, state_class=SensorStateClass.MEASUREMENT, fallback_name="A kör - Teljesítmény")
    .tuya_sensor(dp_id=113, attribute_name="current_a", type=t.uint32_t, divisor=1000, unit=UnitOfElectricCurrent.AMPERE, device_class=SensorDeviceClass.CURRENT, state_class=SensorStateClass.MEASUREMENT, fallback_name="A kör - Áramerősség")

    # --- "B" ÁRAMKÖR ---
    .tuya_sensor(dp_id=105, attribute_name="power_b", type=t.uint32_t, divisor=10, unit=UnitOfPower.WATT, device_class=SensorDeviceClass.POWER, state_class=SensorStateClass.MEASUREMENT, fallback_name="B kör - Teljesítmény")
    .tuya_sensor(dp_id=114, attribute_name="current_b", type=t.uint32_t, divisor=1000, unit=UnitOfElectricCurrent.AMPERE, device_class=SensorDeviceClass.CURRENT, state_class=SensorStateClass.MEASUREMENT, fallback_name="B kör - Áramerősség")

    # --- TELJESÍTMÉNYTÉNYEZŐ (DP 110) ---
    # Itt a φ jelölés a mértékegység helyén
    .tuya_sensor(dp_id=110, attribute_name="cos_phi", type=t.uint32_t, divisor=100, unit="φ", device_class=SensorDeviceClass.POWER_FACTOR, state_class=SensorStateClass.MEASUREMENT, fallback_name="Teljesítménytényező")

    # --- NETTÓ PILLANATNYI ÖSSZEGZŐ (DP 115) ---
    # Divisor=10, Watt mértékegység az energia be/ki méréséhez (A+B eredő)
    .tuya_sensor(dp_id=115, attribute_name="net_power_watt", type=t.uint32_t, divisor=10, unit=UnitOfPower.WATT, device_class=SensorDeviceClass.POWER, state_class=SensorStateClass.MEASUREMENT, fallback_name="Nettó Teljesítmény (A+B)")

    .add_to_registry()
)

boennhoff, 2026-03-08:
I also took the chance of testing the quirk from @epenet:
* replaced the quirk
* removed the existing devices from ZHA
* restarted HA
* newly added the devices (pressing the pairing button until it flashes repeatedly)

I have two identical devices with 4 clamps in total. The first device seems to work exactly like expected now, but the second one is missing the second "Instantaneous demand" attribute now. No matter how often I remove and re-pair them, the amount of attributes stays the same... any clues?

boennhoff, 2026-03-08:
Thanks @Legolas66-teszt for any help...

**Device 1:**
* TS0601 by _TZE284_81yrt3lo
* Firmware: 0x0000004e
* Zigbee info
  * Nwk: 0xc1a1
  * Quirk: zhaquirks.tuya.builder.EnchantedDeviceV2

**Device 2:**
* TS0601 by _TZE284_81yrt3lo
* Firmware: 0x0000004e
* Zigbee info
  * Nwk: 0x5af7
  * Quirk: zhaquirks.tuya.builder.EnchantedDeviceV2

I tried it again, and followed the "press the pair button for 30 seconds and than cut mains" to eventually factory reset the devices. After trying to add them both again I had another strange behavior (for both the same but I describe only one):
1. Adding them showed all attributes I also had with the former quirks (channels a, b, virtual), so, many more attributes (and even some renamed ones that I used before trying the quirk from @epenet)
2. A lot of them showing Unknown values
3. Immediately removed the device again, and readded it a second time, now showing way less attributes, none with unknown value anymore
4. BUT, now I again only have one power attribute - but now it is at least wrong on both devices... haha

Another thing I recognized is (mentioned in 1.) HA seems to remember former settings even after removing devices, despite how many times I restart it or if I remove statistics... I found some infos about a hidden registry, one should better not touch... Is there a safe way to really remove a ZHA device completely from HA (and all historical traces of it)? I assume if HA tries to reuse formerly saved things (like names) it gets confused when quirks are completely handling things differently - which might actually be my problem instead of any issue with the quirk itself... just a thought...

Legolas66-teszt, 2026-03-09:
"Thanks for the kind words! I've been digging deep into the DP (Data Point) mapping because these 2-clamp Tuya meters are notoriously 'shy' when it comes to reporting both clamps simultaneously in ZHA.

Regarding your question about the dev tools:

The 'Live' Method: I’m mostly using the ZHA Custom Quirk folder in Home Assistant. It’s a bit of a 'trial and error' loop: modify the .py file, restart HA, and check the Device Info / Clusters page.

Analysis: I use the ZHA Network Visualization and the Manage Zigbee Device cluster command tools to 'sniff' what the device is actually sending.

Simulation: Sadly, there isn't a great 'dry run' simulator for Tuya-specific DPs in ZHA yet, so I rely on comparing findings with Zigbee2MQTT's external converters source code.

Two important things to add:

Reverse Flow Testing: I must emphasize that I cannot test the reverse flow (energy export/solar feed-in) in my current setup. If anyone has a solar system and could test how the DPs (likely 107 and 109) behave during export, that would be a huge help!

New Experimental Code: I have a new version of the quirk that successfully shows data from both CT clamps separately AND the total sum simultaneously. However, it hasn't been long-term tested yet and might have some stability issues.

If you are willing to participate in testing this 'alpha' version, I’m happy to publish the code! Just a heads-up: please make sure to back up your current working quirk/configuration before trying it out.

Let me know if you're interested!"

boennhoff, 2026-03-10:
Yes, please hand it over, i'll provide feedback!

JosGit, 2026-03-17:
MUCHISIMAS GRACIAS !!!! SU QUIRK FUNCIONO A LA PRIMERA EN ZHA para este dispositivo:  
`pinza amperimétrica PJ-CT-110301 (2 unidades), junto a su modulo zigbee PJ-1203A de dos canales`

codigo en /homeassistant/custom_zha_quirks/ts0601_din_power.py

---

```
"""
TS0601 _TZE284_81yrt3lo – Nation N32G430 + HLW8112 precision measurement.
A/B channel meter with Import/Export energy registers.
"""

import zigpy.types as t
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant import (
    UnitOfElectricCurrent,
    UnitOfElectricPotential,
    UnitOfEnergy,
    UnitOfFrequency,
    UnitOfPower,
)
from zigpy.quirks.v2.homeassistant.sensor import (
    SensorDeviceClass,
    SensorStateClass,
)

(
    TuyaQuirkBuilder("_TZE284_81yrt3lo", "TS0601")
    .applies_to("_TZE284_81yrt3lo", "TS0601")

    # --- NETWORK DATA ---
    .tuya_sensor(
        dp_id=111, attribute_name="f", type=t.uint32_t, divisor=100,
        unit=UnitOfFrequency.HERTZ, device_class=SensorDeviceClass.FREQUENCY,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Frequency"
    )
    .tuya_sensor(
        dp_id=112, attribute_name="v", type=t.uint32_t, divisor=10,
        unit=UnitOfElectricPotential.VOLT, device_class=SensorDeviceClass.VOLTAGE,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Voltage"
    )

    # --- CHANNEL A ---
    .tuya_sensor(
        dp_id=101, attribute_name="w_a", type=t.uint32_t, divisor=10,
        unit=UnitOfPower.WATT, device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Power A"
    )
    .tuya_sensor(
        dp_id=113, attribute_name="a_a", type=t.uint32_t, divisor=1000,
        unit=UnitOfElectricCurrent.AMPERE, device_class=SensorDeviceClass.CURRENT,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Current A"
    )
    .tuya_sensor(
        dp_id=110, attribute_name="pf_a", type=t.uint32_t, divisor=100,
        unit="φ", device_class=SensorDeviceClass.POWER_FACTOR,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Power Factor A"
    )

    # --- CHANNEL B ---
    .tuya_sensor(
        dp_id=105, attribute_name="w_b", type=t.uint32_t, divisor=10,
        unit=UnitOfPower.WATT, device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Power B"
    )
    .tuya_sensor(
        dp_id=114, attribute_name="a_b", type=t.uint32_t, divisor=1000,
        unit=UnitOfElectricCurrent.AMPERE, device_class=SensorDeviceClass.CURRENT,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Current B"
    )
    .tuya_sensor(
        dp_id=121, attribute_name="pf_b", type=t.uint32_t, divisor=100,
        unit="φ", device_class=SensorDeviceClass.POWER_FACTOR,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Power Factor B"
    )

    # --- TOTAL INSTANTANEOUS ---
    .tuya_sensor(
        dp_id=115, attribute_name="w_sum", type=t.uint32_t, divisor=10,
        unit=UnitOfPower.WATT, device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT, fallback_name="Total Instantaneous Power"
    )

    # --- ENERGY REGISTERS (kWh) ---
    .tuya_sensor(
        dp_id=1, attribute_name="energy_import", type=t.uint32_t, divisor=100,
        unit=UnitOfEnergy.KILO_WATT_HOUR, device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING, fallback_name="Energy (Import)"
    )
    .tuya_sensor(
        dp_id=107, attribute_name="energy_export", type=t.uint32_t, divisor=100,
        unit=UnitOfEnergy.KILO_WATT_HOUR, device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING, fallback_name="Energy (Export)"
    )

    .add_to_registry()
)

```