zigpy/zha-device-handlers #3660: [BUG] Entities for device _TZE284_aao3yzhs do not update - always showing as unavailable

kanemari, 2024-12-31:
### Bug description

Have tried with 3 different _TZE284_aao3yzhs moisture sensor devices. All of them can be joined successfully and have entities created for soil moisture, temperature, battery, firmware, LSI and RSSI. However, the entity values show as unavailable and never update. It appears the devices never reconnect after their initial interview. Last seen date / time never changes after the initial join.

### Steps to reproduce

1. Hold join button on device for 6-8 seconds - light flashes
2. Add to ZHA network
3. Interview completes successfully - entities created.
4. Entity values remain as unavailable

### Expected behavior

Entity values should update periodically.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0408",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_aao3yzhs",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"

### Diagnostic information

{
  "home_assistant": {
[... 964 lines cut ...]
                "attribute": "ZCLAttributeDef(id=0x0002, name='time_zone', type=<class 'zigpy.types.basic.int32s'>, zcl_type=<DataTypeId.int32: 43>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0009": {
                "attribute": "ZCLAttributeDef(id=0x0009, name='valid_until_time', type=<class 'zigpy.types.named.UTCTime'>, zcl_type=<DataTypeId.UTC: 226>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

prairiesnpr, 2024-12-31:
>All of them can be joined successfully

I don't think you are actually getting them joined. I have several of these and they have been connected and working since August 2024. I would suggest the usual battery powered device join tricks, check interference, push the join button every couple seconds for the first thirty seconds after joining, etc.

igorsantos07, 2025-06-28:
I've bought 4 of the same sensor, and while all of them can join my Sonoff-bridge-based ZHA, only one of them show the TH sensors - and just as the OP, always unavailable.

The partially working one says it's using `Quirk: zigpy.quirks.v2.CustomDeviceV2`. Is that correct?
All the other simply identify as `unk_model by unk_manufacturer`, and only include a firmware sensor with unknown status. I've been trying to re-join one of them a bunch of times already this morning, with the exact same results.

According to #3305, it is indeed included since #3315 merged in August/24.

[image]

I am running 2025.6.3, so I can't really understand why would a device join buggy as the OP described, and other three wouldn't even be properly recognized. I don't know much about Zigbee inner workings, though, I know understand about writing YAML and pressing join buttons :))

Here's the diag file for the unrecognized one: [zha-4b673429c0955d0e633571e38e6539dd-unk_manufacturer unk_model-853fff0a73cc8973a0283feada59d81d.json](https://github.com/user-attachments/files/20961988/zha-4b673429c0955d0e633571e38e6539dd-unk_manufacturer.unk_model-853fff0a73cc8973a0283feada59d81d.1.json)

And for the one with empty sensors:  [zha-4b673429c0955d0e633571e38e6539dd-_TZE284_aao3yzhs TS0601-d572d325e0cfb497eaa267ea525eadc4.json](https://github.com/user-attachments/files/20961992/zha-4b673429c0955d0e633571e38e6539dd-_TZE284_aao3yzhs.TS0601-d572d325e0cfb497eaa267ea525eadc4.1.json)

closed 2026-01-02 by github-actions[bot]