zigpy/zha-device-handlers #3666: [Device Support Request] Moes Star Ring Smart Thermostat TS0601 _TZE204_lpedvtvr

nholloh, 2025-01-03:
### Problem description

I'd like to request support for [Moes Star Ring Smart Thermostat](https://moeshouse.com/products/zigbee-star-ring-smart-thermostat-temperature-controller-water-boiler-electric-heating?variant=49631043682619). Somebody over at zigbee2mqtt has already started on a [device handler](https://github.com/Koenkk/zigbee2mqtt/issues/25038).

### Solution description

Support for the device. I managed to get the transfer of temperature (current room temp and target temperature) to home assistant working, currently it is not working the other way around, however. If anybody was willing to sit down with me and help me complete this, that would be great! I am having a hard time with the (lack of) documentation, specifically on the home assistant side of things.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_lpedvtvr",
  "model": "TS0601",
  "class": "moes_test.MoesBHTTest"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
[... 1302 lines cut ...]
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHTManufClusterTest,
                    MoesBHTThermostatTest,
                    MoesBHTUserInterfaceTest,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

prairiesnpr, 2025-01-04:
We just added https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_thermostat.py may be a good starting point.

nicmela, 2025-02-28:
Hello, any news on this?

nicmela, 2025-03-03:
A zigbee2mqtt integration is available. There's a lot of informations here: https://www.zigbee2mqtt.io/devices/ZHT-SR.html
would be possible to build a ZHA quirk with this data? I have tried but still no success :(

deanrparry, 2025-04-03:
Yes this would be a fantastic addition as the thermostat looks very smart indeed where things like the the set point and the current temperature are not available and would be nice to be able to synchronize the time on the device as well. Not sure if programming it is possible or whether that would have to be done from the Moe's app only?

deanrparry, 2025-04-05:
WOW! I tried for 2 days to get something working :( sorry guys but this is DIFFICULT for a novice!

nicmela, 2025-04-05:
> WOW! I tried for 2 days to get something working :( sorry guys but this is DIFFICULT for a novice!

Yeah.. I also surrendered... Can't make it work :(

prairiesnpr, 2025-04-05:
I have pretty much zero time right now, but if you post what you tried, we could probably provide some guidance.

nigels0, 2025-05-18:
I’ve got it partially working with the following quirk. 
`"""Map from manufacturer to standard clusters for thermostatic valves."""

from typing import Any

from zigpy.profiles import zha
from zigpy.quirks.v2.homeassistant import PERCENTAGE, UnitOfTemperature, UnitOfTime
from zigpy.quirks.v2.homeassistant.binary_sensor import BinarySensorDeviceClass
from zigpy.quirks.v2.homeassistant.sensor import SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.hvac import RunningState, Thermostat

from zhaquirks.tuya import TUYA_CLUSTER_ID
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaThermostatSystemMode(t.enum8):
    """Tuya thermostat system mode enum."""

    Auto = 0x00
    Heat = 0x01
    Off = 0x02
#    Heat = 0x00
#    Off = 0x01

class TuyaThermostatSystemModeV02(t.enum8):
    """Tuya thermostat system mode enum, auto and manual."""

    Auto = 0x00
    Manual = 0x02

class TuyaThermostatEcoMode(t.enum8):
    """Tuya thermostat eco mode enum."""

    Comfort = 0x00
    Eco = 0x01

class ScheduleState(t.enum8):
    """Schedule state option."""

    Disabled = 0x00
    Enabled = 0x01

class State(t.enum8):
    """State option."""

    Off = 0x00
    On = 0x01

class TuyaHysteresis(t.enum8):
    """Tuya hysteresis mode."""

    Comfort = 0x00
    Eco = 0x01

class TuyaPresetMode(t.enum8):
    """Tuya preset mode."""

    Manual = 0x00
    Auto = 0x01
    Off = 0x02
    Heat = 0x03

class TuyaThermostatV2(Thermostat, TuyaAttributesCluster):
    """Tuya local thermostat cluster."""

[... 164 lines cut ...]
        type=t.uint16_t,
        unit=UnitOfTemperature.CELSIUS,
        min_value=5,
        max_value=30,
        step=1,
        multiplier=0.1,
        translation_key="eco_temperature",
        fallback_name="Eco temperature",
    )
    .tuya_number(
        dp_id=101,
        attribute_name=TuyaThermostatV2NoSchedule.AttributeDefs.local_temperature_calibration.name,
        type=t.int32s,
        min_value=-10,
        max_value=10,
        unit=UnitOfTemperature.CELSIUS,
        step=1,
        translation_key="local_temperature_calibration",
        fallback_name="Local temperature calibration",
    )    

    .skip_configuration()
    .add_to_registry()
)
`

PirvuCatalin, 2025-06-20:
@nigels0  can you describe a bit what's working and what not? I'll also try your quirk and maybe try to improve it. Thanks!

nigels0, 2025-06-20:
Ah, I see what you mean. You can change the value programmatically, but it resets after a few seconds. I don’t know how that can be bypassed.

PirvuCatalin, 2025-06-20:
Yes, I meant exactly that.
It wouldn't bother me that much to have a different temperature shown, what bothers me is that the sensor internal "decision" of turning on/off is based on that temperature. (of course, this wouldn't be a big deal if you use this thermostat only as a display and not actually using the internal relay connected to a heating system... which I do)

So the only workaround I can think of right now is creating an automation that checks changes in the sensor "external temperature" and a sensor you have in the house => When either changes, calculate the offset and set the "local_temperature_calibration" in such a way that the temperature shown (and thus the on/off relay) will be correct.

PirvuCatalin, 2025-06-20:
Running state would actually just change the HVAC_Mode operation from None to Heating. Which is one part of actually making it work.
This won't turn on the Heating.
You can also try yourself by setting the sensor mode to Floor. This would change the temperature shown to 4 degrees celsius and the relay will not turn on or off whatever you do (even changing the running state, aka heating mode)

PirvuCatalin, 2025-06-23:
Hi @nigels0 ,
Seems that there's no other way of correcting the temperature shown, the only way being the local temperature offset.

For this, I successfully implemented an automation that dynamically changes the zht-sr calibration. I needed it to be dynamic because I have multiple temperature sensor across the house and the house actual temperature has a complex logic behind.

I created a new sensor in config.yml that calculates the difference between the real (actual) house temperature and the sensor of ZHT-SR.

```
house_vs_thermostat_temperature_diff:
        unique_id: house_vs_thermostat_temperature_diff
        friendly_name: House VS Thermostat Temperature Diff
        icon_template: mdi:thermometer
        unit_of_measurement: "°C"
        value_template: >
          {{ ((states.sensor.house_temperature.state | float) - (states.climate.tze204_lpedvtvr_ts0601_2.attributes.current_temperature | float)) | int }}
```

The ideal would be to set this difference to be equal to 0.

In order to achieve this I created a new automation that looks like this:

[image]

I also added the automation yaml:

```
alias: Update Central Heating Thermostat temperature offset when it changes
triggers:
  - trigger: state
    entity_id:
      - sensor.house_vs_thermostat_temperature_diff
    for:
      hours: 0
      minutes: 0
      seconds: 30
conditions: []
actions:
  - action: number.set_value
    metadata: {}
    data:
      value: >-
        {{ (states.sensor.house_vs_thermostat_temperature_diff.state | int) +
        (states.number.tze204_lpedvtvr_ts0601_local_temperature_offset_2.state |
        int) }}
    target:
      entity_id: number.tze204_lpedvtvr_ts0601_local_temperature_offset_2
mode: single
```

If anyone has a better solution for this, I'll be glad to try it.
I included a 30 second delay for the "diff" because otherwise the measurements won't be correctly read (i.e. it takes a couple of seconds until the thermostat propagates the new offset.

Liquidmasl, 2025-11-16:
Any progress on this?
Could this be put in a PR?

~~I seam to be having an issue with this, as i get a lot of configuration entites in the device, but i co not get the 'thermostat' it is unknown.. Which is a bit of a bummer ~~

nevermind, just had to swithc modes back and forth and now it seams to be fine!

Rick-Rina, 2026-01-12:
I added this https://github.com/zigpy/zha-device-handlers/issues/3666#issuecomment-3677051654 as a custom quirk (in use for ~3 weeks now). Not testing but "production" in my case (I am just a "user" with technical background, have no idea how all this works) if you will, heating my flat and relying on it (set up a custom thermostat in HA though and use it with average temps around the flat as the internal sensor is kinda weird, nothing to do with this change though). No issues so far and I like to thank you for that.

jogobelo, 2026-10-11:
Quirk for `_TZE204_lpedvtvr` (Moes Star Ring ZHT-SR, 16 A electric floor heating) with a **working weekly schedule**, confirmed on hardware (relay switched exactly at the programmed time).

**Weekly program format** (DP 102–108 = Mon…Sun, type RAW):
`[day 1-7] + 4 x [hour, minute, temp_hi, temp_lo]`, temperature x10 big-endian (0x00C8 = 20.0 C).
- Exactly 4 slots per day; a 5th slot is ignored by the device.
- Only active in "Program" preset (DP 2 = 2).
- Day numbering confirmed for Saturday (6 = DP 107).
- Default plan: `06060000c80b1e00c80d1e00c8111e00c8` (06:00, 11:30, 13:30, 17:30 at 20 C).

**Writing**: `TuyaClusterData.attr_value` is int, so RAW can't go through the standard path. The quirk uses a custom `TuyaMCUCluster.write_attributes` that sends a RAW `TuyaData`. Attribute type is `t.CharacterString` and the value is hex text (spaces allowed), so `zha.set_zigbee_cluster_attribute` accepts it, e.g.
`attribute: week_program_6`, `value: "06 06 00 00 c8 0b 1e 00 c8 0d 1e 00 c8 11 1e 00 c8"`.
Reads are decoded to `06:00 20.0 | 11:30 20.0 | ...`.

**Other notes**: abs_min/max heat setpoint limits set to 500/4500 (otherwise ZHA caps max at 30 C). DP47: 0 = heating, 1 = idle. DP50 setpoint x10. Sensor mode IN/AL/OU on DP32. Passes the repo quirk tests.

Credit to Alokin90 and Liquidmasl for the original DP findings. Related: #4591.

[moes_zhtsr_en.txt](https://github.com/user-attachments/files/33306100/moes_zhtsr_en.txt)

[moes_zhtsr_en.txt](https://github.com/user-attachments/files/33306117/moes_zhtsr_en.txt)