zigpy/zha-device-handlers #3667: Tuya Siren (new) TS0601 by _TZE284_nlrfgpny

cibermike20, 2025-01-03:
**Problem description**
I have a tuya zigbee siren. It is a new model i guess. How can i get this working with ZHA?
I can only see it in home assistant but it has no entities (just LQI/RSSI) and can not be controlled (alarm fired).

TS0601
by _TZE284_nlrfgpny

Bought here: [https://es.aliexpress.com/item/1005007022197993.html?spm=a2g0o.order_list.order_list_main.23.4fb4194dPGOGjw&gatewayAdapt=glo2esp](url)

**Solution description**
That i can start and stop the alarm in home assistant and read it's status.

Screenshots/Video

Device signature

Diagnostic information

Logs

Custom quirk

Additional information

Thanks!

prairiesnpr, 2025-01-03:
I'll probably regret it, but ordered one, maybe that will get the woodpecker to leave my house alone. Once it's here I'll add the quirk for it.

cibermike20, 2025-01-03:
> I'll probably regret it, but ordered one, maybe that will get the woodpecker to leave my house alone. Once it's here I'll add the quirk for it.

That would be great. Even if at the end we can just use it as woodpeckers house. : ))
I will wait for that quirk.

Thanks a lot!

prairiesnpr, 2025-01-11:
@cibermike20 the siren came today, added PR but it will take some work to use as a custom quirk since this has to be enchanted or it will just flash continuously on join. If you do want to test, you will need to sublcass `TuyaQuirkBuilder` and add the `tuya_enchantment` method.

cibermike20, 2025-01-11:
> @cibermike20 the siren came today, added PR but it will take some work to use as a custom quirk since this has to be enchanted or it will just flash continuously on join. If you do want to test, you will need to subclass `TuyaQuirkBuilder` and add the `tuya_enchantment` method.

Thanks a lot. I will check it out.
I'm not an expert in custom quirks but I will have a look at one I have and try to use the subclass/method mentioned.

Keep in touch!

cibermike20, 2025-01-11:
> @cibermike20 the siren came today, added PR but it will take some work to use as a custom quirk since this has to be enchanted or it will just flash continuously on join. If you do want to test, you will need to subclass `TuyaQuirkBuilder` and add the `tuya_enchantment` method.

Hi,
I had some time to try the custom quirk. I had to replace "TuyaPowerConfigurationClusterOther" by "TuyaPowerConfigurationCluster". Otherwise, it gave me an error when importing.

Anyway, after doing so, I get now the following error:
File "/config/custom_components/zha_quirks/TS0601_TZE284_nlrfgpny.py", line 85, in <module>
    .tuya_enchantment()
     ^^^^^^^^^^^^^^^^
AttributeError: 'TuyaQuirkBuilder' object has no attribute 'tuya_enchantment'

I guess that I should probably do what you mentioned to subclass TuyaQuirkBuilder and add the tuya_enchantment but, honestly, I don't know how to do that.

Best regards,