zigpy/zha-device-handlers #3672: [Device Support Request] ZigBee 24G MmWave Radar Human Body Presence Motion Sensor

sblite, 2025-01-04:
### Problem description

ZigBee 24G MmWave Radar Human Body Presence Motion Sensor - not showing entities.

### Solution description

Suport added

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

https://www.aliexpress.com/item/1005005652743663.html?spm=a2g0o.order_list.order_list_main.35.17231802lXbWGM&fbclid=IwY2xjawHmQ8FleHRuA2FlbQIxMQABHSfVIeFRUJLqNJjGpYJoKWdcG948k3FPQWaNTiutn20GFAyvWTlk-fvvHA_aem_i7X3a0YiOqK3mgHbo2FLEA

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_mtoaryre",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 682 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

prairiesnpr, 2025-01-05:
Are you willing to test a quirk?

sblite, 2025-01-05:
Of course, please let me know what i need to do. Your help is apprecaited.

prairiesnpr, 2025-01-05:
https://github.com/zigpy/zha-device-handlers/pull/3676 has a potential quirk for these devices, please test and let me know if it works for you.

sblite, 2025-01-05:
Sorry, ive not installed a quirk before and the documentation doesnt explain it very good.

sblite, 2025-01-05:
Yes that works. Thanks

closed 2025-01-16 by TheJulianJES