zigpy/zha-device-handlers #3677: [Device Support Request] TS0601 _TZE200_qyflbnbj Tuya outdoor temperature and humidity sensor

EugeneTM, 2025-01-05:
### Problem description

Negative temps are displayed incorrectly. Real temp: -3.3C. Displayed temp: 6550C.

### Solution description

A fix is necessary for V2 quirk for this device. Possibly fix a signed/unsigned int type fix for the value.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qyflbnbj",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 920 lines cut ...]

<details><summary>Logs</summary>

```python
2025-01-05 17:20:09.186 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 5, 14, 20, 9, 186899, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9AA2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=228, profile_id=260, cluster_id=61184, data=Serialized[b'\tH\x01\x00\xd8\x01\x02\x00\x04\x00\x00\xff\xe0'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=156, rssi=-61)
2025-01-05 17:20:09.188 DEBUG (MainThread) [zigpy.zcl] [0x9AA2:1:0xef00] Received ZCL frame: b'\tH\x01\x00\xd8\x01\x02\x00\x04\x00\x00\xff\xe0'
2025-01-05 17:20:09.189 DEBUG (MainThread) [zigpy.zcl] [0x9AA2:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x09>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=72, command_id=1, *direction=<Direction.Server_to_Client: 1>)
2025-01-05 17:20:09.191 DEBUG (MainThread) [zigpy.zcl] [0x9AA2:1:0xef00] Decoded ZCL frame: TuyaReplacementCluster:get_data(data=TuyaCommand(status=0, tsn=216, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xe0', *payload=65504))]))
2025-01-05 17:20:09.193 DEBUG (MainThread) [zigpy.zcl] [0x9AA2:1:0xef00] Received command 0x01 (TSN 72): get_data(data=TuyaCommand(status=0, tsn=216, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xe0', *payload=65504))]))
2025-01-05 17:20:09.194 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x9AA2:1:0x0402]: cluster_handler[temperature] attribute_updated - cluster[Temperature Measurement] attr[measured_value] value[655040]
2025-01-05 17:20:09.194 DEBUG (MainThread) [zha] Emitting event cluster_handler_attribute_updated with data ClusterAttributeUpdatedEvent(attribute_id=0, attribute_name='measured_value', attribute_value=655040, cluster_handler_unique_id='94:b2:16:ff:fe:f3:45:db:1:0x0402', cluster_id=1026, event_type='cluster_handler_event', event='cluster_handler_attribute_updated') (1 listeners)
2025-01-05 17:20:09.194 DEBUG (MainThread) [zha] Emitting event state_changed with data EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SENSOR: 'sensor'>, unique_id='94:b2:16:ff:fe:f3:45:db-1-1026', device_ieee=94:b2:16:ff:fe:f3:45:db, endpoint_id=1, group_id=None) (1 listeners)
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

V1 quirk was introduced possible here: https://github.com/zigpy/zha-device-handlers/issues/2113

Also for V2 it does not work.

</details>

prairiesnpr, 2025-01-05:
Please try the changes in https://github.com/zigpy/zha-device-handlers/pull/3678

EugeneTM, 2025-01-05:
It is fixed! Thank you.

closed 2025-01-05 by EugeneTM