zigpy/zha-device-handlers #3696: [BUG] Philips RDM002 long press indistinguishable from dial left rotation

fholzer, 2025-01-10:
### Bug description

Long-pressing buttons will correctly emit long_press/hold events, but they will also emit LevelControl events. Dial rotation will emit only LevelControl events. Due to the args/params attached to the LevelControl events emitted by button long-press, they are indistinguishable from dial left rotation. Dial right rotation uses different arguments, so can be distinguished. See https://github.com/zigpy/zha-device-handlers/pull/2340#issuecomment-2577754688 for more details.

### Steps to reproduce

1. enable debug logging on zha integration, or start listening to `zha_event` in dev tools.
2. perform left rotation on RDM002 dial
3. hold any button for several seconds
4. observe the same step_with_on_off with indistinguishable args/params for both of above actions

### Expected behavior

Ideally holding a button shouldn't emit LevelControl events.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4107,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 128,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 128,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x1000",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "RDM002",
  "class": "zhaquirks.philips.rdm002.PhilipsRDM002"
}
```
</details>

gtunes-dev, 2025-01-10:
Thanks for looking into this and creating the issue, @fholzer. Of course I'd personally appreciate it if you considered fixing it, since I think you're the expert on this integration now :)

I have a question for you. Someone noticed that the step_with_on_off events fired by log presses have "transition_time" param of 8. The similar rotation events appear to me to consistently have a transition_time of 4. 

Do you have an opinion on whether or not this is a reliable way to discriminate between the two in a blueprint or script until the underlying issue is fixed? Thanks!

fholzer, 2025-01-10:
<details><summary>Debug logs, filtered</summary>

```python
# Dial rotation left
2025-01-10 19:30:54.771 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 10, 18, 30, 54, 771539, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x514D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=203, profile_id=260, cluster_id=8, data=Serialized[b'\x01\x1c\x06\x01\x0e\x04\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=220, rssi=-45)
2025-01-10 19:30:54.772 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0x0008] Received ZCL frame: b'\x01\x1c\x06\x01\x0e\x04\x00'
2025-01-10 19:30:54.772 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x01>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=28, command_id=6, *direction=<Direction.Client_to_Server: 0>)
2025-01-10 19:30:54.772 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0x0008] Decoded ZCL frame: LevelControl:step_with_on_off(step_mode=<StepMode.Down: 1>, step_size=14, transition_time=4)
--
2025-01-10 19:30:54.780 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 10, 18, 30, 54, 780510, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x514D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=204, profile_id=260, cluster_id=64512, data=Serialized[b'\x1d\x0b\x10\x1d\x00\x14\x00\x010\x01)\xe2\xff!d\x00)\xe2\xff!d\x00)\xe2\xff!\x90\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=224, rssi=-44)
2025-01-10 19:30:54.780 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Received ZCL frame: b'\x1d\x0b\x10\x1d\x00\x14\x00\x010\x01)\xe2\xff!d\x00)\xe2\xff!d\x00)\xe2\xff!\x90\x01'
2025-01-10 19:30:54.781 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x1D>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4107, tsn=29, command_id=0, *direction=<Direction.Server_to_Client: 1>)
2025-01-10 19:30:54.781 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Decoded ZCL frame: PhilipsRdm002RemoteCluster:notification(button=20, param2=3145984, press_type=1, param4=41, param5=65506)

# Dial rotation right
2025-01-10 19:30:57.166 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 10, 18, 30, 57, 166520, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x514D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=205, profile_id=260, cluster_id=8, data=Serialized[b'\x01\x1e\x06\x00\x0e\x04\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=220, rssi=-45)
2025-01-10 19:30:57.167 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0x0008] Received ZCL frame: b'\x01\x1e\x06\x00\x0e\x04\x00'
2025-01-10 19:30:57.167 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x01>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=30, command_id=6, *direction=<Direction.Client_to_Server: 0>)
2025-01-10 19:30:57.168 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0x0008] Decoded ZCL frame: LevelControl:step_with_on_off(step_mode=<StepMode.Up: 0>, step_size=14, transition_time=4)
--
2025-01-10 19:30:57.175 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 1, 10, 18, 30, 57, 175599, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x514D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=206, profile_id=260, cluster_id=64512, data=Serialized[b'\x1d\x0b\x10\x1f\x00\x14\x00\x010\x01)\x1e\x00!d\x00)\x1e\x00!d\x00)\x1e\x00!\x90\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=220, rssi=-45)
2025-01-10 19:30:57.176 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Received ZCL frame: b'\x1d\x0b\x10\x1f\x00\x14\x00\x010\x01)\x1e\x00!d\x00)\x1e\x00!d\x00)\x1e\x00!\x90\x01'
2025-01-10 19:30:57.176 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x1D>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4107, tsn=31, command_id=0, *direction=<Direction.Server_to_Client: 1>)
[... cut ...]
2025-01-10 20:32:08.113 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x1D>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4107, tsn=82, command_id=0, *direction=<Direction.Server_to_Client: 1>)
2025-01-10 20:32:08.113 DEBUG (MainThread) [zigpy.zcl] [0x514D:1:0xfc00] Decoded ZCL frame: PhilipsRdm002RemoteCluster:notification(button=20, param2=3145984, press_type=2, param4=41, param5=651)
```
</details>

fholzer, 2025-01-11:
Okay, seems like option 2 isn't that easy after all. To maintain compatibility, any emitted events would need to have `params` set, though that field is generated by zha's `Device:emit_zha_event` only if the cluster's commands dict has a proper ZCLCommandDef present. The issue is that the fields command definition would be different between button presses and dial rotations...

fholzer, 2025-01-17:
I created a branch for both approaches I mentioned earlier.

1. Override LevelControl's listener_event, and filter out those events that have `transition_time == 8`, which originate from long-pressing buttons. Rotary events have a transition_time of 4. https://github.com/zigpy/zha-device-handlers/compare/dev...fholzer:zha-device-handlers:blackhole-longpress-levelcontrol
2. Completely mute all LevelControl events by disabling that cluster, and simulate LevelControl events. The drawback of this is that the `params` field isn't set, and the event is emitted from the wrong cluster, so the cluster_id in events is wrong. https://github.com/zigpy/zha-device-handlers/compare/dev...fholzer:zha-device-handlers:mute-levelcontrol

While i dislike relying on magic numbers that might change during firmware updates, I would prefer option 1. Let me know what do you think @TheJulianJES. I can open a PR if you want.

gtunes-dev, 2025-01-19:
The automations I've written for this controller rely trigger.event.data.params.step_mode and trigger.event.data.params.step_size. Do those values stop getting set in your #2 approach? 

I understand your concern about #1 but it's what I would probably do in this situation. I might elaborate a bit more in the comment you added to the code just to be clear that the events are, from the perspective of the quirk, spurious.

fholzer, 2025-01-19:
> Do those values stop getting set in your https://github.com/zigpy/zha-device-handlers/pull/2 approach?

That's exactly what happens, which is why I think it's not a viable solution. Users could still access these parameters via `event.data.args.<...>` instead of `event.data.params.<...>`, but this would for sure break things for anyone using this right now.

<details><summary>LevelControl zha_event for option (1):</summary>

```python
event_type: zha_event
data:
  device_ieee: 00:17:88:01:0d:35:cf:2a
  unique_id: 00:17:88:01:0d:35:cf:2a:1:0x0008
  device_id: 6098e4744fb110396a517798e8fdc92b
  endpoint_id: 1
  cluster_id: 8
  command: step_with_on_off
  args:
    - 1
    - 8
    - 4
  params:
    step_mode: 1
    step_size: 8
    transition_time: 4
origin: LOCAL
time_fired: "2025-01-19T20:30:14.326924+00:00"
context:
  id: 01JJ044H9PEZQMGKTKDVG12NFE
  parent_id: null
  user_id: null
```

</details>

<details><summary>LevelControl zha_event for option (2):</summary>

```python
event_type: zha_event
data:
  device_ieee: 00:17:88:01:0d:35:cf:2a
  unique_id: 00:17:88:01:0d:35:cf:2a:1:0xfc00
  device_id: 6098e4744fb110396a517798e8fdc92b
  endpoint_id: 1
  cluster_id: 64512
  command: step_with_on_off
  args:
    command_id: 6
    step_mode: 1
    step_size: 30
    transition_time: 4
    args:
      - 20
      - 3145984
      - 1
      - 41
      - -30
  params: {}
origin: LOCAL
time_fired: "2025-01-19T20:26:12.463673+00:00"
context:
  id: 01JJ03X53FH2TVCEGRWZEHD52Y
  parent_id: null
  user_id: null
```

</details>

gtunes-dev, 2025-01-20:
I'll state the obvious: the second option is a breaking change and, even with advance notice, there's no simple way to write an automation that works before and after the change.

Option 2 not only removes the named values in params, it causes the indexical values currently in data.args to move to data.args.args and the number of, and indexes of, values change. 

I am strongly in favor of simply dropping step_with_on_off events that can be identified as spurious, as your first approach does.

I think you need to hear @TheJulianJES's opinion before deciding, though.

TheJulianJES, 2025-01-23:
(The answers from below were written as I read this, so the most up-to-date answer is at the bottom. Still wanted to let you know my thoughts about the other questions though.)
  
> 1. Write a custom implementation of the LevelControl cluster, inspecting the `transition_time` field to discriminate between dial roation and button long press event sources, only emitting events for the dial. The drawback here is that we rely on Philips not changing these values in a future firmware update.
> 2. Completely discarding all events we get for the LevelControl cluster and instead adding logic to the PhilipsRdm002RemoteCluster. This is viable because we also receive button events for dial rotations, though at the moment we discard those because the device also sends LevelControl events. We could trigger our own LevelControl that are based on the button events, while discarding the LevelControl events from the device altogether. (I hope this makes sense. If not, let me know and I can explain in more detail)
> 
> [...] I thinks option 2 would be preferable. 

From reading this, I'd also prefer option 2. Isn't that also what we do for the Hue dimmers or am I mistaken about this?

> Okay, seems like option 2 isn't that easy after all. To maintain compatibility, any emitted events would need to have `params` set, though that field is generated by zha's `Device:emit_zha_event` [...]

Ah, ok. So we don't have device triggers for the dial rotation, right? Only for the clicks? Hmm..

> I created a branch for both approaches I mentioned earlier.
> [...]
> While i dislike relying on magic numbers that might change during firmware updates, I would prefer option 1

Yes, I'd also prefer that. The other option of emitting "fake ZCL" events isn't ideal IMO. I'd rather have a clear "custom/Hue event" in that case, but option 1 just seems simpler for this.

This device also isn't as common as the Hue dimmers and all remote generally seem to rarely be getting firmware update, so I think option 1 should be fine.

PR #3793 (2025-01-28): Ignore Hue dial `RDM002` LevelControl long press events
## Proposed change
Override LevelControl's listener_event, and filter out those events that have transition_time == 8, which originate from long-pressing buttons. Rotary events have a transition_time of 4, and won't be muted.

This fixes #3696

## Additional information
Details in #3696

## Checklist

- [X] The changes are tested and work correctly
- [X] `pre-commit` checks pass / the code has been formatted using Black
- [X] Tests have been added to verify that the new code works
closed 2025-01-28 by TheJulianJES