zigpy/zha-device-handlers #3710: [Device Support Request] TS0601 _TZE200_qrztc3ev Tuya outdoor temperature and humidity sensor

puyopsy, 2025-01-12:
### Problem description

Following this issue (https://github.com/zigpy/zha-device-handlers/issues/2694), I have to applied the same custom_quirk to the temperature/humidity sensor TS0601 _TZE200_qrztc3ev without success:

[image]

I have tried to apply the following quirk:
```
from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE200_qrztc3ev", "TS0601")
    .tuya_temperature(dp_id=1, scale=10)
    .tuya_humidity(dp_id=2, scale=10)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)
```

### Solution description

Have the humidity, battery and temperature sensor working.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4742,
    "maximum_buffer_size": 74,
    "maximum_incoming_transfer_size": 404,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 404,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0020",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qrztc3ev",
[... 922 lines cut ...]
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE200_qrztc3ev", "TS0601")
    .tuya_temperature(dp_id=1, scale=10)
    .tuya_humidity(dp_id=2, scale=10)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)

puyopsy, 2025-01-12:
[home-assistant_zha_2025-01-12T22-10-32.263Z.log](https://github.com/user-attachments/files/18391361/home-assistant_zha_2025-01-12T22-10-32.263Z.log)
Here is a log file of me trying to add one of the _TZE200_qrztc3ev (I have 3 of them).

Somehow one is working and the two others are not:
device 1:  [image]
device 2: [image]
device 3: [image]

prairiesnpr, 2025-01-12:
I'll look at your logs when I'm back to a computer. The scaling can be adjust with the scale value, just remove it and it will use the default of 100. If it's working for one and not the other two, I'd suspect they didn't really join properly or they just haven't reported in yet. Try repairing or just give them a couple hours and see if the fall off or start reporting.

puyopsy, 2025-01-13:
I have tried to repair and waited 12 hours but no luck. Somehow the only device that was reporting only reported like one time so I do no trust the data:
[image]

prairiesnpr, 2025-01-16:
I did look at the join logs, I see the join, but the devices never report any values. Unless this needs the Enchanted cluster, didn't look like z2m had it, not sure what's up. Other than these sensors, you have a robust zigbee network and everything else works?

puyopsy, 2025-01-16:
I think my Zigbee network is very strong, I have about 96 devices and many of them are also acting as routers:
[image]
Also I tried to remove and add the 3 devices like 30 times using different Quirks. But only the default one provide me at least the battery level.

Maybe I did a something wrong while trying to capture the logs.
I will try again this evening when I am back home.

puyopsy, 2025-01-16:
Here is another try that I have just made:
[home-assistant_zha_2025-01-16T22-27-38.343Z.log](https://github.com/user-attachments/files/18446031/home-assistant_zha_2025-01-16T22-27-38.343Z.log)

The quirk that I try to use is not working so it seems to skip it to use the regular setting. As a consequence, both temperature and humidity sensors do not report:

[image]

Is there anything else that I can provide?

prairiesnpr, 2025-01-16:
Ok, a few questions. do you have another quirk somewhere in your custom quirks directory? There aren't any quirks in the main repo for `_TZE200_qrztc3ev`. If so, make sure you only have the one for `_TZE200_qrztc3ev`. If that's correct, it should either load the v2 quirk I provided or not get a quirk and all sensors should be unavailable.

dn-gt, 2025-01-17:
Sorry, where can I find the v2 quirk? Having still issues about the decimal for humidty.

[image]

I'm using the quirk from here https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py so far and just added an entry ("_TZE200_qrztc3ev", "TS0601") - I also experimtented with changing RH_MULTIPLIER from 10 to 100 but without success.

[image]

At least the device reports something :-)

dn-gt, 2025-01-17:
Update: I was not waiting long enough, in the meantime after changing the temperature I received an update with correct humidity settings!

[image]

puyopsy, 2025-01-17:
@dn-gt is actually referring to this quirk https://github.com/Peketr/zha-device-handlers/blob/Peketr-patch-zht05/zhaquirks/tuya/ts0601_sensor.py, a V1 that was originally proposed in this issue https://github.com/zigpy/zha-device-handlers/issues/2694 but somehow it doesn't work for my device (though we have the same product _TZE200_qrztc3ev).

prairiesnpr, 2025-01-17:
>  but somehow it doesn't work for my device (though we have the same product _TZE200_qrztc3ev).

Thanks, that's what I assumed, but wanted to be sure I understood.

@dn-gt please try this custom quirk. 

```python
from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE200_qrztc3ev", "TS0601")
    .tuya_temperature(dp_id=1, scale=10)
    .tuya_humidity(dp_id=2, scale=10)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)
```

It's based on the current repo v2 quirk and matches the v1 in the repo that @puyopsy linked, so it should work, but doesn't for @puyopsy and I'm trying to understand why.

puyopsy, 2025-01-20:
Is there anything else I can do to unlock the situation?

dn-gt, 2025-01-20:
> Is there anything else I can do to unlock the situation?

Thanks a lot I just implemented the V2 quirk and it seems to work properly for me. Getting correct temperature an humidity information.

I have added the code snippet to ts0601_sensor.py file and commented out the additional row I have entered.

from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE200_qrztc3ev", "TS0601")
    .tuya_temperature(dp_id=1, scale=10)
    .tuya_humidity(dp_id=2, scale=10)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)

[image]

dn-gt, 2025-01-20:
Sorry, after further refresh again the humidity scaling is wrong with just 4.6 %

prairiesnpr, 2025-01-20:
> Sorry, after further refresh again the humidity scaling is wrong with just 4.6 %

Change `.tuya_humidity(dp_id=2, scale=10)` to `.tuya_humidity(dp_id=2)`

prairiesnpr, 2025-01-20:
@dn-gt does this have a clock? I realized that I had https://github.com/zigpy/zha-device-handlers/pull/3683 which should be the correct quirk for `_TZE200_qrztc3ev` but we can't seem to get the set_time to work correctly. May be worth testing and see if that adds more functionality.

dn-gt, 2025-01-20:
> [@dn-gt](https://github.com/dn-gt) does this have a clock? I realized that I had [#3683](https://github.com/zigpy/zha-device-handlers/pull/3683) which should be the correct quirk for [...]

No clock on the display, just information for temperature, humidity and battery status.

puyopsy, 2025-01-20:
Mine has a clock
[image]

prairiesnpr, 2025-01-20:
Thanks, I'm still trying to track down the set time issue, I know why the device ignores it, but I can't duplicate it in the tests. Basically, the device want's the set time command to be sent as a non-manufacturer specific, yet for some reason it's still being sent as manufacturer specific even when that's overridden.

closed 2025-01-23 by TheJulianJES

prairiesnpr, 2025-01-24:
> does this mean that it will be fixed in the next release?

Yes, this has been merged. Once 2025.2 is released, delete any custom quirks and you should be good.

Meerhoven, 2025-02-04:
He guys, I want to implement the temperature / humidity sensor with clock as well. I must be doing something really basic wrong, but I 'm not able to collect the data (temperature / humidity) from the _TZE200_qrztc3ev.
I have setup a quirk (ts0601_sensor) in custom_zha_quirks and added the code from nous.ts file mentoined above by prairiesnpr to the quirk.
But this doesn't seem to work. Can you help me in the right direction?

Meerhoven, 2025-02-04:
> [@Meerhoven](https://github.com/Meerhoven) This has been merged, support will be available in HA 2025.2 which is in beta now.

Thank you for the quick reply, is the quirk no longer needed in version 2025.2? If so, will this also be for other quirks?

prairiesnpr, 2025-02-04:
Any quirk in this repo is included by default on Home Assistant. If you have a custom quirk that overlaps, please delete it.

PR #3683 (2025-01-23): Add Nous (Tuya) SZ-T04
## Proposed change

Adds Nous (Tuya) SZ-T04

Manufacturers:

- _TZE200_lve3dvpy
- _TZE200_c7emyjom
- _TZE200_locansqn
- _TZE200_qrztc3ev
- _TZE200_snloy4rw
- _TZE200_eanjj2pa
- _TZE200_ydrdfkim
- _TZE284_locansqn

## Additional information

See: https://github.com/Koenkk/zigbee-herdsman-converters/blob/40e72bc2515deb67ef35dddab04437342b7b3eb9/src/devices/nous.ts#L21

Closes: https://github.com/zigpy/zha-device-handlers/issues/1286
Closes: #3710

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted usin [... cut]