zigpy/zha-device-handlers #3713: [Device Support Request] _TZE200_pay2byax

flipper, 2025-01-14:
### Problem description

I found the model and manufacturer mentioned here under temp/humidity??

https://github.com/zigpy/zha-device-handlers/blob/57bcec8cf963499b9fca76484619ec57e4a7c088/zhaquirks/tuya/ts0601_sensor.py#L84C1-L97C23

The device works for me with that quirk. But there is a duplicate Illuminance entity and the battery type returns 3 AAA instead of the single CR2032 in the device.

So its currently "partially working"

### Solution description

The device should show correct battery type and only a single Illuminance entity.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4742,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_pay2byax",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

flipper, 2025-01-14:
I managed to fix it for myself but unsure what the main cause is. Currently doing a hacky quirk to fix it.
Stole some code from the IKEA quirks for the CR2032... But seems like its ignored anyways.

```python
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t

from zigpy.quirks import DEVICE_REGISTRY

del DEVICE_REGISTRY._registry_v2[("_TZE200_pay2byax", "TS0601")]

from zhaquirks.tuya.builder import TuyaQuirkBuilder

from zigpy.zcl.clusters.general import (PowerConfiguration)

from zhaquirks.tuya import (TuyaLocalCluster)

BATTERY_VOLTAGE = PowerConfiguration.attributes_by_name["battery_voltage"].id
BATTERY_SIZE = PowerConfiguration.attributes_by_name["battery_size"].id
BATTERY_QUANTITY = PowerConfiguration.attributes_by_name["battery_quantity"].id
BATTERY_RATED_VOLTAGE = PowerConfiguration.attributes_by_name[
    "battery_rated_voltage"
].id

class CR2032TuyaPowerConfigurationCluster(TuyaLocalCluster, PowerConfiguration):
    """Custom Power Configuration cluster that represents battery reports."""

    _CONSTANT_ATTRIBUTES = {
        BATTERY_SIZE: 10,
        BATTERY_QUANTITY: 1,
        BATTERY_RATED_VOLTAGE: 30
    }
(
    TuyaQuirkBuilder("_TZE200_pay2byax", "TS0601")  # Cusam ZG-102ZL
    .tuya_sensor(
        dp_id=101,
        attribute_name="measured_value",
        type=t.uint16_t,
        fallback_name="Illuminance",
        device_class=SensorDeviceClass.ILLUMINANCE,
        state_class=SensorStateClass.MEASUREMENT,
    )
    .tuya_contact(dp_id=1)
    .tuya_battery(dp_id=2, power_config=CR2032TuyaPowerConfigurationCluster)
    .skip_configuration()
    .add_to_registry()
)
`

prairiesnpr, 2025-01-15:
Does the above still result in a duplicate Illuminance cluster?

flipper, 2025-01-16:
> Does the above still result in a duplicate Illuminance cluster?

Nope. My hacky quirk above fixes it.
I'm unsure why...
But also feels a bit weird the sensor is under "temp/humidity" in the repository.

flipper, 2025-01-17:
> @flipper Sure about the battery type? I didn't knew what battery type is inside and as the device is on top of my wardrobe and unless i borrow a big ladder from neighbours, inaccessible for me so I just used the default config. The duplicate illuminance sensor may have been an artifact but nevertheless, there is a PR that will fix the missing unit of measurement https://github.com/zigpy/zha-device-handlers/pull/3718
> 
> 

Yeah I'm sure about the battery type.
Bought it off Aliexpress: https://www.aliexpress.com/item/1005007307128850.html

I have also opened it up. It's a coin cell battery

Thanks for the PR!

PR #3752 (2025-01-24): Add `BatterySize` enum
## Proposed change

Sets correct battery size and voltage for `_TZE200_pay2byax TS0601`

Fixes #3713
## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-01-24 by TheJulianJES