zigpy/zha-device-handlers #3714: [Device Support Request]  CO Sensor - TS0601 TZE284_rjxqso4a

waffelmaik, 2025-01-15:
### Problem description

Hey,
I’ve been using ZHA for some time now and all devices were working OOTB. So I didn’t give it a second thought when ordering some random carbon monoxide sensor from china.
Turns out that it’s not that easy with this one.

the manufacturer calls himself MOES on some other places it’s called differently. The model name is HS-720ES.

### Solution description

I’d like to ask for help developing the necessary quirk to get it running.
Any idea on where to get started?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_rjxqso4a",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[... 675 lines cut ...]
    configure: tuya.configureMagicPacket,
    exposes: [
        /* DP  1 */ e.carbon_monoxide(),
        /* DP  2 */ e.co(),
        /* DP  9 */ tuya.exposes.selfTestResult(),
        /* DP 15 */ e.battery(),
        /* DP 16 */ tuya.exposes.silence()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'carbon_monoxide', tuya.valueConverter.trueFalse0],
            [2, 'co', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw]
        ]
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

</details>

duhow, 2025-01-16:
Try copying the file into `custom_quirks` and adding your variant like in #3693 , test if that works and make a PR

prairiesnpr, 2025-01-16:
> Try copying the file into `custom_quirks` and adding your variant like in [#3693](https://github.com/zigpy/zha-device-handlers/pull/3693) , test if that works and make a PR

That's changed a bit, it looks like this now.
https://github.com/zigpy/zha-device-handlers/blob/e61ecbb4f64291c663fd19d23f95f7db5d33e1f0/zhaquirks/tuya/tuya_co.py#L22

But https://github.com/zigpy/zha-device-handlers/pull/3635 should fix this, you could test as a custom quirk, but would need to comment out the units, there is still a ZHA bug on those we are waiting for.

closed 2025-01-20 by TheJulianJES