zigpy/zha-device-handlers #3725: [BUG] _TZE200_ntcy3xu1 does not show battery or open detection

Nillorian, 2025-01-18:
### Bug description

I am not sure what information to provide so bare with me.

I used a custom quirk for my Tuya smoke sensor and it stopped working.

It seems like a similar issue to https://github.com/zigpy/zha-device-handlers/issues/3589 since the quirks look similar.
 
I researched for updates in the basic quirks and saw the v2 update.
https://github.com/zigpy/zha-device-handlers/pull/3596
(I dont know why that was needed, but that maybe broke my custom quirk)

This is some device information:
TS0601 von _TZE200_ntcy3xu1
Verbunden über [Texas Instruments CC2652](http://homeassistant.local:8123/config/devices/device/7535884d24d78dc3a8c2f827cc056f3d)
Eigenart: zigpy.quirks.v2.CustomDeviceV2

--

Core 2025.1.2
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250109.0

---

I also got this one TS0205 / _TZ3210_up3pngle that works fine

### Steps to reproduce

NA

### Expected behavior

Battery Level 
Opening Detection 

should be available for device _TZE200_ntcy3xu1

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
[... 817 lines cut ...]
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaIasZone,
                    TuyaSmokeDetectorCluster,
                    TuyaPowerConfigCluster,
                    TuyaTamperDetection,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
`

prairiesnpr, 2025-01-18:
Try #3726, you can test like so.

```python
from zigpy.quirks import DEVICE_REGISTRY
from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant.binary_sensor import BinarySensorDeviceClass

from zhaquirks.tuya.builder import TuyaPowerConfigurationCluster2AAA, TuyaQuirkBuilder

del DEVICE_REGISTRY._registry_v2[("_TZE200_ntcy3xu1", "TS0601")]

(
    TuyaQuirkBuilder("_TZE200_ntcy3xu1", "TS0601")
    .tuya_smoke(dp_id=1)
    .tuya_binary_sensor(
        dp_id=4,
        attribute_name="tamper",
        device_class=BinarySensorDeviceClass.TAMPER,
        entity_type=EntityType.DIAGNOSTIC,
        translation_key="tamper",
        fallback_name="Tamper",
    )
    .tuya_dp(
        dp_id=14,
        ep_attribute=TuyaPowerConfigurationCluster2AAA.ep_attribute,
        attribute_name="battery_percentage_remaining",
        converter=lambda x: {0: 10, 1: 80, 2: 200}[x],
    )
    .adds(TuyaPowerConfigurationCluster2AAA)
    .skip_configuration()
    .add_to_registry()
)
```

closed 2025-01-19 by TheJulianJES

Nillorian, 2025-02-09:
@TheJulianJES Thanks for your reply
I added your code as custom quirk - though I expected the code already included in the main release since it is inclueded here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/tuya_smoke.py

The device still does not get recognized properly:

[image]

currently i am running on 
Core 2025.2.1
Supervisor 2025.02.0
Operating System 14.2

TheJulianJES, 2025-02-09:
Since you're already on HA 2025.2.x, you can remove the custom quirk. Please wait a few hours and see if the battery level shows up. Maybe also try to completely remove and re-pair the device.

Nillorian, 2025-02-09:
Removing the battery of the smoke detector did the trick. Thanks a lot.

TheJulianJES, 2025-02-09:
> Resetting one device made all devices reset and no show the correct values.

"not showing" or "now showing"?

If not showing: it might take a few minutes until they show again. I think we can't read these values during pairing and instead need to wait for the device to send us these values.

Nillorian, 2025-02-10:
> "not showing" or "now showing"?

now! 

how one missing letter could end a war :)
thanks a lot and keep up the great work.