zigpy/zha-device-handlers #3729: [Device Support Request] Tuya _TZE204_1youk3hj presence sensor 230v with output switch

andpaw88, 2025-01-19:
### Problem description

Purchase Link: [https://smartbudynek.pl/pl/products/czujnik-obecnosci-z-wykrywaniem-ruchu-podtynkowy-wpuszczany-zigbee-tuya-1190.html](url)

Device information:
TS0601
producent: _TZE204_1youk3hj
IEEE: a4:c1:38:98:de:22:cb:3c
Nwk: 0xc1a5
Device Type: Router

### Solution description

Please support in quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_1youk3hj",
  "model": "TS0601",
[... 1181 lines cut ...]
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

</details>

prairiesnpr, 2025-01-19:
Test #3730 and let me know if that works.

andpaw88, 2025-01-19:
@prairiesnpr  Thank you!
It works, now i have to try with configuration, because it does not detect my daughter in the bathroom, but it look that works corectly with distance, presence, human motion state

andpaw88, 2025-01-24:
I would ask you for help becouse i have problems with zha after addindg quirk for TZE204_1youk3hj
I noticed that sometimes during making snapshot of vm with HA SkyConnect is disconnected and then reconnected. But after reconnect there is problem with starting zha and the reason is duplicated quirk:

<details><summary>Logs</summary>

```python
Rejestrator: homeassistant.config_entries
Źródło: config_entries.py:640
Pierwsze zdarzenie: 13:19:30 (1 zdarzenia)
Ostatnio zalogowany: 13:19:30

Error setting up entry SkyConnect v1.0, s/n: 4830d46bdd93ed1196c67ff23b20a988 - Nabu Casa for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        app_config, auto_form=False, start_radio=False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 239, in new
    await app._load_db()
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 111, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.13/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/registry.py", line 125, in get_device
    raise MultipleQuirksMatchException(
        f"Multiple matches found for device {device}: {matches}"
    )
zigpy.exceptions.MultipleQuirksMatchException: Multiple matches found for device <Device model='TS0601' manuf='_TZE204_1youk3hj' nwk=0xC1A5 ieee=a4:c1:38:98:de:22:cb:3c is_initialized=True>: [QuirksV2RegistryEntry(quirk_file=PosixPath('/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/builder/__init__.py'), quirk_file_line=92, manufacturer_model_metadata=(ManufacturerModelMetadata(manufacturer='_TZE204_1youk3hj', model='TS0601'),), friendly_name=None, device_alerts=(), filters=(), custom_device_class=None, device_node_descriptor=None, skip_device_configuration=True, adds_metadata=(AddsMetadata(cluster=<class 'TZE204_1youk3hj.TuyaOccupancySensing'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({})), AddsMetadata(cluster=<class 'TZE204_1youk3hj.TuyaIlluminanceCluster'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({}))), removes_metadata=(), replaces_metadata=(ReplacesMetadata(remove=RemovesMetadata(cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>), add=AddsMetadata(cluster=<class 'zhaquirks.tuya.builder.TuyaQuirkBuilder.add_to_registry.<locals>.TuyaReplacementCluster'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({}))),), replaces_cluster_occurrences_metadata=(), adds_endpoint_metadata=(), removes_endpoint_metadata=(), replaces_endpoint_metadata=(), entity_metadata=(ZCLEnumMetadata(entity_platform=<EntityPlatform.SENSOR: 'sensor'>, entity_type=<EntityType.STANDARD: 'standard'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='human_motion_state', fallback_name='Human motion state', enum=<enum 'TuyaHumanMotionState'>, attribute_name='human_motion_state', reporting_config=None), NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='fading_time', fallback_name='Fading time', attribute_name='fading_time', reporting_config=None, min=3, max=600, step=1, unit=<UnitOfTime.SECONDS: 's'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DURATION: 'duration'>), NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='detection_distance_max', fallback_name='Maximum range', attribute_name='detection_distance_max', reporting_config=None, min=150, max=600, step=75, unit=<UnitOfLength.CENTIMETERS: 'cm'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DISTANCE: 'distance'>), NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='radar_sensitivity', fallback_name='R [... cut ...] EntityType.STANDARD: 'standard'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='led_indicator', fallback_name='LED indicator', attribute_name='find_switch', reporting_config=None, force_inverted=False, invert_attribute_name=None, off_value=0, on_value=1)), device_automation_triggers_metadata=frozendict.frozendict({}))]
```

</details>

But i have only 2 files in folder with quirks:

[image]

prairiesnpr, 2025-01-24:
This is a known zigpy bug, see https://github.com/zigpy/zigpy/issues/1508, it was fixed and should be good in HA 2025.2.

oapy123, 2025-03-05:
I have just added a [Haozee Human Presence Sensor Light Switch](https://www.amazon.com/dp/B0DNF4KK5G) to Home Assistant using Zigbee2MQTT that gets recognized as _TZE204_1youk3hj.  Everything is working except 'motion sensitivity value' reports unknown and when I try to adjust slider I get this error in log: 

`z2m: Publish 'set' 'motion_sensitivity_value' to 'Adams Office Presence' failed: 'Error: No datapoint defined for 'motion_sensitivity_value''`