zigpy/zha-device-handlers #3733: Question: how to consume tuyaDatapointData in custom zha quirk? (No datapoint handler issue)

MrdUkk, 2025-01-20:
Hello! I can't find some documentation or other information how to add processing TuyaDatapointData that is not known to quirk and zha to date?
i got some MOES Dimmer switch that is constantly spammig network with two datapoints (each second two DPs) and ZHA answering each request to device 'attibute unsupported'.

[zigpy.zcl] [0x66DA:1:0xef00] Received ZCL frame: b'\t\x8e\x02\x00]\x0c\x02\x00\x04\x0fU\xa4\xc2'

[zigpy.zcl] [0x66DA:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x09>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=142, command_id=2, *direction=<Direction.Server_to_Client: 1>)

[zigpy.zcl] [0x66DA:1:0xef00] Decoded ZCL frame: TuyaLevelControlManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=93, datapoints=[TuyaDatapointData(dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x0fU\xa4\xc2', *payload=257270978))]))

[zigpy.zcl] [0x66DA:1:0xef00] Received command 0x02 (TSN 142): set_data_response(data=TuyaCommand(status=0, tsn=93, datapoints=[TuyaDatapointData(dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x0fU\xa4\xc2', *payload=257270978))]))

[zigpy.zcl] [0x66DA:1:0xef00] No datapoint handler for TuyaDatapointData(dp=12, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x0fU\xa4\xc2', *payload=257270978))

[zigpy.zcl] [0x66DA:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl<0x10>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=142, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 0>)

[zigpy.zcl] [0x66DA:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)

maybe someone can point me to right place where i can read some documentation or example quirks that consume custom attributes ?

prairiesnpr, 2025-01-20:
The docs and wiki would be the place to start.

For supporting Tuya devices, see [using the TuyaQuirkBuilder](https://github.com/zigpy/zha-device-handlers/blob/dev/tuya.md).

https://github.com/zigpy/zha-device-handlers/wiki/Tuya-%E2%80%90-v2-Quirk-with-TuyaQuirkBuilder

MrdUkk, 2025-01-20:
unfortunately that documents covers only v2 quirks. i have only v1 quirk. he is working but don't have support for datapoints i observed in zha logs.

MrdUkk, 2025-01-20:
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster, TuyaDimmerSwitch
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOffNM,
)

class TuyaInWallLevelControlNM(NoManufacturerCluster, TuyaInWallLevelControl):
    """Tuya Level cluster for inwall dimmable device with NoManufacturerID."""

class TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch):

    signature = {
        MODELS_INFO: [
            ("_TZE204_hlx9tnzb", "TS0601"), #moes ring star series
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaLevelControlManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaLevelControlManufCluster,
                    TuyaOnOffNM,
                    TuyaInWallLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

closed 2025-08-02 by github-actions[bot]