zigpy/zha-device-handlers #3738: [Device Support Request] _TZ3000_402vrq2i TS004F Tuya Rotary Button

SpeedyGoneZales, 2025-01-20:
### Problem description

Please support Tuya `_TZ3000_402vrq2i`, which is a rotary button, purchased from https://www.aliexpress.us/item/3256807534367914.html

As far as I can see, it is identical to `_TZ3000_abrsvsou`, which is already supported in the [TuyaSmartRemote004FROK](https://github.com/zigpy/zha-device-handlers/blob/b7f2bc7a3becaf5c1d97d7420ff6fccc1d1e2785/zhaquirks/tuya/ts004f.py#L80) class.
I have both, the zha events match.    
I tried adding `("_TZ3000_402vrq2i", "TS004F"),` in line 81 and import it as a custom quirk, but I can see the _TZ3000_402vrq2i continues to be handled by the `TuyaSmartRemote004F` class, and as a result offers incorrect commands (e.g. TURN OFF and TURN ON instead of Toggle).
I'm not sure if this is because of a conflict between the native ZHA quirk and my custom one or for some other reason.

Thanks.

### Solution description

This device should be handled by the [TuyaSmartRemote004FROK](https://github.com/zigpy/zha-device-handlers/blob/b7f2bc7a3becaf5c1d97d7420ff6fccc1d1e2785/zhaquirks/tuya/ts004f.py#L80) class.

Operations supported:

- Toggle
- Long press
- Double press
- Rotate right (slow)
- Rotate left (slow)
- Rotate right (fast)
- Rotate left (fast)
- Rotate right while pressed 
- Rotate left while pressed 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
[... 819 lines cut ...]
  }

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2025-01-20:
First look on the manage Zigbee device from the device card (tre dots right from device reconfigure). Look on signature tab on the last line

MattWestb, 2025-01-20:
I think its only missing group in cluster so try deleting it in line 90 and restart HA ans see if ZHA is loading the quirk.

lowman39, 2025-01-21:
+1 for this, received this rotary knob and cannot get it to operate correctly. Or load a quirk

lowman39, 2025-01-21:
same issue over on zigbee2mqtt/issues https://github.com/Koenkk/zigbee2mqtt/issues/23543

SpeedyGoneZales, 2025-01-23:
I can make it work somewhat by directly working with `zha_events` - here they are:

### Single Press
```
event_type: zha_event
data:
  device_ieee: a4:c1:38:ec:7a:54:ab:27
  unique_id: a4:c1:38:ec:7a:54:ab:27:1:0x0006
  device_id: 24649a18ec0cb8aade6a1eb621154d0d
  endpoint_id: 1
  cluster_id: 6
  command: toggle
  args: []
  params: {}
origin: LOCAL
time_fired: "2025-01-23T23:20:15.865994+00:00"
context:
  id: 01JJAQEQQSM8Z9M2XV6PFBW83M
  parent_id: null
  user_id: null   
```

### Right turn

```
event_type: zha_event
data:
  device_ieee: a4:c1:38:ec:7a:54:ab:27
  unique_id: a4:c1:38:ec:7a:54:ab:27:1:0x0008
  device_id: 24649a18ec0cb8aade6a1eb621154d0d
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 0
    - 13
    - 1
  params:
    step_mode: 0
    step_size: 13
    transition_time: 1
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2025-01-23T23:28:55.575487+00:00"
context:
  id: 01JJAQYK8QSNEJNNYK61T7EV1F
  parent_id: null
  user_id: null
```

### Left turn
```
event_type: zha_event
data:
  device_ieee: a4:c1:38:ec:7a:54:ab:27
  unique_id: a4:c1:38:ec:7a:54:ab:27:1:0x0008
  device_id: 24649a18ec0cb8aade6a1eb621154d0d
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 1
    - 13
    - 1
  params:
    step_mode: 1
    step_size: 13
    transition_time: 1
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2025-01-23T23:29:21.806346+00:00"
context:
  id: 01JJAQZCWECQ8ERV19R1QZANXF
  parent_id: null
  user_id: null

```

### Fast right turn
```
event_type: zha_event
data:
  device_ieee: a4:c1:38:ec:7a:54:ab:27
  unique_id: a4:c1:38:ec:7a:54:ab:27:1:0x0008
  device_id: 24649a18ec0cb8aade6a1eb621154d0d
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 0
    - 109
    - 3
  params:
    step_mode: 0
    step_size: 109
    transition_time: 3
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2025-01-23T23:29:36.077637+00:00"
context:
  id: 01JJAQZTTDMT4SMW0V26EQ7R43
  parent_id: null
  user_id: null
```

### Fast left turn
```
event_type: zha_event
data:
  device_ieee: a4:c1:38:ec:7a:54:ab:27
  unique_id: a4:c1:38:ec:7a:54:ab:27:1:0x0008
  device_id: 24649a18ec0cb8aade6a1eb621154d0d
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 1
    - 157
    - 3
  params:
    step_mode: 1
    step_size: 157
    transition_time: 3
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2025-01-23T23:30:14.563267+00:00"
context:
  id: 01JJAR10D3J26697X8BDFYKWQA
  parent_id: null
  user_id: null
```

I did not get any events for double click, press release (long press), hold and turn.

MattWestb, 2025-01-24:
Double click is only supported in scene mode and normally being changed by setting the attribute but is not possible if the quirk is not loaded. But if triple clicking fast (real fast not slow motion) it changing mode nd you is getting different events.

For getting custom quirk working try the suggested change and if not working look if its more cluster / endpoints changed and implanting it.
Then all working well both device automatons for light commands and scenes shall working OK from the device card.

SpeedyGoneZales, 2025-01-27:
Am just spending some time comparing the `_TZ3000_abrsvsou` (which I also have and which works) with the `_TZ3000_402vrq2i`

### Differences

|Property|_TZ3000_abrsvsou|_TZ3000_402vrq2i|
| :---:   | :---: | :---: |
|nwk|34655|10047|
|Manufacturer code|4742|4417|
|Output clusters|0x0300|-|
|app_version|145|66|
|hw_version|null|1|
|reporting_status|null|0|
|current_file_version|null|66|
|switch_mode|1|0|

_TZ3000_abrsvsou has an array of additional attributes in the 0x300 cluster:

<details><summary>Attributes</summary>   

```yaml   
"unsupported_attributes": []
          },
          "0x0300": {
            "endpoint_attribute": "light_color",
            "attributes": {
              "0xfffd": {
                "attribute": "ZCLAttributeDef(id=0xFFFD, name='cluster_revision', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x400a": {
                "attribute": "ZCLAttributeDef(id=0x400A, name='color_capabilities', type=<flag 'ColorCapabilities'>, zcl_type=<DataTypeId.map16: 25>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x4002": {
                "attribute": "ZCLAttributeDef(id=0x4002, name='color_loop_active', type=<class 'zigpy.types.basic.uint8_t'>, zcl_type=<DataTypeId.uint8: 32>, access=<ZCLAttributeAccess.Read|Scene: 17>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x4003": {
                "attribute": "ZCLAttributeDef(id=0x4003, name='color_loop_direction', type=<class 'zigpy.types.basic.uint8_t'>, zcl_type=<DataTypeId.uint8: 32>, access=<ZCLAttributeAccess.Read|Scene: 17>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x4005": {
                "attribute": "ZCLAttributeDef(id=0x4005, name='color_loop_start_enhanced_hue', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x4006": {
                "attribute": "ZCLAttributeDef(id=0x4006, name='color_loop_stored_enhanced_hue', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x4004": {
                "attribute": "ZCLAttributeDef(id=0x4004, name='color_loop_time', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Scene: 17>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0008": {
                "attribute": "ZCLAttributeDef(id=0x0008, name='color_mode', type=<enum 'ColorMode'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x003c": {
                "attribute": "ZCLAttributeDef(id=0x003C, name='color_point_b_intensity', type=<class 'zigpy.types.basic.uint8_t'>, zcl_type=<DataTypeId.uint8: 32>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x003a": {
                "attribute": "ZCLAttributeDef(id=0x003A, name='color_point_b_x', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x003b": {
                "attribute": "ZCLAttributeDef(id=0x003B, name='color_point_b_y', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0038": {
[... 154 lines cut ...]
              },
              "0xfffe": {
                "attribute": "ZCLAttributeDef(id=0xFFFE, name='reporting_status', type=<enum 'AttributeReportingStatus'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x4010": {
                "attribute": "ZCLAttributeDef(id=0x4010, name='start_up_color_temperature', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0030": {
                "attribute": "ZCLAttributeDef(id=0x0030, name='white_point_x', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0031": {
                "attribute": "ZCLAttributeDef(id=0x0031, name='white_point_y', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": []
          },
```
</details>

I find this odd, as this is not a device that could control color.   
The remainder of the differences seems insignificant, e.g. things like battery voltage or RSSI.

MattWestb, 2025-01-27:
First color cluster is being used in tuya TS004F devices (LIDL is using it of color / temperature lights but not in very Zigbee standard way) so is not very strange only tuya Zigbee devices.
More strange and can making problem is the manufacture code is different.
We is casting one "tuya magic spell" to the device then it joining for fixing some locked functionality like the scene functionality on TS004F devices. If the manufacture code is changed it very likely we need updating the spell 2.
Also the extra color cluster and attribute can being unlocked by the spell (we have seen that also adding more endpoints then getting one spell like the TS004F Dimmer Switch is doing).
Try making one new device class and fixing the in and out cluster list on the endpoints and if its working well the quirk shall being loaded. Then we need see if all functions is working OK or we need digging deeper in the (tuy) rabbit hole.

SpeedyGoneZales, 2025-02-03:
Haven't made much progress, though I found some relevant log entries:

```
2025-02-03 03:15:59.861 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3000_402vrq2i TS004F (a4:c1:38:4c:4b:51:76:63)
2025-02-03 03:15:59.861 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'q2i.TuyaSmartRemote004FROK'>
2025-02-03 03:15:59.861 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
2025-02-03 03:15:59.861 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts004f.TuyaSmartRemote004F'>
2025-02-03 03:15:59.861 DEBUG (MainThread) [zigpy.quirks] Device matches filter signature - device ieee[a4:c1:38:4c:4b:51:76:63]: filter signature[{'model': 'TS004F', 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.DIMMER_SWITCH: 260>, 'input_clusters': [0, 1, 3, 4, 6, 4096], 'output_clusters': [25, 10, 3, 4, 5, 6, 8, 4096]}}}]
```

I tried to cross-reference the clusters with [zigpy](https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/general.py).

```py
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,               #0x0000 (0)
                    PowerConfiguration.cluster_id,  #0x0001 (1)
                    Identify.cluster_id,            #0x0003 (3)
                    Groups.cluster_id,              #0x0004 (4)
                    OnOff.cluster_id,               #0x0006
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,          #0x0019 (25)
                    Time.cluster_id,         #0x000a (10)
                    Identify.cluster_id,     #0x0003 (3)
                    Groups.cluster_id,       #0x0004 (4)
                    Scenes.cluster_id,       #0x0005 (5)
                    OnOff.cluster_id,        #0x0006 (6)
                    LevelControl.cluster_id, #0x0008 (8)
                    LightLink.cluster_id,
                ],
```

I could not find 0x1000 (4096), is this the LightLink.cluster_id?
I'm not sure what to make of the ` device_type mismatch on at least one endpoint` log entry?

SpeedyGoneZales, 2025-02-07:
Able to confirm that `0x1000` appears to be [LightLink](https://github.com/zigpy/zigpy/blob/a082f1257c0acebcdb3cfa3a1bff7535c5a9fa12/zigpy/zcl/clusters/lightlink.py#L81).   
The question is then: 
Why does HA log `Fail because device_type mismatch on at least one endpoint` when everything matches?

SpeedyGoneZales, 2025-02-08:
I finally got it working, here is the result:   
<details><summary>402vrq2i.py</summary>

```
"""Tuya TS004F Rotary Button."""

from __future__ import annotations

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    BUTTON,
    BUTTON_1,
    CLUSTER_ID,
    COMMAND,
    COMMAND_STEP_COLOR_TEMP,
    COMMAND_STEP,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PARAMS,
    PROFILE_ID,
    RIGHT,
    ROTATED,
    SHORT_PRESS,
)
from zhaquirks.tuya import (
    EnchantedDevice,
    TuyaNoBindPowerConfigurationCluster,
    TuyaSmartRemoteOnOffCluster,
)

class TuyaSmartRemote004FRotary(EnchantedDevice):
    """Tuya Smart (rotating) Knob device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=260, device_version=1, input_clusters=[0, 1, 3, 4, 6, 4096], output_clusters=[25, 10, 3, 4, 5, 6, 8, 4096])
        MODELS_INFO: [
            ("_TZ3000_402vrq2i", "TS004F")
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,               # 0x0000 (0)
                    PowerConfiguration.cluster_id,  # 0x0001 (1)
[... 88 lines cut ...]
            ENDPOINT_ID: 1,
            CLUSTER_ID: 6,
        },
        (SHORT_PRESS, LEFT): {
            COMMAND: LEFT,
            ENDPOINT_ID: 1,
            CLUSTER_ID: 6,
        },
    }

```
</details>

A few thoughts:

* It would be great if we could query the device to check if attribute 32772 is `0` (dimmer / remote mode) or `1` (scene mode) and accordingly only assign the top part or bottom part to the `device_automation_triggers` dictionary.     
Currently, all methods are shown in HA, but not all can be used at the same time (depending on which mode the switch is in), which makes configuration confusing. I suspect this is possible somehow, but I'm not sure how.
* There doesn't seem to be an explicit "Hold and Turn" method, for which reason I made the regular turn dim_up / dim_down and the hold-and-turn left and right.    
It's not perfect, but I couldn't think of a better solution.
* I couldn't get a long release event from my device, even though this is documented in the enclosed leaflet. Perhaps my fingers aren't nimble enough...   
* The implementation is different in spirit to the other rotary devices (e.g rotate instead of button press), which is why I didn't submit a pull request for `ts004f.py`. Perhaps @MattWestb you have some thoughts whether I should?
* I noticed HA only loads the custom custom quirks _after_ the custom quirks, which is why I assumed my quirk wasn't loading when in fact it was (after some time)
* I had zha debug logs enabled for a while which broke a surprising number of things.

Many thanks for the help!

MattWestb, 2025-02-09:
Changing selectness DA of the device mode is good and bad.
Normally starting doing one automation from device card you thinking what to do and selecting the DA and then have done it you is testing it and then going back to the device card for doing more so its braking the logic and only changing mode of the device then testing the DA.
The hold and turn is not having any precision and was nearly impossible getting "clean" commands from the device so we was not implanting it (still possible using event as trigger).
The device in scene mode is not sending command for start long pressing only release so we is using it as long press (miss led but also strange having release DA but no long press).
Then i testing quirk i giving then completely different name so i easy can see if its the test one or the internal one that the device is using.
How is your device looking ? If its looks and working like the original ROC or the little larger one it wold being great adding it to ZHA with the "standard" DA so user with different version getting the same functionality and still can using your custom  quirk if like having your DAs functionality.

### Great work done !!

SpeedyGoneZales, 2025-02-14:
I also added a Blueprint for Tuya Smart Knob Dimmer rotary button TS004F _TZ3000_402vrq2i, which is here: 
https://gist.github.com/SpeedyGoneZales/e01dc9f73fac9398ff4f38863bc2d08b    

I added two photographs in the Problem description above.   

I found the precision for press and hold to be accurate, it sends one event per "click", just like it does without holding it down.    

With regards to dimmer mode vs switch mode, it's attribute 32772 that determines the mode, `0` is dimmer, `1` is scene:

<details><summary>32772</summary>     

```       
event_type: zha_event
data:
  device_ieee: a4:c1:38:59:ef:3d:0d:7f
  unique_id: a4:c1:38:59:ef:3d:0d:7f:1:0x0006
  device_id: 62566cfbda274ccf0a7d2343a2ddf526
  endpoint_id: 1
  cluster_id: 6
  command: attribute_updated
  args:
    attribute_id: 32772
    attribute_name: switch_mode
    value: 0
  params: {}
origin: LOCAL
time_fired: "2025-02-14T01:27:27.892736+00:00"
context:
  id: 01JM112QWMECMBEHJSJH0YDPY8
  parent_id: null
  user_id: null
```    
</details>

Is there a way to query the device for the state of an attribute (I was not able to find anything in the documentation)?

MattWestb, 2025-02-14:
The modes is light and scene mode then the tuya scene mods is not switch commands is only numbers used for changing scenes. (look in the latest ZCL or matter spec)
My ROK (Rotating Knob) is reporting the attribute changes:
```
event_type: zha_event
data:
  device_ieee: 00:3c:84:ff:fe:b1:5b:b3
  unique_id: 00:3c:84:ff:fe:b1:5b:b3:1:0x0006
  device_id: b83611f96747bead3cbb01e3fee2133e
  endpoint_id: 1
  cluster_id: 6
  command: attribute_updated
  args:
    attribute_id: 32772
    attribute_name: switch_mode
    value: 1
  params: {}
```
and the DMS (Dimmer Switch) 2:
```
event_type: zha_event
data:
  device_ieee: 60:a4:23:ff:fe:fd:dc:14
  unique_id: 60:a4:23:ff:fe:fd:dc:14:1:0x0006
  device_id: 186bdaffc5452ae2ec1eb05db4b997b3
  endpoint_id: 1
  cluster_id: 6
  command: attribute_updated
  args:
    attribute_id: 32772
    attribute_name: switch_mode
    value: 1
  params: {}
```
then being changed on the device.
If like its possible implanting it as one switch in ZHA GUI that maks it possible changing mode from the GUI and also getting the updated status if changed in the device then the information is in the system but its out of my coding knowledge.

I can only recommending looking on other implementations how they have doing it and try do the same .