zigpy/zha-device-handlers #3749: [BUG] Smoke detector TZE284_0zaf1cr8 battery_level and battery_low

toms001, 2025-01-22:
### Bug description

The Nous E8 smoke detector TZE284_0zaf1cr8 can also output the battery level and the battery_level_low status.
https://nous.technology/share/uploads/source/E8/TZE284_0zaf1cr8.txt
So far, none of this has been recorded.

### Steps to reproduce

.

### Expected behavior

.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

toms001, 2025-01-22:
I may not know enough, but you can find the following in my link (nous-website):

```
       tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.trueFalse0],  // Ð”Ð°Ñ‚Ñ‡Ð¸Ðº Ð´Ñ‹Ð¼Ð°
            [14, 'battery_low', tuya.valueConverter.trueFalse0],  // ÐÐ¸Ð·ÐºÐ¸Ð¹ Ð·Ð°Ñ€ÑÐ´ Ð±Ð°Ñ‚Ð°Ñ€ÐµÐ¸
            [15, 'battery', tuya.valueConverter.raw]  // Ð£Ñ€Ð¾Ð²ÐµÐ½ÑŒ Ð·Ð°Ñ€ÑÐ´Ð° Ð±Ð°Ñ‚Ð°Ñ€ÐµÐ¸
        ],
```

prairiesnpr, 2025-01-22:
What size and quantity of battery?

prairiesnpr, 2025-01-22:
> What size and quantity of battery?

It should look like this, just use existing quirk and add this.

```python
.tuya_binary_sensor(
        dp_id=14,
        attribute_name="battery_low",
        translation_key="battery_low",
        fallback_name="Battery low",
    )
.tuya_battery(dp_id=15, battery_type=BatterySize.AA, battery_qty=2)

```

PR #3768 (2025-01-26): Add battery sensors to Tuya smoke `TZE284_0zaf1cr8`
## Proposed change

Adds battery sensors to `TZE284_0zaf1cr8`
Adds `TZE200_0zaf1cr8`

## Additional information

Closes: #3749

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-01-26 by TheJulianJES