zigpy/zha-device-handlers #3751: [Device Support Request]

maloko416, 2025-01-22:
### Problem description

I cannot get the values of the Avatto din power meter. I tried differents quirks but no success.

### Solution description

No solution

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_goecjd1t",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

[... 680 lines cut ...]

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter_devices/zhaquirks/tuya/ts0601_energy_meter.py

with this modification in the function TuyaEnergyMeter_1CHB
```python
    MODELS_INFO: [("_TZE204_ac0fhfiq", "TS0601"),("_TZE204_goecjd1t", "TS0601")],
```

</details>

### Additional information

The quirk which had an effect was https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter_devices/zhaquirks/tuya/ts0601_energy_meter.py, but the values stay unavailable.

maloko416, 2025-01-22:
product page : https://a.aliexpress.com/_EzIIrC4

Walmtech, 2025-04-09:
Hi I have bough the same device and just like maloko416 can not get it to work. I notice this has been sorted now with the zigbee-herdsman-converters for Zigbee2MQTT https://github.com/Koenkk/zigbee-herdsman-converters/pull/8640 if that helps anyone ti implement this change. Thanks

Bluicie, 2025-04-10:
I got it working after modifying the already mentioned ZHA Quirk. But this is in no means a final solution. Would need to dive a bit deeper in the how ZHA Quirks actually work to make a proper Quirk.

My updated Quirk:
https://github.com/Bluicie/zha-device-handlers/blob/ts0601_energy_meter_devices/zhaquirks/tuya/ts0601_energy_meter.py