zigpy/zha-device-handlers #3765: [Device Support Request] TS0601_TZE204_ogx8u5z6

Rowleys, 2025-01-23:
### Problem description

The TS0601_TZE204_ogx8u5z6 is not supported in any files I have found.

### Solution description

Hi. Would we be able to add the support for the TS0601_TZE204_ogx8u5z6

Thanks

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

prairiesnpr, 2025-01-23:
It's already supported https://github.com/zigpy/zha-device-handlers/blob/4f64c181d6a98c7c733bccfb6a7e52e47ad9f146/zhaquirks/tuya/ts0601_trv.py#L1920

Rowleys, 2025-01-23:
I looked everywhere .. 

Thanks

closed 2025-01-23 by Rowleys

TheJulianJES, 2025-01-23:
Just noting that you don't need to install any custom quirks. Support will officially be added with Home Assistant Core 2025.2.0.

Rowleys, 2025-01-23:
Where is the repo file. 

Don’t think I have added this repo.

TheJulianJES, 2025-01-23:
All quirks that are in the repos are automatically included in Home Assistant Core. This one will be included with Home Assistant Core 2025.2.0.

You can manually install quirks (as custom quirks), though this is only intended for development and they are likely to break at some point.