zigpy/zha-device-handlers #3782: [Device Support Request] _TZ3000_r6buo8ba TS011F Tyua smart plug -- not showing up in energy panel

peterhalicky, 2025-01-26:
### Problem description

The device isn't showing up in HA Energy panel, most likely because the _summation_delivered state is lacking a kWh unit.

### Solution description

The device should show up in HA Energy panel.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 0,
    "maximum_buffer_size": 80,
    "maximum_incoming_transfer_size": 80,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 80,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0019",
        "0xe001"
      ]
    }
  },
  "manufacturer": "_TZ3000_r6buo8ba",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

Attached as a file; github refused the pasted file contents here due to length exceeding 64k.

</details>

### Logs

### Additional information

Originaly I've reported the issue here: https://github.com/home-assistant/core/issues/113941 and @TheJulianJES asked to create a new issue here based on the diagnostic info I've provided.

[zha-5891f4232a680f344d200a653743f6cf-_TZ3000_r6buo8ba TS011F-0a296d2fb90395c43cf9f7d0f6938782.json](https://github.com/user-attachments/files/18549208/zha-5891f4232a680f344d200a653743f6cf-_TZ3000_r6buo8ba.TS011F-0a296d2fb90395c43cf9f7d0f6938782.json)

TheJulianJES, 2025-01-26:
You can try to download the full file from https://github.com/zigpy/zha-device-handlers/pull/3784/files and use that as a custom quirk until 2025.2.0 is released. Otherwise, wait for the release and it should likely work then.

peterhalicky, 2025-01-26:
I got the file, used it as a custom quirk. I got notifications that the unit has changed for the 3 devices I have and was offered to convert the historical values. So far so good, but the devices are still not showing up in the energy panel :-(

peterhalicky, 2025-01-26:
My bad -- I still have to add the devices to the panel -- and I now can. Problem solved!

TheJulianJES, 2025-01-26:
Awesome! Fix will be included in 2025.2.0 then. Until then, you can keep the custom quirk file installed, but please remove it when the version is out, as it can cause issues in the future otherwise.

closed 2025-01-26 by TheJulianJES