zigpy/zha-device-handlers #3788: [Device Support Request] thermostat Moes BHT-002 GA _TZE204_xalsoe3m

tux1337, 2025-01-26:
### Problem description

I ordered via Moes homepage and Aliexpress some of the BHT-002 GA devices with Zigbee. 

[Moes Shop](https://moeshouse.com/products/zigbee-002-thermostat-for-heating-boiler?variant=39802433011793)
[Aliexpress](https://de.aliexpress.com/item/1005008266328948.html?spm=a2g0o.order_list.order_list_main.5.484f5c5fFvBqZ5&gatewayAdapt=glo2deu)

[image]

Both of them has a new zigbee model ID: _TZE204_xalsoe3m

it looks 100% identical to the old models and have the same features:
TS0601 | _TZE200_aoclfnxz | _TZE200_ztvwu4nk | _TZE200_u9bfwha0

ZHA and HA do not bring entities except RSI and LQI, so its not controllable.

### Solution description

It seems that it needs a new quirk. 

The Zigbee2MQTT project did some work and figured out that its the same than: _TZE204_aoclfnxz
https://github.com/Koenkk/zigbee-herdsman-converters/pull/8628

Therefore I tried to add the new ID to an existing quirk of [_TZE204_aoclfnxz
](https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto.py)
=> this don't work unfortunately. Only Power On / off is working properly and reading the temperature after math adoption. 

Adopted quirk is attached for reference. 

It seems that there is some more work for an quirk for ZHA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
[... 813 lines cut ...]
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [AvattoChildLock],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.CONSUMPTION_AWARENESS_DEVICE,
                INPUT_CLUSTERS: [AvattoTempCalibration],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.CONSUMPTION_AWARENESS_DEVICE,
                INPUT_CLUSTERS: [AvattoDeadzoneTemp],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

</details>

ShogunMan, 2025-05-05:
@tux1337 Have you made any progress on this? I have the same thermostat and its severely limited with current quirk.

tux1337, 2025-05-06:
No I ordered a other device.

ShogunMan, 2025-05-06:
> No I ordered a other device.

What device did you go for?

attilasim, 2025-09-20:
I bought the same device have the same problem.