zigpy/zha-device-handlers #3792: [Device Support Request] Tuya Soil Sensor _TZE284_rqcuwlsa

Nortrix0, 2025-01-27:
### Problem description

This is another variant of the Tuya Soil Sensor, I was able to piece together enough to get the temperature and soil moisture (mostly) working.

Current Issues with the quark I have:
- Soil Moisture shows as Humidity instead of Moisture
- Electric Conductivity does not show at all
- Battery information is not showing and wrong BatteryType

Let me know if any other info is needed and what I can do to help finish this

### Solution description

Help finish the quirk and add it into the tuya_sensor.py

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0408",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_rqcuwlsa",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}

[... 1008 lines cut ...]
    .tuya_temperature(dp_id=5, scale=10)
    #Shows Unknown as 2x AAA instead of the correct 3x AA
    .tuya_battery(dp_id=15) 
    #.tuya_dp(dp_id=15, ep_attribute=TuyaPowerConfigurationCluster3AA.ep_attribute, attribute_name="battery_percentage_remaining",converter=lambda x: {0: 50, 1: 100, 2: 200}[x],)
    
    #Electric Conductivity sensor that is found in the zigbee-herdsman-converters
    #.tuya_number(dp_id=1, attribute_name="ec", type=t.uint16_t, unit="µS/cm",entity_type=EntityType.CONFIG,translation_key="ec", fallback_name="ec",)
    
    #Shows as humidity vs Moisture
    .tuya_soil_moisture(dp_id=3)
    .skip_configuration()
    .add_to_registry()
)

```

</details>

### Additional information

Zigbee-Herdsman-Coverters work:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/46d86e8c7bb2c3f930039b3ae856fce7b6443830/src/devices/neo.ts#L365

Item via Aliexpress:
https://www.aliexpress.us/item/3256808041967058.html?spm=a2g0n.order_detail.order_detail_item.2.6e79f19cm6HtWF&gatewayAdapt=glo2usa

prairiesnpr, 2025-01-27:
It can take a long time for the battery to show. If you reboot does the soil sensor still show as moisture?

Nortrix0, 2025-01-27:
I'm letting it run overnight to see if the battery shows, but yes both reboots of the device and home assistant.

Nortrix0, 2025-01-27:
Battery still shows as Unknown, is there any good documentation of quirks V2 tuya?  I saw some where it's just using .tuya_dp but without guessing what is required and what home assistant would see it as I couldn't find too much on it.

prairiesnpr, 2025-01-27:
I wouldn't say they are good, but check out https://github.com/zigpy/zha-device-handlers/blob/dev/tuya.md and https://github.com/zigpy/zha-device-handlers/wiki/Tuya-%E2%80%90-v2-Quirk-with-TuyaQuirkBuilder

You may need to enchant the device, but that won't be available until 2025.2.

ArqJang, 2025-02-23:
Hey Nortrix0 andpaririesnpr, i stumbled up on this thread because I bought this exact soil sensor from aliexpress and realized today I couldn't simply just add it to my home assistant. Looks like you've found a way to pull some of the useful information from it into HA even if it's not perfect but this is a whole different rabbit hole I have yet to go down. What are the steps I should take to get the temperature and soil "moisture" readings from this sensor? I have not yet dabbled in installing Zigpy on my HomeAssistant yet.. seems like I need to learn that first? Thank you for any assistance you can offer!

ArqJang, 2025-02-23:
Thanks @Nortrix0  I'll give this a shot!

Nortrix0, 2025-03-03:
Looks like it's being added to zha dev

https://github.com/zigpy/zigpy/pull/1544#issue-2814130438
https://github.com/zigpy/zigpy/pull/1543#issue-2814129146

tasomaniac, 2025-04-15:
Thanks a lot for this contribution. I just purchased this device and it was a nice surprise that it's already supported. 

I have one little problem though. The values seem to be updated very less frequently (like in 9 hours in some cases). Is there a way to fix that?

Nortrix0, 2025-04-15:
I've had similar experience in both this and z2m.  Some things like temperature update fine, but others can go hours without anything changing.  I personally haven't looked deeply into it to see if it's a hardware thing, it not wanting to update until the value changes by X amount, or if it needs something else configured.

tasomaniac, 2025-05-16:
It could be that. But maybe not. When I press the button underneath, it updates immediately and the soil moisture level is 2-3% different. So if it internally stores the value and only reports when it is changed, then it should've reported this change.

Also when I put it in water, other Tuya sensors report 100% immediately (or after watering my plants). This sensor does not update immediately after such big changes

closed 2025-11-21 by github-actions[bot]