zigpy/zha-device-handlers #3799: Request support for the following device: zemismart 3P energy meter _TZE284_ny94onlb TS0601

thomo66, 2025-01-28:
### Problem description

Here is the signature for it:

  "data": {
    "ieee": "**REDACTED**",
    "nwk": 38625,
    "manufacturer": "_TZE284_ny94onlb",
    "model": "TS0601",
    "name": "_TZE284_ny94onlb TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4098,
    "power_source": "Mains",
    "lqi": 95,
    "rssi": -82,
    "last_seen": "2025-01-28T17:23:53",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": {
        "logical_type": 1,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 142,
        "manufacturer_code": 4098,
        "maximum_buffer_size": 82,
        "maximum_incoming_transfer_size": 82,
        "server_mask": 11264,
        "maximum_outgoing_transfer_size": 82,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xed00",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE284_ny94onlb",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "update.energie_fewo_firmware",
        "name": "_TZE284_ny94onlb TS0601"
      }
    ],

https://de.aliexpress.com/item/1005006801518635.html?spm=a2g0o.productlist.main.19.11f990yS90yS1O&algo_pvid=d48b39a5-9f24-4e3d-ae4d-7da8bd02fb6f&algo_exp_id=d48b39a5-9f24-4e3d-ae4d-7da8bd02fb6f-9&pdp_npi=4%40dis%21EUR%21102.95%2141.39%21%21%21763.16%21306.83%21%40210391a017380938485937877e5300%2112000038351435930%21sea%21DE%211992006277%21X&curPageLogUid=PZFD3bhGDc3i&utparam-url=scene%3Asearch%7Cquery_from%3A

### Solution description

Please add support for this device either natively or in a quirck.
Thanks!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

thomo66, 2025-01-30:
I got the device working with this work. 
[https://app.codecov.io/gh/zigpy/zha-device-handlers/commit/90dcb4e969c96045ad6c12b09dfdf892de75188b](url)
Of course, I still had to enter my model number. 
Thank you for the fantastic work.

closed 2025-01-30 by thomo66

theveninjeremy, 2025-02-12:
Hi @thomo66 
May I ask you how to use your work ? I'm a complet noob for the quirks. :( 
I got the exact same model tze200_ny94onlb. 

Thanks in advance.

thomo66, 2025-02-12:
Super 👍