zigpy/zha-device-handlers #3810: [BUG] IKEA of Sweden RODRET Dimmer quirk not loaded with firmware 1.0.57

PluueeR, 2025-01-30:
### Bug description

I have two IKEA of Sweden RODRET Dimmers. One with firmware 1.0.47 and 1.0.57. For 1.0.47 the quirk zhaquirks.ikea.twobtnremote.IkeaRodretRemote2Btn is applied. For 1.5.7 no quirk is being applied.

When looking at the signatures for both devices. The class differs.

In the ZHA says the following about de device with 1.0.57:

2025-01-29 22:52:50.016 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden RODRET Dimmer (d4:48:67:ff:fe:fe:f5:7d)
2025-01-29 22:52:50.017 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.twobtnremote.IkeaRodretRemote2Btn'>
2025-01-29 22:52:50.017 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint

In the same debug log for device 1.0.47

2025-01-29 22:52:49.990 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden RODRET Dimmer (6c:fd:22:ff:fe:66:a8:30)
2025-01-29 22:52:49.990 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.twobtnremote.IkeaRodretRemote2Btn'>
2025-01-29 22:52:49.990 DEBUG (MainThread) [zigpy.quirks] Device matches filter signature - device ieee[6c:fd:22:ff:fe:66:a8:30]: filter signature[{'models_info': [('IKEA of Sweden', 'RODRET Dimmer')], 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.NON_COLOR_CONTROLLER: 2080>, 'input_clusters': [0, 1, 3, 32, 4096, 64636], 'output_clusters': [3, 4, 6, 8, 25, 4096]}}}]

### Steps to reproduce

1. Open ZHA integration.
2. Find defected device.
3. Examine the info.
4. Notice no quirk is being applied.

### Expected behavior

Quirk zhaquirks.ikea.twobtnremote.IkeaRodretRemote2Btn is being applied to IKEA of Sweden RODRET Dimmer devices with firmware 1.0.57.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature 1.0.47</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4476,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "RODRET Dimmer",
  "class": "zhaquirks.ikea.twobtnremote.IkeaRodretRemote2Btn"
}

```

</details>
<details><summary>Device signature 1.0.57</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4476,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "RODRET Dimmer",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

MattWestb, 2025-01-30:
The 1.0.57. update have getting more cluster on the endpoint so the first quirk is not matching and ZHA is not loading it.
Look in the signature and finding the different and making one new device class and look if you can getting it loading after restarting HA and i hope all commands used so the Device Automation is working OK.

If getting all working OK pleas so one PR for getting the quirk updated (i using light controller for searing bounded lights / group of lights and have not time for digging deep in it).

PluueeR, 2025-01-30:
@MattWestb Thanks! Somehow I missed that one completely. In my defense, it was already quite late when I looked at it yesterday.

Works like a charm!

MattWestb, 2025-01-31:
Great !!
Can you posting your changed quirk so other user that  have getting there device updated getting the DA working ?
Also if you dont like doing one PR i can doing it so it going in ZHA and dont new custom quirks / patching HA.

PluueeR, 2025-01-31:
[twobtnremote.py.txt](https://github.com/user-attachments/files/18615630/twobtnremote.py.txt)

I just added another class to the file. Don't know if this is the way to go. You can do the PR if you like.

MattWestb, 2025-01-31:
## Great thanks !!!
Have making on PR and adding you for credit !!
I was asking the maintainer if we shall covering the quirk 2 V2 for 2 weeks ago but have no getting any response so going for your V1 is more then OK for me.
Also one good things is not adding the group in on the replacement then if doing it its showing up in ZHA light groupe as one light that is not working with good zigbee controllers (some tuya TS004F must have it then then binding it not working ad tuya standard).

Have testing that  is loading OK and DA is being added on the device card but not testing every DA in automatons but the zha_events looks being OK.

One more thanks for the fix !!

MattWestb, 2025-01-31:
Looks like the dim down resease is changed from only stop to:
```
  endpoint_id: 1
  cluster_id: 8
  command: stop_with_on_off
  args: []
  params: {}
```
So if some like hacking little more for fixing it then its being great.