zigpy/zha-device-handlers #3815: [BUG] Tuya contact sensor quirks are not loaded

tr4nt0r, 2025-01-31:
### Bug description

The V2 quirks for the tuya contact sensors (`_TZE200_pay2byax`) are not loaded in Home Assistant 2025.2.0b2 (zha==0.0.47). Couldn't find any errors in the logs. When copying the quirk to a file in the custom quirks folder it is successfully loads the quirk.

### Steps to reproduce

see above

### Expected behavior

quirk is loaded

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-02-01:
Can you provide the diagnostics file for the contact sensor when the quirk doesn't apply?
Also, enable debug logs for at least `zha`, `zigpy`, and `zhaquirks` in your `configuration.yaml`, then re-start HA and PM me the debug logs:

```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_znp: debug
    zha: debug
    zhaquirks: debug
```

tr4nt0r, 2025-02-03:
@TheJulianJES sent you the logs via discord

tr4nt0r, 2025-02-06:
not reproducible anymore

closed 2025-02-06 by tr4nt0r