zigpy/zha-device-handlers #3818: [BUG] TS0041 TS0044 TS0042 - missing presses and LED flashing

provinzio, 2025-02-01:
### Bug description

TS0041 TS0044 TS0042

sometimes does not react and run associated automation on button press, even thought LED flashes
LED(s) flashing at some interval about once and hour or so even when no button is pressed

### Steps to reproduce

have TS0041 TS0044 TS0042

### Expected behavior

react on press all the time and do not flash leds Lrandomly"

### Device signature

<details><summary>Device signature</summary>

```json

{
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 128,
        "manufacturer_code": 4098,
        "maximum_buffer_size": 82,
        "maximum_incoming_transfer_size": 82,
        "server_mask": 11264,
        "maximum_outgoing_transfer_size": 82,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0006",
          "input_clusters": [
            "0x0000",
            "0x0001",
            "0x0006"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "2": {
          "profile_id": "0x0104",
          "device_type": "0x0006",
          "input_clusters": [
            "0x0001",
            "0x0006"
          ],
          "output_clusters": []
        },
        "3": {
          "profile_id": "0x0104",
          "device_type": "0x0006",
          "input_clusters": [
            "0x0001",
            "0x0006"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "_TZ3000_gbm10jnj",
      "model": "TS0043"
    }

```

</details>

### Diagnostic information

Diagnostic information

[zha-1b4a270315098112f906ebf272ff388d-_TZ3000_gbm10jnj TS0043-47d617482e6562762573c21860a32c41.json](https://github.com/user-attachments/files/18628542/zha-1b4a270315098112f906ebf272ff388d-_TZ3000_gbm10jnj.TS0043-47d617482e6562762573c21860a32c41.json)

### Additional information

Following the issue https://github.com/zigpy/zha-device-handlers/issues/1746, the problem was fixed for a while. But starting after a few months ago, the exact same problem came back.

neilenns, 2025-02-13:
I'm having the same/similar issue. My TS0044 button presses are no longer getting handled by HomeAssistant. The LED lights up when I press the button, but the HomeAssistant automations no longer run and the battery levels report as unknown.

provinzio, 2025-03-31:
I had the issue for months and replaced 2032 button cell batteries with fresh 2025 button cell batteries. It's currently working as expected.

closed 2025-03-31 by provinzio

neilenns, 2025-03-31:
> I had the issue for months and replaced 2032 button cell batteries with fresh 2025 button cell batteries. It's currently working as expected.

In my case the batteries were all fine. I had to remove and re-add the buttons to get things working again.