zigpy/zha-device-handlers #3820: [Device Support Request] TS0601 _TZE284_n4ttsck2

beppenike, 2025-02-01:
### Problem description

I bought this type of sensor but can't get it to work even using the quirk currently around.
i noticed that by searching google for this type of sensor there are no threads about it.
Would anyone be able to help me

Thank you in advance

### Solution description

working quirk file for this device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

prairiesnpr, 2025-02-01:
What is this device? I don't see it listed in z2m, do you have a Tuya hub to capture the data points with?

beppenike, 2025-02-01:
[image]

prairiesnpr, 2025-02-01:
You can try something like this, since we don't have any details on the DPs.

```python
(
    TuyaQuirkBuilder("_TZE284_n4ttsck2", "TS0601")
    .tuya_smoke(dp_id=1)
    .tuya_battery(dp_id=15, battery_type=BatterySize.CR123A, battery_qty=1)
    .skip_configuration()
    .add_to_registry()
)
```

beppenike, 2025-02-01:
i was using this quirk [https://github.com/zigpy/zha-device-handlers/blob/e8a1570203012ba12777fecd18df82ecd1ff4883/zhaquirks/tuya/ts0601_smoke.py](url)

how to integrate your code in it?

prairiesnpr, 2025-02-01:
> i was using this quirk [ts0601_smoke.py](https://github.com/zigpy/zha-device-handlers/blob/e8a1570203012ba12777fecd18df82ecd1ff4883/zhaquirks/tuya/ts0601_smoke.py) How do I integrate your code into it?

Delete that file, create a new file, something like `test_tuya_smoke.py` place above code into that file.

beppenike, 2025-02-01:
[image]
nothing is changed

prairiesnpr, 2025-02-01:
Logs? Quirk isn't loaded.

beppenike, 2025-02-01:
<details><summary>Diagnostic information</summary>

```json

"data": {
    "ieee": "**REDACTED**",
    "nwk": 29172,
    "manufacturer": "_TZE284_n4ttsck2",
    "model": "TS0601",
    "name": "_TZE284_n4ttsck2 TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 168,
    "rssi": -58,
    "last_seen": "2025-02-01T20:44:16",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 128,
        "manufacturer_code": 4417,
        "maximum_buffer_size": 66,
        "maximum_incoming_transfer_size": 66,
        "server_mask": 10752,
        "maximum_outgoing_transfer_size": 66,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xed00",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE284_n4ttsck2",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "update.tze284_n4ttsck2_ts0601_firmware",
        "name": "_TZE284_n4ttsck2 TS0601"
      },
      {
        "entity_id": "update.sf_camino_firmware",
        "name": "_TZE284_n4ttsck2 TS0601"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "SF camino",
    "device_reg_id": "0c589f216d47a40f1a6e6a0541402ea2",
    "area_id": null,
    "cluster_details": {
```

I think that quirk is loaded ( i have another device that works with it in another quirk conf.)

prairiesnpr, 2025-02-01:
> "quirk_applied": false,
>     "quirk_class": "zigpy.device.Device",

It's not matching, turn on debug logging, reload ZHA and then see why it didn't match the quirk.

beppenike, 2025-02-01:
```
Logger: zhaquirks
Source: /usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py:481
First occurred: 21:16:43 (1 occurrences)
Last logged: 21:16:43

Unexpected exception importing custom quirk 'ts0601_smoke'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_smoke.py", line 2, in <module>
    from zhaquirks.tuya import BatterySize
ImportError: cannot import name 'BatterySize' from 'zhaquirks.tuya' (/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/__init__.py)
```

prairiesnpr, 2025-02-01:
Ahh, that's part of 2025.2, we can remove it for now.

```python
from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE284_n4ttsck2", "TS0601")
    .tuya_smoke(dp_id=1)
    .tuya_battery(dp_id=15)
    .skip_configuration()
    .add_to_registry()
)
```

beppenike, 2025-02-01:
OMG it's working!!!

[image]

prairiesnpr, 2025-02-01:
Turn on debug logging and see if you have any messages about unknown data points for that device. Sometimes it takes awhile for the battery to check in, but could also be the wrong DP. Also, verify that HA shows alarm when triggered.

beppenike, 2025-02-01:
OK it's working perfectly now. Thank you so much mate <3

[image]

PR #3823 (2025-02-06): Add Tuya smoke detector `_TZE284_n4ttsck2`
## Proposed change

Add Tuya smoke detector `_TZE284_n4ttsck2`
Closes: #3820 

## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-02-06 by TheJulianJES