zigpy/zha-device-handlers #3830: [Device Support Request] Philips Hue Wall Switch RDM001

cbornemann, 2025-02-05:
### Problem description

Signature for Philips Hue Wall Switch RDM001 is not correct.

### Solution description

Custom Quirk work for RDM001 and RDM004.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4107,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 128,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 128,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0830",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "RDM001",
  "class": "wall_switch.PhilipsWallSwitch"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

"""Signify wall switch devices (RDM001 and RDM004)."""

from typing import Final

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PowerConfiguration,
)
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    RIGHT,
    SHORT_PRESS,
    SHORT_RELEASE,
    TURN_ON,
)
from zhaquirks.philips import (
    PHILIPS,
    SIGNIFY,
    Button,
    PhilipsBasicCluster,
    PhilipsRemoteCluster,
    PressType,
)

class SwitchMode(t.enum8):
    """Wall switch modes. See https://zigbee.blakadder.com/Philips_RDM001.html."""

    SingleRocker = 0x00
    SinglePush = 0x01
    DoubleRocker = 0x02
    DoublePush = 0x03

class PhilipsWallSwitchBasicCluster(PhilipsBasicCluster):
    """Philips wall switch Basic cluster."""

    class AttributeDefs(PhilipsBasicCluster.AttributeDefs):
        """Attribute definitions."""

        mode: Final = ZCLAttributeDef(
            id=0x0034,
            type=SwitchMode,
            is_manufacturer_specific=True,
        )

    attr_config = {
        **PhilipsBasicCluster.attr_config,
        AttributeDefs.mode.id: SwitchMode.DoublePush,
    }

class PhilipsWallSwitchRemoteCluster(PhilipsRemoteCluster):
    """Philips wall switch remote cluster."""

    BUTTONS = {
        1: Button(LEFT, TURN_ON),
        2: Button(RIGHT),
    }

    PRESS_TYPES: dict[int, PressType] = {
        1: PressType(LONG_PRESS, "hold"),
        3: PressType(LONG_RELEASE, "long_release", "hold_release"),
    }

    SIMULATE_SHORT_EVENTS = [
        PressType(SHORT_PRESS, "press"),
        PressType(SHORT_RELEASE, "short_release"),
    ]

class PhilipsWallSwitch(CustomDevice):
    """Philips RDM001 or RDM004 device."""

    signature = {
        MODELS_INFO: [
            (PHILIPS, "RDM001"),
            (SIGNIFY, "RDM001"),
            (PHILIPS, "RDM004"),
            (SIGNIFY, "RDM004"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.NON_COLOR_SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PhilipsWallSwitchRemoteCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.NON_COLOR_SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    PhilipsWallSwitchBasicCluster,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PhilipsWallSwitchRemoteCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    device_automation_triggers = (
        PhilipsWallSwitchRemoteCluster.generate_device_automation_triggers()
    )

```

</details>

nhunzaker, 2025-05-14:
Confirming that this is working for me also with my RDM001 by Signify Netherlands B.V. devices. Thanks!

tundratier, 2025-05-23:
Can you describe in detail how you got this working? I've tried it using the custom_quirks thing in HOAS, but it does not seem to be loading them at all. How can I verify that I did setup them correctly?

nhunzaker, 2025-05-24:
@tundratier here are my notes:

Just to summarize - from my understanding, the Hue switch adapter module supports up to two switches but it uses a non-standard method of doing so. The Zigbee for Home Assistant add-on handles this through its quirks feature, but there's a bug in the implementation for the RDM001 unit. 

I had to explicitly enable quirks mode and specify a directory for additional quirks:

```yaml
# configuration.yaml
zha:
  enable_quirks: true
  # Important! I have a home assistant green and homeassistant is an alias of this config directory. I had to use config and could not use the homeassistant directory.
  custom_quirks_path: /config/zha_quirks/
```

Now, paste the quirk python module above into that directory. Something like `/config/zha_quirks/wall_switch_custom.py`. The name doesn't seem to matter.

After restarting Home Assistant and re-adding the devices, the quirk loads! You'll see it in the device info:

[image]

I just had to set the mode for the switch to `SwitchMode.DualRocker` (from the modes [here](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/philips/wall_switch.py#L44-L50)) to get it to behave like two separate switches:

[image]

I seem to have to do this part pretty soon after pairing, otherwise ZHA will fail to write the value. I'm not sure why.
 
One other weird thing - the left switch is labeled as "turn on" while the right switch is labeled "right button". I suspect this is just bad nomenclature. Either way, it works now.

Hope it helps!

closed 2025-08-24 by TheJulianJES