zigpy/zha-device-handlers #3831: [BUG] Battery not supported for lumi.motion.agl04 while implemented in a quirk

rrooggiieerr, 2025-02-06:
### Bug description

I have an Aqara motion sensor:
[image]
https://www.aliexpress.com/item/1005006461768670.html

Which should is supported by ZHA:
[image]

Unfortunately the battery status is not working:
[image]

In the same time the code seems to be in place to support the battery:
https://github.com/zigpy/zha-device-handlers/blob/8b3560ee6d9cc105df7b698cec021ceacf14af1f/zhaquirks/xiaomi/aqara/motion_agl04.py#L63-L113

### Steps to reproduce

See description

### Expected behavior

The battery sensor should show the battery state.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-02-06:
Did it populate after a few hours?

rrooggiieerr, 2025-02-06:
No, it doesn't

TheJulianJES, 2025-02-06:
What firmware are you running on the Raspbee II coordinator?
Some newer Xiaomi devices require Xiaomi manufacturer codes during pairing to unlock all functionality. Newer coordinator firmware sends this.

Otherwise, could be an issue in the quirk as well.

rrooggiieerr, 2025-02-07:
Interesting! I just paired the device with an other HA instance I have running which uses a SONOFF Zigbee 3.0 USB Dongle Plus V2 and there the Battery percentage is correctly handled. And I now also got a device temperature!

Only thing is that the battery size and quantity is incorrect. The device uses 2 CR2450 batteries.
[image]

The quirk states CR1632, so that's an error in the quirk and a quick fix, however the quirk also states `battery_quantity = 2`, so there's something going wrong there.

rrooggiieerr, 2025-02-07:
I'm using firmware 0x26690700 on the Raspbee II coordinator. That's what it came with, but there is a newer version available

rrooggiieerr, 2025-02-07:
Updating the RaspBeeII firmware to 0x26780700 (latest) indeed solved the problem with the battery state. Battery size and quantity still incorrect.

rrooggiieerr, 2025-02-10:
After updating the firmware of my Raspbee II coordinator the problems with the battery sensor were resolved, however an other issue came up. The sensor now creates false triggers, about every 55 minutes:

[image]

Running on HA 2025.2.1

TheJulianJES, 2025-02-15:
We might need to implement something like this: https://github.com/zigpy/zha-device-handlers/pull/2779/files

TheJulianJES, 2025-03-26:
Hmm, we had a similar fix we reverted in https://github.com/zigpy/zha-device-handlers/pull/3091.
Would be really good to get some feedback on if the above fix works for you. We might need to reintroduce the fix again and tell those people to upgrade firmware + re-pair the device possibly.

whalemare, 2025-09-24:
[image]

Exactly the same issue with the same sensor `lumi.motion.agl04`, but it emits false positive value each 45 minutes.

45 min "off"
10 min "on"

```
IEEE: 54:ef:44:10:00:6d:b7:17
Nwk: 0xe415
Device Type: EndDevice
LQI: 236
RSSI: -41
Last seen: 2025-09-24T19:01:38
Power source: Battery or Unknown
Quirk: zhaquirks.xiaomi.aqara.motion_agl04.LumiLumiMotionAgl04
```

Signature
```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 127,
    "maximum_incoming_transfer_size": 100,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 100,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0406",
        "0x0500",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.agl04",
  "class": "zhaquirks.xiaomi.aqara.motion_agl04.LumiLumiMotionAgl04"
}
```

Video:
https://github.com/user-attachments/assets/697d9659-7c2a-4648-9a7b-f9bdfb52b16e

whalemare, 2025-09-24:
I've add custom quirk and apply patch from [your comment](https://github.com/zigpy/zha-device-handlers/issues/3831#issuecomment-2660625000),

after that Home Assistant stopping displaying any updates from sensor. Once I revert it, everything starts working as it was before

ge33ek, 2026-10-10:
Hi All,

i've made a fix for this issue on firmware 0x0000001b specifically, i didn't have this issue pre-upgrade. I ran it in debug and identified that the problem could be mitigated with a custom quirk.

I've  tested it now for over a week and haven't had any misfires. I have my sensitivity set to "low" in the config for ZHA. Occupancy remains after PIR clear for 600 seconds, this can be customised if needed, but have left it as the default for now as it served my particular purpose which was just to ensure correct firing. Hope it helps.

Needs to be in your custom quirks folder with custom quirks enabled in configuration.yaml - will show as "Quirk: motion_agl04:LumiLumiMotionAgl04HeartbeatFiltered" in the Zigbee info of the device if applied correctly after reboot.

```
"""Experimental selective heartbeat filter for LUMI lumi.motion.agl04.

For the built-in quirk shipped with Home Assistant 2026.10 / zha-quirks 2.3.0.
An occupied report is held for 0.5 seconds. It is discarded only if BOTH
manufacturer status attributes 0x00F7 and 0x00DF arrive during that window.
Battery and configuration reports continue through the built-in handler.

This pattern matches the debug logs i gathered in troubleshooting the issue; it is not an upstream-approved fix.
Real motion coinciding with that status sequence could also be discarded.
"""

from __future__ import annotations

import asyncio
from copy import deepcopy
import logging

from zigpy.zcl import foundation

from zhaquirks.const import ENDPOINTS, INPUT_CLUSTERS
from zhaquirks.xiaomi import OccupancyCluster
from zhaquirks.xiaomi.aqara.motion_agl04 import (
    LumiLumiMotionAgl04 as BuiltinMotionAgl04,
    OppleCluster,
)

_LOGGER = logging.getLogger(__name__)
FILTER_WINDOW_SECONDS = 0.5
STATUS_ATTRIBUTES = frozenset({0x00F7, 0x00DF})

class HeartbeatFilteredOccupancyCluster(OccupancyCluster):
    """Hold occupied reports until the accompanying messages are known."""

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._pending_report = None
        self._pending_handle = None
        self._status_attributes = set()

    def _cancel_pending(self):
        if self._pending_handle is not None:
            self._pending_handle.cancel()
        self._pending_handle = None
        self._pending_report = None
        self._status_attributes.clear()

    def _release_pending(self):
        report = self._pending_report
        self._cancel_pending()
        if report is not None:
            hdr, args, kwargs = report
            _LOGGER.debug(
                "%s: accepted standalone occupancy report",
                self.endpoint.device.ieee,
            )
            super().handle_cluster_general_request(hdr, args, **kwargs)

    def status_attributes_reported(self, attribute_ids):
        """Discard a pending occupied report only after the complete pair."""
        if self._pending_report is None:
            return
        self._status_attributes.update(STATUS_ATTRIBUTES.intersection(attribute_ids))
        if self._status_attributes == STATUS_ATTRIBUTES:
            _LOGGER.debug(
                "%s: suppressed occupancy report paired with heartbeat/status messages",
                self.endpoint.device.ieee,
            )
            self._cancel_pending()

    def handle_cluster_general_request(self, hdr, args, **kwargs):
        if hdr.command_id != foundation.GeneralCommand.Report_Attributes:
            return super().handle_cluster_general_request(hdr, args, **kwargs)

        occupancy_reports = [
            attr for attr in args.attribute_reports if attr.attrid == 0x0000
        ]
        if not occupancy_reports:
            return super().handle_cluster_general_request(hdr, args, **kwargs)

        self._cancel_pending()
        if not (int(occupancy_reports[-1].value.value) & 1):
            return super().handle_cluster_general_request(hdr, args, **kwargs)

        self._pending_report = (hdr, args, kwargs)
        self._pending_handle = asyncio.get_running_loop().call_later(
            FILTER_WINDOW_SECONDS, self._release_pending
        )

class HeartbeatAwareOppleCluster(OppleCluster):
    """Identify the status pair while preserving built-in Xiaomi parsing."""

    def handle_cluster_general_request(self, hdr, args, **kwargs):
        if (
            hdr.command_id == foundation.GeneralCommand.Report_Attributes
            and hdr.manufacturer == 0x115F
        ):
            self.endpoint.occupancy.status_attributes_reported(
                {attr.attrid for attr in args.attribute_reports}
            )
        return super().handle_cluster_general_request(hdr, args, **kwargs)

class LumiLumiMotionAgl04HeartbeatFiltered(BuiltinMotionAgl04):
    """Use the built-in device definition with selective motion filtering."""

    replacement = deepcopy(BuiltinMotionAgl04.replacement)
    replacement[ENDPOINTS][1][INPUT_CLUSTERS] = [
        (
            HeartbeatFilteredOccupancyCluster
            if cluster is OccupancyCluster
            else HeartbeatAwareOppleCluster
            if cluster is OppleCluster
            else cluster
        )
        for cluster in BuiltinMotionAgl04.replacement[ENDPOINTS][1][INPUT_CLUSTERS]
    ]
```