zigpy/zha-device-handlers #3834: [BUG] Saswell TRV _TZE200_c88teujp no longer able to change heating mode

vifino, 2025-02-07:
### Bug description

I upgraded my HA from 2025.1.2 to 2025.2.0 and my two Saswell TRVs lost function.

I previously renamed the "_thermostat" entity to remove the double "thermostat_thermostat". That entity got lost in the process, a new "thermostat_thermostat" entity appeared.

When changing the heating mode from HA, nothing happens. Setting it on the thermostat itself shows up in HA though.

The new integration shows more knobs and sensors, but I'm not sure they work. The new HVAC action and Valve Position sensors are unavailable.

Removing the device and repairing did not help.

### Steps to reproduce

1. Have _TZE200_c88teujp TRVs on 2025.1.2
2. Upgrade to 2025.2.0
3. Try to change heating modes
4. Make a sad face as you watch nothing happen

### Expected behavior

The heating mode should change as desired.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_c88teujp",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
```

</details>

### Diagnostic information
[... 952 lines cut ...]
                "value": null
              }
            },
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

prairiesnpr, 2025-02-07:
Please enable debug logging in ZHA then repeat your testing and send me the log file.

vifino, 2025-02-07:
Hey @prairiesnpr!

I had to shorten it down to fit the original issue comment, but in the end it didn't fit anyway.
Here's the gist: https://gist.github.com/vifino/8e2374a40e51d805a32d07909d41f249

Hope it helps!

prairiesnpr, 2025-02-08:
Please make a custom quirk using `tuya_trv.py` and change https://github.com/zigpy/zha-device-handlers/blob/8b92f6925527f18faac9a973218395bce510d72b/zhaquirks/tuya/tuya_trv.py#L128 to 

```python
    .tuya_dp(
        dp_id=101,
        ep_attribute=TuyaThermostatV2.ep_attribute,
        attribute_name=TuyaThermostatV2.AttributeDefs.system_mode.name,
        converter=lambda x: Thermostat.SystemMode.Heat if x == 0x01 else Thermostat.SystemMode.Off,
        dp_converter=lambda x: 0x01 if x == Thermostat.SystemMode.Heat else 0x00,
    )
```

And let me know the results.

vifino, 2025-02-08:
I downloaded the `tuya_trv.py` to the custom quirks path, it seems to be loaded judging from the `__pycache__`.
Changing heating modes still doesn't seem to work.

prairiesnpr, 2025-02-08:
Does HA reflect the state correctly or is it inverted?

vifino, 2025-02-08:
The state in HA is correctly updated when I manually turn on or off the TRV with the physical button.

TheJulianJES, 2025-02-11:
If you change the TRV mode to "OFF" with physical buttons, does OFF to HEAT using HA work then?

TheJulianJES, 2025-02-12:
Also, please try the custom quirk linked in this comment: https://github.com/zigpy/zha-device-handlers/issues/3837#issuecomment-2652472292

vifino, 2025-02-12:
@TheJulianJES No, off to heat using HA also does not work.

vifino, 2025-02-12:
@TheJulianJES I've tried the custom quirk you linked, it still does not work.
The thermostat entities are duplicated (again), off -> heat and heat -> off from HA are still non-functional.

prairiesnpr, 2025-02-12:
Do the other functions work? Can you change set point?

vifino, 2025-02-12:
Yes, I can change the set point once HA and the TRV agree that the TRV is set to heating. Before and after the quirk update.

prairiesnpr, 2025-02-12:
@vifino Is the run state correct? I have another report of the same quirk, but different model saying it's inverted.

vifino, 2025-02-12:
If run state is off/heating, then no, it is not inverted.

prairiesnpr, 2025-02-12:
> If run state is off/heating, then no, it is not inverted.

It actually should say `Heat` or `Idle`. 

[image]

TheJulianJES, 2025-02-12:
We'd love some debug logs from the v1 quirk, where everything is working. Can you do the following?

Use the old v1 quirk from here: https://github.com/zigpy/zha-device-handlers/blob/eda673936ae01ea50fc7806e3fc8ae57fa3432a1/zhaquirks/tuya/ts0601_trv_sas.py

Since the loading order is "custom v2 quirks > built-in v2 quirks > custom v1 quirks > built-in v1 quirks", we need to stop the built-in v2 quirk from loading by adding the following code at the top of the v1 custom quirk file:

```python
from zigpy.quirks import DEVICE_REGISTRY
# other existing imports from custom quirks

# remove upstream quirk from registry
del DEVICE_REGISTRY._registry_v2[("_TZE200_c88teujp", "TS0601")]

# add custom quirk...
```

Then, enable debug logs. They can be enabled on the integrations page -> ZHA -> enable debug logs. Disabling them downloads the entire `home-assistant.log` file to your PC.
We just need the section(s) when you're changing the system mode (via v1 quirk and ideally also via the updated v2 quirk linked above that still doesn't work).
If you don't feel comfortable sharing them here, you can send them to me via email or Discord (or to prairiesnpr).

TheJulianJES, 2025-02-12:
Actually, prairiesnpr seems to have found the issue. Please try this custom v2 quirk first and see if changing `sytem_mode` works now: https://github.com/prairiesnpr/zha-device-handlers/blob/patch-saswell-trv/zhaquirks/tuya/tuya_trv.py

prairiesnpr, 2025-02-13:
Valve position may report in later, that was added based on the z2m converter. HVAC action is odd, that's from ZHA itself and in thisc case should just be based off the running state.

vifino, 2025-02-13:
Still not there. The TRV also seems to randomly reset to 16 degrees for no reason?

TheJulianJES, 2025-02-13:
Does that happen if "Away mode" is on?

vifino, 2025-02-14:
If I turn on Away Mode, I can't control temperature without it turning off again.
While it's on, it shows "LEAV" on the LCD.

No, it seems to just always reset to 16.0 degrees when I turn it from off -> heat.

TheJulianJES, 2025-02-14:
We might need to explicitly set `schedule` to off when writing `system_mode`?

Z2M:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/c4499e662d4c72d0d151f82262e3b9ab5b7a21c4/src/lib/legacy.ts#L5711-L5714

old ZHA PR:
https://github.com/zigpy/zha-device-handlers/pull/1816

TheJulianJES, 2025-02-14:
Does the "schedule" switch toggle on when changing system mode? Does it make a difference if you toggle it once or twice after setting the system mode?

TheJulianJES, 2025-02-15:
On 2025.2.4 (important to use this version), please try the following as a custom quirk: https://github.com/prairiesnpr/zha-device-handlers/blob/patch-_TZE200_c88teujp/zhaquirks/tuya/tuya_trv.py

vifino, 2025-02-15:
Hey!
This updated quirk seems to fix the issue I was having.
Do you still need me to provide any logs?

Thanks!

closed 2025-02-15 by vifino

skumka, 2025-02-21:
When the fixes will be available on official update? Looks like one from Feb 14th does not include it yet.

PR #3853 (2025-02-13): Fix Saswell TRV local temperature calibration and system mode
## Proposed change

This ensures that the off value for system mode is sent as an enum. It also uses a number for local temperature calibration and sets the correct range and step values.

## Additional information

Closes: #3837 
Closes: #3834

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
PR #3867 (2025-02-16): Turn off schedule on `system_mode` change for `_TZE200_c88teujp`
## Proposed change

Sends command to turn schedule off whenever `system_mode` is set.

Closes: #3834

## Additional information

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works