zigpy/zha-device-handlers #3837: [BUG] Tuya TS0601 TRV temperature offset not exposed after 2025.2.0

zolakt, 2025-02-07:
### Bug description

I have bunch of Tuya TRVs, TS0601 by _TZE200_c88teujp.

Now they are loading this quirk, although I don't think they used the same one before.
Quirk: zigpy.quirks.v2.CustomDeviceV2

Previously they exposed a temperature offset numeric input, which is no longer there.
There are a bunch one new sensors like frost/windows/scale protection, but the temperature offset it gone.
This breaks the integration with BetterThermostat

### Steps to reproduce

No special reproduction steps. The old entity is gone. It shows "This entity is no longer being provided by the zha integration. If the entity is no longer in use, delete it in settings."

### Expected behavior

TRVs should expose temperature offset. This is critical functionality

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_c88teujp",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
[... 955 lines cut ...]
              }
            },
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

zolakt, 2025-02-07:
Actually, it even more weird. A few of them have a new control for the offset. But most of them don't. All of these are the same devices.

[image]

[image]

zolakt, 2025-02-07:
I managed to solve it by manually writting the value of `local_temperature_calibration` attribute, and reloading ZHA. After that the missing controls appearead. The value was initialy reading `None`. I've written it to `0`

[image]

[image]

zolakt, 2025-02-07:
However this doesn't work so great, especially with BetterThermostat. The previous offset slider (in the control section) was going from -6/+6C (or possibly -5/+5C, I don't remember). This new one (in configuration section) goes from -2.5/+2.5C. That is way to low and basically unusable for TRVs. 2.5 offset it literally nothing. I can set it to a bigger value via "Manage devices", but BetterThermostat still caps it to 2.5...

Does anyone have a solution for this? To increase the range of the offset. With a custom quirk or something... Seriously there should have been a braking change warning for this in the latest release... This messed up my whole heating in the middle of the winter. If I had knows wouldn't have updated. You really shouldn't mess with TRVs during winter, in general

prairiesnpr, 2025-02-08:
> goes from -2.5/+2.5C

That appears to be the same for all thermostats in ZHA. https://github.com/zigpy/zha/blob/d6bb4a89ff857137e96b2932f711f1bdb2fbe80b/zha/application/platforms/number/__init__.py#L840

zolakt, 2025-02-08:
Just to verify that I'm not hallucinating, I connected one of the TRVs to Tuya directly. Calibration goes from -6/+6C, which I'm pretty sure I had in ZHA previously as well. So this -2.5/+2.5C limitation seems to be purely on ZHA. Maybe this was introduces with v2 quirks?

Does anyone have a workaround for this?

prairiesnpr, 2025-02-08:
I understand how this would be frustrating, I am in the process of rewriting all of the Tuya quirks as v2 quirks, in order to simplify the code base, increase existing support, and bring support for more Tuya devices.

As far as your issue, you have a few quick options. you can revert to the previous version of HA, replace the built-in v2 quirk with the older v1 quirk as a custom quirk, or update the existing v2 quirk using a custom quirk so that it uses a number entity instead.

You could comment out https://github.com/zigpy/zha-device-handlers/blob/8b92f6925527f18faac9a973218395bce510d72b/zhaquirks/tuya/tuya_trv.py#L115 and replace it with something like the following.

```python
    .tuya_number(
        dp_id=27,
        attribute_name="local_temperature_calibration_custom",
        type=t.uint32_t,
        min_value=-6,
        max_value=6,
        unit=UnitOfTemperature.CELSIUS,
        step=1,
        translation_key="local_temperature_calibration",
        fallback_name="Local temperature calibration",
    )
```

Untested, but should provide a separate number entity. No idea if it will work with better thermostat.

We need to understand if the correction should be in ZHA, should the range be increased for all thermostats or should that be set per quirk?

zolakt, 2025-02-08:
Thanks for the reply. As for the solutions:

1. Unfortunately I don't have a backup of the previous HA version, so not sure if I can roll back safely
2. I haven't used a custom quirk for these TRVs previously, it was working with the built-in one. I'll try to find the old one here. Do you have a link maybe?
3. I'll give it a try, but I'm not sure if this will work. How would BetterThermostat be able to conclude that the custom number input is for temperature calibration. I think I would at least need to name the attribute "local_temperature_calibration", not "local_temperature_calibration_custom"

Please don't take my venting personally, but this is the thing that pisses me off about HA the most, by far. Fixed release cycles of totally untested releases (or tested on 1 out of million device) just for the sake of making a release. If something is "in the process" and "we would need to understand"... then you don't release it until the process and the understanding is done. People will gladly rather wait for a stable release, than having half of the devices fall apart every few months. I've been burned by this way too many times, especially with ZHA. That being said, I'm a complete idiot for even installing a x.x.0 release, absolutely never have those gone smoothly

zolakt, 2025-02-08:
On top of existing issues, I just found out that the HVAC mode doesn't do anything anymore... I turn it off, the TRV doesn't close... Just great...

prairiesnpr, 2025-02-08:
I recommend you revert to a previous version of HA, discord should be able to walk you through that. I'm traveling this weekend and will be unable to look at this further until next week.

This PR was available for review for several weeks, reviewed by maintainers, and then part of the beta release. While I understand that this can be frustrating, your expectation that we should purchase every Tuya device made to ensure there are never breaking changes or never make any changes to the code base is unreasonable. We include unit tests where applicable, but that doesn't catch everything.

Ensure any missing functionality is listed and be available for testing and I'll do what I can to help resolve this.

zolakt, 2025-02-08:
I don't expect you to purchase every Tuya device, but I do expect you to test it on at least one, before rolling out a relase. We are not talking about a little irrelevant bug, this is completely broken funadamental functionality. Offset got broken, switching the mode got broken, loading custom quirks doesn't work... Seriously, this wasn't rocket science to detect, if it was tested at all. So what was exactly the point of this (obviously unfinished) release, to roll out half-finished code and hope for the best?

I'm a developer myself, and honestly I don't understand how you guys get aways with this. Pretty much no one else would. Also, there is a thing called "proffesional pride", that should prevent you from rolling out untested releases. You are really, continuosuly, shooting yourself in the foot with these fixed monthly cycle releases. From a professional standpoint that is just ridicolous. A feature is finished when it has been tests, not just because the calendar says it's a new month

prairiesnpr, 2025-02-08:
If you have the experience and the device, probably easier and faster if you just make the changes. Ping me when your PR is in and I'll be happy to review it.

zolakt, 2025-02-08:
I can change the offset range and test that. I'll let you know.

But for the broken HVAC mode, I have no idea. I'm comparing the old and the new quirk, and so far I can't find where that change is. Anyway, right now. changing Auto/Off is completely broken. The TRV doesn't react to it. It only reacts to the target temperature, regardless if HA displays the TRV as on or off. Can you please have a look at that? I can test it. I was working before, I'm 100% sure

zolakt, 2025-02-08:
Btw. the fix with `tuya_number` that you suggested [here](https://github.com/zigpy/zha-device-handlers/issues/3837#issuecomment-2645509035) does not work.

It display both sliders, even though I deleted that old `tuya_dp` call. One goes -2.5/+2.5C the other -6/+6C. They are totally independent from each other, changing the value of one of them does not change the value of the other. I very much doubt the -6/+6 is actually affecting the TRV. Also, even though the slider is capped at -6C, for some TRVs I can see it's set to -25C when I open the chart. I very much doubt BetterThermostat will work with this.

I also tried removing all other mappings of `local_temperature_calibration`, even for other device models. To see if one of those was maybe leeking in, and creating that -2.5/+2.5C slider. It didn't help... There are still 2 sliders

zolakt, 2025-02-09:
For now, I've loaded the old v1 quirk, which works without issues, same as before:

- the calibration slider goes from -6C to +6C, with 1C steps (same as in Tuya)
- HVAC mode Heat/Off actually opens/closes the TRV

I've used the quirk that @prairiesnpr has suggested [here](https://github.com/zigpy/zha-device-handlers/issues/3837#issuecomment-2645675064). You need to add these 2 lines at the beginning, (as suggested by @TheJulianJES [here](https://github.com/zigpy/zha-device-handlers/issues/3841#issuecomment-2645930050)). Possibly change the model in the registry call, if you have a different one.
```
from zigpy.quirks import DEVICE_REGISTRY
del DEVICE_REGISTRY._registry_v2[("_TZE200_c88teujp", "TS0601")]
```
From what I tested so far, it works exactly as it did before. I don't have the new entities that were added in v2, like frost/scale/window protection, but honestly I don't need any of those.

If you guys don't have the devices, and need help testing the fixes for the v2 quirk, let me know. I have around 15 of these TRVs, in various setups and automations, so I can test it pretty thoroughly.

zolakt, 2025-02-09:
I mean, seriosly, this is bordering with insanity... I honestly don't see how can you claim, with a straight face, that this has been tested at all....

Out of pure desperation, I went into `.storage/core.entity_registry` and deleted all of these entities that were added by the v2 quirk. Save... restart... and... they are back... added back to the `core.entity_registry` file... As if there aren't enough problems with these things not working at all, you can even get rid of them...

puddly, 2025-02-09:
@zolakt I want to remind you that quirks are community-contributed and supported by people in their free time. [Your tone earlier](https://github.com/zigpy/zha-device-handlers/issues/3837#issuecomment-2645869429) was unprofessional and not acceptable here.

We're working on tracking down what we can do to fix this issue and what testing infrastructure can be added to prevent this from happening again. If you'd like to participate in the discussion, do so professionally.

TheJulianJES, 2025-02-09:
To address most of what was mentioned in this thread:

> this is the thing that pisses me off about HA the most, by far. Fixed release cycles of totally untested releases (or tested on 1 out of million device) just for the sake of making a release. If [...]

It is absolutely untrue that our releases are completely untested. Of course, we don't merge any quirks where we think they're not ready or that they'll cause issues. Most quirks and devices are tested. But with Tuya devices, it's impossible for us to test every single variation.

ZHA library and zha-quirks versions are released independently from Home Assistant. If something is not done, we either don't merge that particular PR yet or don't bump the lib version in Home Assistant if a released library version is considered unstable. It'll just go in the next major HA version, when we think it's ready and bug free.

But if there are no testers available to test new or updated quirks, the code won't become "more stable" just because we wait a few months before merging a specific PR or releasing a new feature.

> I've been burned by this way too many times, especially with ZHA.
> That being said, I'm a complete idiot for even installing a x.x.0 release, absolutely never have those gone smoothly
> Unfortunately I don't have a backup of the previous HA version, so not sure if I can roll back safely

So, no .0 release ever worked correctly for you before and you still don't have backups set up? I don't really see how you can fault us or HA's release schedule for that then.

Backups are essential to Home Assistant, especially if you completely rely on it for your heating. There's a guide in the "Getting Started" documentation: https://www.home-assistant.io/getting-started/configuration/#creating-a-backup

2025.1 and 2025.2 also added a simpler backup strategy. You should really set it up now if you still haven't: https://www.home-assistant.io/blog/2025/01/03/3-2-1-backup/

> I don't expect you to purchase every Tuya device, but I do expect you to test it on at least one, before rolling out a relase. We are not talking about a little irrelevant bug, this is completely [...]

I think you're underestimating how complex the interaction between Zigbee/ZCL, ZHA, zha-quirks, and Home Assistant is.

ZHA supports devices that follow ZCL spec by default. No quirks are required and they are generally pretty stable across releases. We mention these are devices that are well supported and should be used in the ZHA docs.

Tuya devices, however, go out of their way to detect non-Tuya hubs to block functionality. They also implemented their on protocol on top of ZCL. This is why the issue tracker on this repo is mostly about Tuya devices, see [here](https://github.com/zigpy/zha-device-handlers/issues?q=sort%3Aupdated-desc%20is%3Aissue%20is%3Aopen%20label%3ATuya). And there are a lot of other issues that are just not marked as "Tuya" issues yet.

Major bugs get spotted and fixed during the beta period. This issue wasn't found because not a lot of people have this exact variation of a Tuya TRV apparently and other similar Tuya TRVs were tested and work just fine.
We can't have all Tuya devices. Most are just variations with different firmware. We ask other users to test new quirks for us then. This went well for most quirks added/updated in the 0.0.131 release, just look at [the 0.0.131 changelog](https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.131). Most of these quirks seem to work just fine.

However, some users just miss bugs during testing or do not face them in the first place, sometimes simply because they're using a different coordinator.

Tuya TRVs in particular are very complex, so it's unfortunate that this one doesn't work correctly now. But again, others do though. There are multiple underlying issues that came together for this device to experience these issues.

There will always be issues with particular devices, hopefully less in the future. We're working on v2 quirks and getting up the test coverage a lot.
The v1 quirk for this device was added about 4 to 5 years ago and did not have any test coverage. This makes it a lot harder for us to modernize the codebase, since we need to rely on others to test these devices (since you can't even buy most variations anymore).

> I'm a developer myself, and honestly I don't understand how you guys get away with this. Pretty much no one else would. 

You can just not use Home Assistant nor ZHA then? Home Assistant is open source and relies on community contributions for most integrations. Clearly, it's working well enough for Home Assistant to end up where it did.

> Also, there is a thing called "proffesional pride", that should prevent you from rolling out untested releases. You are continuosuly shooting yourself in the foot with these fixed monthly cycle releases. From a professional standpoint that is just ridicolous. A feature is finished when it has been tested, not just because the calendar says it's a new month.

The great thing about monthly major releases is that we can just wait another month or two if a feature isn't ready.
If HA and ZHA only released a new version every year, there'd be a lot of unsupported devices during that period, and if you can upgrade eventually, you would face way more bugs, as a lot of new functionality and changed code suddenly gets to you at once.
But if you prefer this, just upgrade HA every 6 months for example, and skip `.0` releases.

[... cut ...]
Instead, we should modify the original ZHA entity or remove that via quirks v2 and add a custom v2 entity with the Tuya limits.

There are some issues with this particular TRV model that need to be fixed in the v2 quirk, but now that you're using the old v1 quirk as a custom quirk, what doesn't work correctly at the moment?
If the ZHA entity really bothers you that much, just don't use it or disable it (via the UI).

So, do you still have any actual issues with your current setup you need help with?

zolakt, 2025-02-10:
I apologize for the tone, it was impulsively written. I mean no harm. I do appreciate the work everyone has put in, and I know things are complex with Tuya. In my defense, this is probably the 3-4th time I had massive breaking of a large number of Tuya devices like this, so it does seem things could be better tested there, before they roll out. For some things you can't expect production users to be a testing environment. It's not the end of the world if e.g. a motion sensor gets messed up for a few days, but it is kind of a big deal to have the heating system break in the middle of winter. Same for the betas, even if I did participate, I can't really switch these devices over to a beta env, and have a non functioning heating system while I play around.

That being said, the mess I'm in, and the frustration, is entirely my fault, for jumping on the update button right away, and somehow missing to take a backup, which I usually do. I read the HA release article, didn't see anything alarming, and just went ahead. It would be helpful if major, breaking, updates like this would be mentioned there. Usually (if anything) it just says "bump zha to xx.xx.xx"

----------------------------------------------

Getting back to the problem. I'll summarize everything and try to keep it short:

Before the update I had:

- offset slider that goes from -6/+6C
- HVAC mode Heat/Off was opening/closing the TRV
- no controls for min/max setpoint
- no -2.5/+2.5C slider

After the update:

- the old -6/+6C slider was unavailable, orphaned, and I deleted it manually
- the new -2.5/+2.5C slider was added (on some devices only after writing the calibration value manually, through "manage device")
- min/max setpoint controls were added
- other entities were added like: window/scale/freeze protection
- HVAC mode Heat/Off is NOT opening/closing the TRV

After loading the v1 quirk:
- the -6/+6C slider got added again, and is working
- HVAC mode Heat/Off works again
- most of the entities added by v2 are unavailable, orphaned, and I can delete them
- I can NOT delete these 3 entities: -2.5/+2.5 slider, min and max set point inputs. They are not unavailable, they look like they are available and operational. I tried purging from the UI, and (out of desparation) messing with the entity_registry file. Nothing seems to remove them, they get added back again or reboot. I had the similar problem when I tried patching up the v2 quirk previously. The -2.5/+2.5C just kept getting added back, even though I removed it from the quirk. Right now I have all 3 of them disabled, since that seems like the only thing I can do

prairiesnpr, 2025-02-12:
@zolakt Spent some time going through this, we think it may be related to the manufacturer id. Please capture logs of you changing the system mode and set point, both in the v1 and v2 quirks. Once we have those, we should be able to dig further.

TheJulianJES, 2025-02-12:
Yep, debug logs would be really helpful. They can be enabled on the integrations page -> ZHA -> enable debug logs. Disabling them downloads the entire `home-assistant.log` file to your PC.
We just need the section(s) when you're changing the system mode (via v1 quirk and ideally also via the v2 quirk).
If you don't feel comfortable sharing them here, you can send them to me via email or Discord (or to prairiesnpr).

For the local temp calibration slider, we'll either remove the default (2.5°C) ZHA entity for this device via a future version (either by preventing discovery in 2025.3 or by modifying the attribute cache in 2025.2), or it has to be removed manually by deleting the device from ZHA and then re-pairing it.

TheJulianJES, 2025-02-12:
Here's an updated v2 quirk: https://github.com/prairiesnpr/zha-device-handlers/blob/patch-saswell-trv/zhaquirks/tuya/tuya_trv.py

Please try and see if that solves any issues. It should add a proper "local temperature calibration" entity (though the default ZHA one will still show for now too).
The heating mode may or may not be fixed, we're not sure yet. Please test this.

For the debug log comments above, it's enough to grab the logs from the v1 quirk (so we can add some proper unit tests for this) and from the linked v2 quirk installed as a custom quirk. It should override the built-in v2 quirk automatically, though you can also add the ["registry removal"](https://github.com/zigpy/zha-device-handlers/issues/3841#issuecomment-2645930050) at the top of the file if you really want to be sure.
I'll hopefully come around to adding a way for HA to show which v2 `quirk_file` is used in the future (similar to v1 quirk classes).

zolakt, 2025-02-12:
Hi, thanks for the quick fixes. I didn't have time today. I'll test the new quirk and send logs first thing tomorrow morning

TheJulianJES, 2025-02-12:
Actually, prairiesnpr might have found the issue. Please try this custom v2 quirk first and see if changing `sytem_mode` works now: https://github.com/prairiesnpr/zha-device-handlers/blob/patch-saswell-trv/zhaquirks/tuya/tuya_trv.py

closed 2025-02-13 by TheJulianJES

zolakt, 2025-02-13:
I loaded the quirk you linked. Switching Heat/Off does work, however there are still some weird things, so I'm not sure if  I'm in some frankenstein state from going back and forth.

When I swicthed back to the v1 quirk last time, the v2 entities (window/scale/frost protection, set points etc) bacame unavailable and I deleted them. Now with the new v2 quirk, they came back, but they are all unavailable. Some of the v1 entities seem to be back, and some of v2 also back (but unavailable). e.g. battery used to be a percentage in v1. Then a binary sensor in v2. Now I see a working percentage entity, and an unavailable binary sensor.

Also, the new slider (this is now the 3rd one, the v1 is unavailable, the old v2 one disabled) does not work with BetterThermostat. I assume it's missing some attribute for it to pick it up. Both previous sliders got recognised.

[image]

[image]

prairiesnpr, 2025-02-13:
I think you would be best served to remove and the repair the device with just the v2 quirk, that should remove all the orphaned entities. On the local temp calibration, we will have to discuss, the way it was supported in v1 wasn't ideal.

TheJulianJES, 2025-02-13:
Better Thermostat uses a hacky way to get the "local temperature calibration" entity: https://github.com/KartoffelToby/better_thermostat/blob/6385a794d4edbf1625f44680917f0e02f2754ad0/custom_components/better_thermostat/utils/helpers.py#L297

Do you still have multiple entities present? Please try to just delete the TRV from HA and then re-pair. It should completely delete the other entities.
Better Thermostat just stops at the first match, it seems, even if it's deactivated or unavailable.

PR #3853 (2025-02-13): Fix Saswell TRV local temperature calibration and system mode
## Proposed change

This ensures that the off value for system mode is sent as an enum. It also uses a number for local temperature calibration and sets the correct range and step values.

## Additional information

Closes: #3837 
Closes: #3834

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works