zigpy/zha-device-handlers #3838: [BUG] TS0601 by _TZE204_xnbkhhdr (Heating setpoint from HA is not reflected/updated in the local device)

cibermike20, 2025-02-07:
### Bug description

Hi everyone,
With the last update 2025.2 my Avatto thermostat stopped working properly.

I've been using a custom quirk found here for weeks: https://github.com/zigpy/zha-device-handlers/issues/3326#issuecomment-2625532689

With that custom quirk I could modify the heating setpoint of the thermostat from the HA interface using a control "Thermostat", and the value was set on the local device immediately.

However, since the last update with native support, after removing the custom quirk and pairing again the device, when I modify the heating setpoint this value is NOT reflected on the thermostat. If I set it from the local device, the value is updated on the HA thermostat control... Can you please check or tell me if I'am doing something wrong?

Note: It's not a signal issue as I do not get a message as so and I use it close to my zigbee coordinator.

Thanks!

### Steps to reproduce

1. Modify the heating setpoint in the thermostat control in HA.
2. After a few seconds the value showed in HA shows again the value that appeared in the local device, not being updated in the local one.

### Expected behavior

The value set from the HA panel updates the value on the local device (allowing to change its status from idle to heating and viceversa).

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

delaosa, 2025-02-07:
I've just created  #3839

AlastairMutch, 2025-02-11:
The Avatto _TZE204_lzriup1j thermostat has exactly the same problem with 2025.2.2 (and .1) using the native support now added. I can change the set point in HA but it has no effect on the physical thermostat. The HA displayed value gets updated from the thermostat if you change it on the physical thermostat front panel. 

It was working ok prior to 2025.2.x with a custom quirk. 

Alastair

TheJulianJES, 2025-02-13:
Should be fixed by https://github.com/zigpy/zha-device-handlers/pull/3860, should be included in the next HA patch release.
Thanks for the initial PR and thanks for the issue report!

closed 2025-02-13 by TheJulianJES