zigpy/zha-device-handlers #3841: [BUG] Huge delays on Tuya TS0601 _TZE200_3towulqd motion sensor after 2025.2

zolakt, 2025-02-07:
### Bug description

I have a  Tuya TS0601 by _TZE200_3towulqd motion/illumination sensor. Before 2025.2 it worked just fine, pretty quick. After upgrading I'm having gigantic delays, to the point that it's unusable. I use it to turn on a light. And it's has around 30s delays... Does anyone have an idea what has changed?

### Steps to reproduce

Nothing special, I keep looking at the device, and walk by the sensor. It switches to "occupied" after 20-30s

### Expected behavior

There shouldn't be more than 1s delay, like it was before the update

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0400",
        "0x0406",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_3towulqd",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}

```

</details>

### Diagnostic information
[... 523 lines cut ...]
                "attribute": "ZCLAttributeDef(id=0x0071, name='battery_3_size', type=<enum 'BatterySize'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0076": {
                "attribute": "ZCLAttributeDef(id=0x0076, name='battery_3_volt_min_thres', type=<class 'zigpy.types.basic.uint8_t'>, zcl_type=<DataTypeId.uint8: 32>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0077": {
                "attribute": "ZCLAttributeDef(id=0x0077, name='battery_3_volt_thres1', type=<class 'zigpy.

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

zolakt, 2025-02-08:
I managed to figure something out. It's not actually a delay. The state of the sensor got flipped. It shows Detected when Clear, and vice versa. So it actually turns on the light when the sensor gets Clear.

Does anyone have an idea how to flip this? I really don't want to do automations with inversed logic

prairiesnpr, 2025-02-08:
See https://github.com/zigpy/zha-device-handlers/blob/8b92f6925527f18faac9a973218395bce510d72b/zhaquirks/tuya/tuya_motion.py#L1124

Try `converter=lambda x: x == 0,`

zolakt, 2025-02-08:
Thanks @prairiesnpr. I'll give it a try. Just to double check, I should basically clone that quirk, make the change, and load it as a custom quirk, right? And not mess with the original file directly

Actually, I have another question. I was using a custom quirk (v1) for this device previously, and it was working fine. But now it seems like it doesn't use it anymore, even though the quirk is for that exact device model. Now it just says "Quirk: zigpy.quirks.v2.CustomDeviceV2".

This is the quirk I was using before. Is there any way to force it use it again?

```
import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl import foundation

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Sensitivity level enum."""

    _10_SEC = 0x00
    _30_SEC = 0x01
    _60_SEC = 0x02
    _120_SEC = 0x03

class PirMotionManufCluster(TuyaMCUCluster):
    """Neo manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF0A: ("keep_time", OnTimeValues)})

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "sensitivity_level",
            converter=lambda x: SensitivityLevel(x),
        ),
        10: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "keep_time",
            converter=lambda x: OnTimeValues(x),
        ),
        12: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        10: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [("_TZE200_3towulqd", "TS0601")],
        ENDPOINTS: {
            # endpoints=1 profile=260 device_type=0x0402
            # in_clusters=[0x0000, 0x0001, 0x0500],
            # out_clusters=[0x000a, 0x0019]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    TuyaOccupancySensing,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

prairiesnpr, 2025-02-08:
> I should basically clone that quirk, make the change, and load it as a custom quirk, right? And not mess with the original file directly

This would depend on your install method, I use a venv so I never use custom quirks, just edit the file directly. The usual option is to clone the file and configure custom quirks.

On the v1 quirk not loading anymore, I'm not aware of any changes there, would expect a custom quirk to override a built in one even if it's a v1 overriding a v2, but as I said, I've not tested any of that.

zolakt, 2025-02-08:
Custom quirks don't seem to work at all... Even v2 ones.
I've cloned the quirk, modified that if statement, restarted... It's not loading it.
It is still showing "Quirk: zigpy.quirks.v2.CustomDeviceV2".

I have custom quirks enabled, and before this latest update, all of the custom ones were working fine.

```
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/
```

I've put the modified v2 quirk in `/config/custom_zha_quirks/custom_tuya_motion.py`

prairiesnpr, 2025-02-08:
> 
> UPDATE: Maybe it does work but it just doesn't show the name of the quirk on the device page? My sensor values are now inverted correctly. I hope it's the quirk, and that it didn't just randomly flip

All v2 quirks will show `zigpy.quirks.v2.CustomDeviceV2`

TheJulianJES, 2025-02-08:
The order for quirk loading is as following: custom v2 quirks > v2 quirk > custom v1 quirk > v1 quirks.
You can still override built-in v2 quirks with custom v1 quirks by removing the v2 quirk from the v2 registry in your v1 quirk:
```python
from zigpy.quirks import DEVICE_REGISTRY
# other existing imports from custom quirks

# remove upstream quirk from registry
del DEVICE_REGISTRY._registry_v2[("Signify Netherlands B.V.", "SOC001")]

# add custom quirk...
```

But custom quirks are only intended for development and testing, not for permanent use. They will always break sooner or later.
The built-in v2 quirk should be fixed.

Due to how v2 quirks work, they will currently always show `zigpy.quirks.v2.CustomDeviceV2`. This might change in the future.

zolakt, 2025-02-08:
@TheJulianJES thanks for the reply. If I remove the v2 quirk from the registry, will it get added back once I stop using the v1 quirk?

TheJulianJES, 2025-02-08:
Yes, since the v2 quirk is removed from the device registry in the custom v1 quirk, the v2 quirk will be "re-added" if the custom v1 quirk is removed. The device registry is constructed when ZHA is initialized.

But again, I'd highly recommend fixing the v2 quirk instead of using an old v1 custom quirk.
Did the fix [from the comment above](https://github.com/zigpy/zha-device-handlers/issues/3841#issuecomment-2645513257) work for the v2 quirk? (then loading the modified quirk as a custom v2 quirk)

zolakt, 2025-02-08:
Yes, for this particular problem, the fix suggested above did work.

But I have a much bigger problem with a different device, where I need to bring the v1 quirk back, since the v2 one pretty much broke the whole thing. https://github.com/zigpy/zha-device-handlers/issues/3837

PR #3858 (2025-02-13): Invert occupancy for Tuya `_TZE200_3towulqd`
## Proposed change

Inverts occupancy for 1_TZE200_3towulqd`.

Closes: #3841
## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-02-13 by TheJulianJES