zigpy/zha-device-handlers #3844: [BUG] Humidity, temp "unknown" from "lumi.weather" sensor

ralfbergs, 2025-02-09:
### Bug description

The device is continuously reporting the air pressure, as well as battery, LQI, RSSI. The latter indicate good network (but it shouldn't matter anyway, as the air pressure and battery values **are** received continuously!).

**However**, "humidity" and "temperature" are not displayed at all, they show up as "unknown" in HA. :(

I'm using this device in ZHA, HA 2025.2.1. My coordinator is a SLZB-06M, "Core" firmware 2.7.1, ZB FW 20241127.

### Steps to reproduce

1. Add the sensor to the ZB network.
2. Observe that the air pressure is continuously reported, while humidity and temp are **not** reported.

### Expected behavior

Humidity and temp should be received at the same time as air pressure values are received.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4151,
    "maximum_buffer_size": 127,
    "maximum_incoming_transfer_size": 100,
    "server_mask": 0,
    "maximum_outgoing_transfer_size": 100,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0403",
        "0x0405",
        "0xffff"
      ],
      "output_clusters": [
        "0x0000",
        "0x0004",
        "0xffff"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.weather",
  "class": "zhaquirks.xiaomi.aqara.weather.Weather2"
}
[... 933 lines cut ...]
            "attributes": {},
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

I'm currently collecting debug logs, which I will submit later here in this place. I just would like to send this "draft" now already to avoid my efforts in populating this page so far being lost by some accident.

```python

[Paste the logs here]

```

</details>

ralfbergs, 2025-02-09:
I really have no clue about ZB comms, but the above DEBUG message of "Frame contains trailing data: b'\x04'" sounds to me as if the binary data in the frame might not be properly parsed, and therefore the temp and humidity values cannot be properly extracted.

I'd be extremely grateful if someone more knowledgeable in this area than myself could have a look. I'm of course available to support any investigations, but I need some guidance as I only started using HA a couple of weeks ago.

Many thanks in advance for your support.

ralfbergs, 2025-02-09:
More examples of invalid metrics:

Log for temp, value not ok (but no suspicious frame issues!):
```python
2025-02-09 14:07:52.330 DEBUG (MainThread) [bellows.ash] Received data 5408b1a97d312a15b658964e24ab5592499cbce7677d38459874f5de6e83fc7d5e165733936b7e
2025-02-09 14:07:52.330 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=5, re_tx=0, ack_num=4, ezsp_frame=b'J\x90\x01E\x00\x00\x04\x01\x02\x04\x01\x01\x00\x00\x00\x00\xf2\xc0\xcc\xf5\x8b\xff\xff\x08\x18\r\n\x00\x00)\xf0\xd8')
2025-02-09 14:07:52.330 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=6) + FLAG
2025-02-09 14:07:52.331 DEBUG (MainThread) [bellows.ash] Sending data  8610be7e
2025-02-09 14:07:52.333 DEBUG (MainThread) [bellows.ezsp.protocol] Received command incomingMessageHandler: {'type': <EmberIncomingMessageType.INCOMING_UNICAST: 0>, 'apsFrame': EmberApsFrame(profileId=260, clusterId=1026, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=242), 'lastHopLqi': 192, 'lastHopRssi': -52, 'sender': 0x8BF5, 'bindingIndex': 255, 'addressIndex': 255, 'messageContents': b'\x18\r\n\x00\x00)\xf0\xd8'}
2025-02-09 14:07:52.334 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1026, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=242), 192, -52, 0x8BF5, 255, 255, b'\x18\r\n\x00\x00)\xf0\xd8']
2025-02-09 14:07:52.335 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 2, 9, 13, 7, 52, 335782, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8BF5), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=242, profile_id=260, cluster_id=1026, data=Serialized[b'\x18\r\n\x00\x00)\xf0\xd8'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=192, rssi=-52)
2025-02-09 14:07:52.337 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0402] Received ZCL frame: b'\x18\r\n\x00\x00)\xf0\xd8'
2025-02-09 14:07:52.338 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0402] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=13, command_id=10, *direction=<Direction.Server_to_Client: 1>)
2025-02-09 14:07:52.339 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0402] Decoded ZCL frame: TemperatureMeasurementCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=-10000))])
2025-02-09 14:07:52.340 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0402] Received command 0x0A (TSN 13): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=-10000))])
2025-02-09 14:07:52.341 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0402] Attribute report received: measured_value=-10000
```

Log for humidity, value not ok (duplicate frame!):
```python
2025-02-09 14:07:52.348 DEBUG (MainThread) [bellows.ash] Received data 6408b1a97d312a15b658904124ab5593499c4a9b601d839874efce7383f97b1e43ebc5db4e8fffccdefcd26952327e
2025-02-09 14:07:52.348 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=6, re_tx=0, ack_num=4, ezsp_frame=b'J\x90\x01E\x00\x00\x04\x01\x04\x0b\x01\x01\x00\x01\x00\x00\x04\xbc\xcb\xf0M\xff\xff\x12\x08\x10\n\x05\x05!\xe4\x00\x08\x05!\x00\x00\x0b\x05)\x00\x00')
2025-02-09 14:07:52.348 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=7) + FLAG
2025-02-09 14:07:52.349 DEBUG (MainThread) [bellows.ash] Sending data  87009f7e
2025-02-09 14:07:52.350 DEBUG (MainThread) [bellows.ezsp.protocol] Received command incomingMessageHandler: {'type': <EmberIncomingMessageType.INCOMING_UNICAST: 0>, 'apsFrame': EmberApsFrame(profileId=260, clusterId=2820, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=4), 'lastHopLqi': 188, 'lastHopRssi': -53, 'sender': 0x4DF0, 'bindingIndex': 255, 'addressIndex': 255, 'messageContents': b'\x08\x10\n\x05\x05!\xe4\x00\x08\x05!\x00\x00\x0b\x05)\x00\x00'}
2025-02-09 14:07:52.350 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=2820, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=4), 188, -53, 0x4DF0, 255, 255, b'\x08\x10\n\x05\x05!\xe4\x00\x08\x05!\x00\x00\x0b\x05)\x00\x00']
2025-02-09 14:07:52.351 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 2, 9, 13, 7, 52, 350978, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4DF0), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=4, profile_id=260, cluster_id=2820, data=Serialized[b'\x08\x10\n\x05\x05!\xe4\x00\x08\x05!\x00\x00\x0b\x05)\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=188, rssi=-53)
[... cut ...]
2025-02-09 14:07:52.357 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0405] Decoded ZCL frame: RelativeHumidityCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=65436))])
2025-02-09 14:07:52.357 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0405] Received command 0x0A (TSN 14): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=65436))])
2025-02-09 14:07:52.359 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0405] Attribute report received: measured_value=65436
```

ralfbergs, 2025-02-09:
Example of successful metrics:

<details><summary>Logs for RSSI, LQI, battery, pressure (all values are ok, no "Frame contains trailing data" or "duplicate frame"!)</summary>

```python
2025-02-09 13:52:24.035 DEBUG (MainThread) [bellows.ash] Received data 0044b1a97d312a15b658944a24ab5592499cbfef657d38459874d1de6f83fd817d5d82eaec6d648bde6f98d0f31b1f4007a8ec723ba58e5c7299fd6dbaecd7cb90b51d0e0d9a798a60b37e
2025-02-09 13:52:24.035 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=0, re_tx=0, ack_num=0, ezsp_frame=b'\x06\x90\x01E\x00\x00\x04\x01\x00\x00\x01\x01\x00\x00\x00\x00\xf1\xc8\xce\xf5\x8b\xff\xff,\x18\x0c\n\x01\xffB%\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@')
2025-02-09 13:52:24.035 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=1) + FLAG
2025-02-09 13:52:24.035 DEBUG (MainThread) [bellows.ash] Sending data  8160597e
2025-02-09 13:52:24.037 DEBUG (MainThread) [bellows.ezsp.protocol] Received command incomingMessageHandler: {'type': <EmberIncomingMessageType.INCOMING_UNICAST: 0>, 'apsFrame': EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=241), 'lastHopLqi': 200, 'lastHopRssi': -50, 'sender': 0x8BF5, 'bindingIndex': 255, 'addressIndex': 255, 'messageContents': b'\x18\x0c\n\x01\xffB%\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@'}
2025-02-09 13:52:24.037 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=241), 200, -50, 0x8BF5, 255, 255, b'\x18\x0c\n\x01\xffB%\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@']
2025-02-09 13:52:24.037 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2025, 2, 9, 12, 52, 24, 37521, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8BF5), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=241, profile_id=260, cluster_id=0, data=Serialized[b'\x18\x0c\n\x01\xffB%\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=200, rssi=-50)
2025-02-09 13:52:24.038 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0000] Received ZCL frame: b'\x18\x0c\n\x01\xffA%\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@'
2025-02-09 13:52:24.039 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=12, command_id=10, *direction=<Direction.Server_to_Client: 1>)
2025-02-09 13:52:24.039 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0000] Decoded ZCL frame: BasicCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0xFF01, value=TypeValue(type=LVBytes, value=b'\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@'))])
2025-02-09 13:52:24.039 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0000] Received command 0x0A (TSN 12): Report_Attributes(attribute_reports=[Attribute(attrid=0xFF01, value=TypeValue(type=LVBytes, value=b'\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@'))])
2025-02-09 13:52:24.040 DEBUG (MainThread) [zigpy.zcl] [0x8BF5:1:0x0000] Attribute report received: 0xFF01=b'\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@'
2025-02-09 13:52:24.040 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x8BF5:1:0x0000]: cluster_handler[basic] attribute_updated - cluster[BasicCluster] attr[65281] value[b'\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@']
2025-02-09 13:52:24.040 DEBUG (MainThread) [zha] Emitting event cluster_handler_attribute_updated with data ClusterAttributeUpdatedEvent(attribute_id=65281, attribute_name=65281, attribute_value=b'\x01!\xb3\x0b\x04!\xa8C\x05!r\x93\x06$\x01\x00\x04\x00\x00d)\xf0\xd8e!\x9c\xfff+\xe0\x8d\x01\x00\n!\x9c@', cluster_handler_unique_id='00:15:8d:00:03:6b:d3:9d:1:0x0000', cluster_id=0, event_type='cluster_handler_event', event='cluster_handler_attribute_updated') (2 listeners)
2025-02-09 13:52:24.040 DEBUG (MainThread) [zha] Emitting event state_changed with data EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SENSOR: 'sensor'>, unique_id='00:15:8d:00:03:6b:d3:9d-1-0-rssi', device_ieee=00:15:8d:00:03:6b:d3:9d, endpoint_id=1, group_id=None) (1 listeners)
2025-02-09 13:52:24.040 DEBUG (MainThread) [homeassistant.components.zha.entity] sensor.wetter_rssi: Handling event from entity: EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SENSOR: 'sensor'>, unique_id='00:15:8d:00:03:6b:d3:9d-1-0-rssi', device_i [... cut ...]
2025-02-09 13:52:24.044 DEBUG (MainThread) [zha] Emitting event state_changed with data EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SENSOR: 'sensor'>, unique_id='00:15:8d:00:03:6b:d3:9d-1-1027', device_ieee=00:15:8d:00:03:6b:d3:9d, endpoint_id=1, group_id=None) (1 listeners)
2025-02-09 13:52:24.044 DEBUG (MainThread) [homeassistant.components.zha.entity] sensor.wetter_pressure: Handling event from entity: EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SENSOR: 'sensor'>, unique_id='00:15:8d:00:03:6b:d3:9d-1-1027', device_ieee=00:15:8d:00:03:6b:d3:9d, endpoint_id=1, group_id=None)
```
</details>

MattWestb, 2025-02-09:
I have one with the same symptom but i was on the balcony so its little damage of the humidity so likely its one hardware problem so i have not looking more.
After your post i was looking more and i have some first gen and all is working OK but i also having one V2 in my sleeping room and its also working OK so i was wandering if its somthing in ZHA code that have changing and the balcony is added later so have different data setting then being initiated.
Then looking on the quirk side is not changed for around 2 years so shall not being any problem but it can being changes in zigpy / ZHA that making things failing (like python version used) but im not shore.
So for my it can being hardware or update that have making 1/2 version 2 not working OK.

Ping @TheJulianJES Do you having one lumi weather v2 for verifying it working OK ??
I cant saying its software related then one of my is working OK and the other have stop working around the same time as @ralfbergs

ralfbergs, 2025-02-09:
> I have one with the same symptom but i was on the balcony so its little damage of the humidity

It's interesting that you mention this, because my device was also outside for a while -- although it was never directly hit by rain, because it was under a protective cover. I wonder whether I should open it and check it out?

Anyway, maybe someone experienced enough with low-level Zigbee protocol issues can have a look at my logs and find out whether this is anything wrong with parsing the values returned by the device...

MattWestb, 2025-02-09:
I was having one V1 on the balcony with little protection and was working very well until moving it  being unprotected then renovating the underground and was getting direct rain on it and it was completely corroded inside = hardware brick.
The second one (V2) have being outside for little more then one year and was having the same protecting and the position but was living shorter.

They is not very easy open without braking them but take out the battery and then and have 6 point (2 on the upper and lower side and 1 on each side) being hold but its looks best starting on the upper side on the side of the pairing button and then its little loos then on the same side but in the middle and then the under then upper other side and all shall falling in part with or without damages holding points.

The humidity and temperature have the sensor on the small "head" of PCB that is the most exposed and also the soldering points of the sensor.

So very likely the sensor is list and the device can only sending pressure and battery status then no communicating the TH-chip.
Photo V1 little damaged then opening :

[image]

ralfbergs, 2025-02-09:
> They is not very easy open without braking them 

Thanks, Matt.

I just came back from opening it up when I saw your update. ;-)

I didn't even take out the battery. Just used a kitchen knife and brute force to spread the two case parts apart -- virtually no damage to the case or anything else. ;-)

This is how it looked like... Pretty much corroded, eh?

[image]

I've now cleaned it up a bit, and it looks **much** better already:

[image]

Unfortunately that didn't help anything -- I still only get the air pressure...

I really wonder whether it's the corrosion, or whether it's not some bad interpretation of Zigbee protocol data... Especially a temp sensor should be pretty robust in my understanding... 🤔

Anyway, thanks for your help, and let's see if we find some protocol experts here who are willing to look into the case...

MattWestb, 2025-02-09:
### Great !!!
Looks like my first V1 before i was cleaning it in alcohol and drying it on low temperature in the owen.

The blue / green is copper that have being oxidated and with little current running around it making it galvanic move or the copper.

The barometer is the metallic "thing" very near TP21 and its working.
The TH sensor is on the other side and on the small head and shall have 4 connection on 2 sides. 
Adafrutis version: https://www.adafruit.com/product/5064
All connection must being OK or the MCU cant reading the data from it or the sensor is damaged inside.

MattWestb, 2025-02-09:
One other module with pinout and circuit connection for getting it working.
https://www.adafruit.com/product/5064