zigpy/zha-device-handlers #3845: [BUG] _TZE204_upagmta9 does not receive temperature/humidity/battery information, but can set display unit

bleuthoot-sven, 2025-02-09:
### Bug description

When setting up the sensor in Home Assistant, the temperature sensor, humidity sensor and battery sensor show up in Home Assistant, but don't receive any data. I am able to set the temperature display of the device and it changes accordingly.

### Steps to reproduce

1. Connect the device to Home Assistant using the extension
2. Wait a few hours
3. Unknown status does not go away

### Expected behavior

For the temperature/humidity/battery to give a value

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Device signature</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.2.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.1",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.6.62-haos-raspi",
    "supervisor": "2025.02.0",
    "host_os": "Home Assistant OS 14.2",
    "docker_version": "27.2.0",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "sonoff": {
      "documentation": "https://github.com/AlexxIT/SonoffLAN",
      "version": "3.8.2",
      "requirements": [
[... 902 lines cut ...]
              "0x0000": {
                "attribute": "ZCLAttributeDef(id=0x0000, name='time', type=<class 'zigpy.types.named.UTCTime'>, zcl_type=<DataTypeId.UTC: 226>, access=<ZCLAttributeAccess.Read|Write_Optional: 5>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0001": {
                "attribute": "ZCLAttributeDef(id=0x0001, name='time_status', type=<flag 'TimeStatus'>, zcl_type=<DataTypeId.map8: 24>, access=<ZCLAttributeAccess.Read|Write_Optional: 5>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0002": {
                "attribute": "ZCLAttributeDef(id=0x0002, name='time_zone', type=<class 'zigpy.types.basic.int32s'>, zcl_type=<DataTypeId.int32: 43>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0009": {
                "attribute": "ZCLAttributeDef(id=0x0009, name='valid_until_time', type=<class 'zigpy.types.named.UTCTime'>, zcl_type=<DataTypeId.UTC: 226>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}
```

prairiesnpr, 2025-02-10:
This has been removed and then added again? It needs to apply the enchantment to work and that's done on wehn paired.

bleuthoot-sven, 2025-02-10:
> This has been removed and then added again?

I did remove and add it a few times, with a few hours in between the re-adding.

> It needs to apply the enchantment to work and that's done when paired.

I don't exactly get what you mean by that. This is basically what I did:

1. Put the device in paring mode
2. Go to the "Add device" page for Zigbee
3. Wait for the device to be paired and give it a name
4. Wait about an hour, hoping it will update the value.
5. See that the device status didn't get updated, so I remove the device from the Zigbee extension device list and attempted it again.

I only did this about 2 - 3 times.

prairiesnpr, 2025-02-12:
Do you see any traffic in the logs from the device? Looking for reports of temperature, humidity, etc.

bleuthoot-sven, 2025-02-15:
> Do you see any traffic in the logs from the device? Looking for reports of temperature, humidity, etc.

These are some potential relevant logs I could find form the time I tested:

<details>
<summary>Logs</summary>

```
2025-02-08 12:53:16.167 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0019]: 'async_configure' stage succeeded
2025-02-08 12:53:16.167 DEBUG (MainThread) [zha] Emitting event zha_channel_cfg_done with data ClusterHandlerConfigurationComplete(device_ieee=a4:c1:38:65:c9:ac:4b:dc, unique_id=a4:c1:38:65:c9:ac:4b:dc, event_type='zha_channel_message', event='zha_channel_cfg_done') (0 listeners)
2025-02-08 12:53:16.167 DEBUG (MainThread) [zha.zigbee.device] [0x2BC9](TS0601): completed configuration
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.device] [0x2BC9](TS0601): started initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:ZDO](TS0601): 'async_initialize' stage succeeded
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0001]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0402]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0000]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0405]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0xef00]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0019]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0019]: Skipping cluster handler initialization
2025-02-08 12:53:16.168 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0001]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0402]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0000]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0405]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0xef00]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0019]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x2BC9:1:0x0019]: 'async_initialize' stage succeeded
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.device] [0x2BC9](TS0601): power source: Battery or Unknown
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.zigbee.device] [0x2BC9](TS0601): completed initialization
2025-02-08 12:53:16.169 DEBUG (MainThread) [zha.application.gateway] Platform entity data: EnumSelectInfo(fallback_name='Display unit', unique_id='a4:c1:38:65:c9:ac:4b:dc-1-display_unit', platform=<Platform.SELECT: 'select'>, class_name='ZCLEnumSelectEntity', translation_key='display_unit', device_class=None, state_class=None, entity_category=<EntityCategory.CONFIG: 'config'>, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='TuyaClusterHandler', generic_id='cluster_handler_0xef00', endpoint_id=1, cluster=ClusterInfo(id=61184, name='Tuya Manufacturer Specific', type='server', commands=[ZCLCommandDef(id=0x25, name='mcu_connection_status_rsp', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.mcu_connection_status_rsp'>, is_manufacturer_specific=True), ZCLCommandDef(id=0x03, name='query_data', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.query_data'>, is_manufacturer_specific=True), ZCLCommandDef(id=0x04, name='send_data', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.send_data'>, is_manufacturer_specific=True), ZCLCommandDef(id=0x00, name='set_data', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.set_data'>, is_manufacturer_specific=True), ZCLCommandDef(id=0x24, name='set_time', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.set_time'>, is_manufacturer_specific=True)]), id='1:0xef00', unique_id='a4:c1:38:65:c9:ac:4b:dc:1:0xef00', status='INITIALIZED', value_attribute=None)], device_ieee=a4:c1:38:65:c9:ac:4b:dc, endpoint_id=1, available=True, group_id=None, enum='TuyaTempUnitConvert', options=['Celsius', 'Fahrenheit'])
2025-02-08 12:53:16.170 DEBUG (MainThread) [zha.application.gateway] Platform entity data: SensorEntityInfo(fallback_name=None, unique_id='a4:c1:38:65:c9:ac:4b:dc-1-1', platform=<Platform.SENSOR: 'sensor'>, class_name='Battery', translation_key=None, device_class=<SensorDeviceClass.BATTERY: 'battery'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='PowerConfigurationClusterHandler', generic_id='cluster_handler_0x0001', endpoint_id=1, cluster=ClusterInfo(id=1, name='Power Configuration', type='server', commands=[]), id='1:0x0001', unique_id='a4:c1:38:65:c9:ac:4b:dc:1: [... cut ...]
    await self.context_manager.__aexit__(exc_type, exc, traceback)
  File "/usr/local/lib/python3.13/site-packages/zigpy/datastructures.py", line 184, in __aexit__
    self.release()
    ~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/datastructures.py", line 160, in release
    raise ValueError("Semaphore released too many times")
ValueError: Semaphore released too many times
```

</details>

Not sure if these are good enough, since the file is massive and could contain private information (Which I think I filtered out now)

bleuthoot-sven, 2025-03-04:
Well, I don't know what we did, but it suddenly started working again. No idea what has changed to make it work again.

closed 2025-03-04 by bleuthoot-sven